openscad/openscad #2061: Rendering is not completing and getting stuck...

kdgkp98, 2017-06-18:
We are working on a module which has a input parameter as 'nsa' and we found the relations but it is not rendering completely i.e it is repeatedly getting stuck at rendering(at 985/1000) and if rendering got stuck, I was unable to either open/render new Openscad files. I have to close and open it once more to open any other Openscad files.
I think the problem is with the too much (extra)  precision of sin and cos values Openscad considers by default and due to that it is being stuck??
I don't know whether my guess is correct?? If it is right please help me to reduce the precision of Sin and Cos values up to 2 decimal places.(Is there any special way to use 'round' command to decrease precision in this case??) 
If my guess is wrong, Please suggest your solutions and  help me to complete my project...

My code is

module pattern(nsa)
{
    for (nsa=[125:0.1:132])
        { 
difference()
{
//main-polyhedron 
polyhedron( 
    points=[ [0,0,0],[54.5,0,0],[54.5,9,0],[1.88+20*sin(nsa)-3*cos(nsa),17.9-20*cos(nsa)-3*sin(nsa),0],[1.88+10*sin(nsa)-3*cos(nsa),17.9-10*cos(nsa)-3*sin(nsa),0],[1.88+10*sin(nsa),17.9-10*cos(nsa),0],[1.88,17.9,0],[54.5,9,9],[19.13,28.46,11],[54.5,0,9],[0,0,11],[1.88,17.9,11],[9.54,24.33,11],[11.47,22.03,11] ],
    faces=[ [0,1,2,3,4,5,6],[2,7,8,3],[7,9,10,11,12,13,8],[10,9,1,0],[11,10,0,6],[5,12,11,6],[3,8,13,4],[4,13,12,5],[1,9,7,2] ]
    );
    
  //cylinder
    $fn=100;
    translate([80.06,104,0])
    cylinder(h=11,r=98.4);
}  
 //middle-polyhedron   
translate([54.5,0,0])
polyhedron(
    points=[ [0,0,0],[54.5,0,0],[54.5,6,0],[0,9,0],[54.5,6,6],[0,9,9],[54.5,0,6],[0,0,9] ],
    faces=[ [0,1,2,3],[2,4,5,3],[4,6,7,5],[7,6,1,0],[5,7,0,3],[2,1,6,4] ]
    ); 
    
   //tria_tip(tip-polyhedron) 
    translate([109,0.1,0.5])
    polyhedron(
    points=[ [0,0,0],[29,0,0],[29,3,0],[0,5.8,0],[29,3,3],[0,5.8,5],[29,0,3],[0,0,5] ],
    faces=[ [0,1,2,3],[2,4,5,3],[4,6,7,5],[7,6,1,0],[5,7,0,3],[2,1,6,4] ]
    );
//u-plate(arc_latest) 
 translate([9.54,21.4+0.2*(nsa-125),5.5])
 rotate([(nsa-90),270,270])
    union(){
        
//semi-circular-part
$fn=100;
difference(){
    translate ([0,10.5,-1.5])
    cylinder(h=3,r=10.5);
    cube([21,20,3],true);
}
//rectangular-part
translate([0,5,0])
rotate([0,0,90])
cube([10,21,3],true);

//circular-extrusions
$fn=100;
translate([0,5,0])
cylinder(h=15,r=4.9);
translate([0,5,15])
cylinder(h=15,r1=5.5,r2=5);

//side-faces
//side-1
translate([8,-5,0])
cube([5,10,3],true);

//side-2
translate([-8,-5,0])
cube([5,10,3],true);

//tips
//tip-1
translate([5.5,-10,1.5])
rotate([90,90,0])
  polyhedron(
    points=[ [0,0,0],[10,0,0],[0,5,0],[10,0,3],[0,5,3],[0,0,3] ],
    faces=[ [0,1,2],[1,3,4,2],[3,5,4],[5,3,1,0],[4,5,0,2] ]
    );
    
//tip-2
translate([-5.5,-13,1.5])
rotate([0,90,90])
  polyhedron(
    points=[ [0,0,0],[10,0,0],[0,5,0],[10,0,3],[0,5,3],[0,0,3] ],
    faces=[ [0,1,2],[1,3,4,2],[3,5,4],[5,3,1,0],[4,5,0,2] ]
    );

}
}
 
}
pattern(129);

CONSOLE output after F5:

Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 81
Geometry cache size in bytes: 149592
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 923 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

If I press F6:
This is the situation
[image]
 

I am using OpenSCAD-2015.03-2 version on windows
Please help me...
Thank you for your valuable time....

t-paul, 2017-06-18:
It's probably just very slow as this looks like it's creating a large amount of vertices. Maybe try decreasing the `for()` steps and `$fn` values to see if it works in general.

kdgkp98, 2017-06-18:
Thanks for your suggestion but it didn't work,Though there is decrease in rendering speed till reaching 933/1000 and getting stuck there...
Please help me....

kdgkp98, 2017-06-18:
Thanks for your suggestion. Our code is rendering completely if changed for() from 0.1 to 0.5 and $fn=100 to $fn=30 but the output model is like the combination of new and old dimensions (like hull command), it is only at the curved and round shapes as shown in the images

[image]
[image]

Where as the output model should be like this

[image]
[image]

I was unable to identify the mistake in my code...
Please help me...

t-paul, 2017-06-18:
I suspect that's unrelated to the resolution changes as the initial post already produces the same (wrong) result.

Please post general questions about your model in the Forum at http://forum.openscad.org/ - this here is meant for bug tracking and feature requests. It will also reach more people that may be able to help there.

I'll close this for now, if the discussion would point to an actual bug in OpenSCAD, feel free to re-open.

closed 2017-06-18 by t-paul