openscad/openscad #2062: "View All" for empty preview sets camera at infinity

thehans, 2017-06-27:
Steps to reproduce:
1) Start openscad, with new, empty document.
2) Hit F5 to preview
3) Click the "View All" toolbar button

Observe bottom status bar now says "distance: inf"

Scroll wheel zoom no longer is able to affect the camera distance at this point.

This happens in both perspective and orthogonal views.

MichaelAtOz, 2017-06-27:
(well it is viewing _all_)

thehans, 2017-12-06:
This now freezes OpenSCAD and requires a force quit.

thehans, 2017-12-11:
I was looking at the various bugs I have reported and #2014 caught my eye as being the same issue(with OpenSCAD freezing now) as this one.

In that bug, @teepee pointed out that an infinite loop was occurring in GLView::showScalemarkers
So I tried turning off `Show Scale Markers` in the View menu, and the freezing no longer happens (of course camera distance still goes to infinity though)

thehans, 2018-12-01:
Fixed in PR #2610

closed 2018-12-01 by thehans