openscad/openscad #2063: Where is the projection?

RonaldoCMP, 2017-07-02:
<code>
epsilon = 0.001;
projection() cube([epsilon,10,10]);
</code>
This projection is void; however
<code>
epsilon = 0.001;
square([epsilon,10]);
</code>
is not.

PRouzeau, 2017-07-02:
cube vector is [x,y,z]
projection is done in z on plane x,y, so you are projecting a very thin line.
Projection have no thickness, while 2D primitives have a thickness of 1. Projections are for drawing export.
Square z default to 1, so you see it, but if you project it, you don't see more than on your first example

RonaldoCMP, 2017-07-02:
I am projecting to a very narrow rectangle to be precise.

Preview of 2D polygons seems to have a thickness of 1 (and this is a debatable topic ) but its render()ization has no thickness. The operator projection() generates a 2D polygon like any other and it has no thickness. 

Note that everything would be fine if epsilon was set greater than 0.001.

thehans, 2017-07-02:
Probably related to this issue:  https://github.com/openscad/openscad/issues/999

You can work around the issue with something like this:

```
epsilon = 0.001;
scaleAdjust = 100;

scale(1/scaleAdjust) 
  projection() cube(scaleAdjust*[epsilon,10,10]);
```

RonaldoCMP, 2017-07-04:
It seems to be.

The work around I have adopted is:

epsilon = 0.001;
scale([epsilon,1]) projection() cube([epsilon,10,10]);

kintel, 2017-09-09:
Closing as duplicate of #999.
@RonaldoCMP Feel free to add your test as a comment to #999 and we'll pick that up eventually

closed 2017-09-09 by kintel