openscad/openscad #2064: How to select custom editor font? Also, where's the menu?

mk-pmb, 2017-07-05:
I was happy to discover that there's an openscad PPA for Ubuntu trusty again. I added it, installed the openscad package, and it renders nicely, now even with diameter support for cylinders. However, it seems to have bad luck choosing the editor font: [Looks quite fancy](http://abload.de/img/openscad-default-fon93su2.jpeg) but I'd prefer boring monospace as shown in the leafpad editor for comparison.

I tried clicking the word "Editor" in hopes of a menu, but there's no reaction, and the squared x right of it probably means it's a panel caption masquerading for a main menu. I deleted `~/.config/OpenSCAD/OpenSCAD.conf` and tried again, same result. The config file re-appeared but doesn't seem to have a setting for the font either. I searched the documentation link in the readme for words like "setting", "config", "font", no luck. Couldn't find "font" in [the FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ) or the [UI chapter of the manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_User_Interface) either.

For now I'll try and work with my usual editor and use auto-reload as described in ["How to use an external editor"](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_an_external_Editor_with_OpenSCAD), and… nope, can't do that without the "Design" menu. So, even more urgent than fixing the font: How do I enable the main menu?

Openscad is the latest that apt can find in the "openscad/releases" PPA, probably 2015.03 because the status bar shows these numbers on the right. OS is Ubuntu 14.04.5 LTS trusty, desktop env is Xfce 4.10.

MichaelAtOz, 2017-07-05:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#I.27m_not_getting_any_menubar_when_running_OpenSCAD_in_Ubuntu.2C_how_can_I_get_it_back.3F

Once you get menus, Edit/Preferences/Editor can do your font.

mk-pmb, 2017-07-05:
Thanks for the quick reply! The fix worked, and now it even quits on Ctrl+Q, nice!

Would it seem feasible to add a "Known bugs for your OS" info box to the greeting screen for stuff like that? Of course now it looks like I should have been able to search the FAQ for "menu", but at the time I didn't actually expect I could really just enable it. Bad experience with other projects led me to believe the new toolbars with icon buttons were actually a replacement for the menu and the docs for external editor were out of date.

To have the least clutter in openscad core, the box could show a text file from a user-independent place like `/usr/share/doc/openscad/bughints.txt` if it exists, with the platform-specific content generated by platform-specific installers (in my case, a script in the PPA).

Corona688, 2017-08-04:
I doubt psychic prediction will be added to OpenSCAD any time soon.

mk-pmb, 2017-08-04:
@Corona688
> I doubt psychic prediction will be added to OpenSCAD any time soon.

That's why I suggested the inferior, inaccurate mechanism of text in an infobox.

t-paul, 2018-04-06:
Closing as there's not much more we can do than have an entry in FAQ.

closed 2018-04-06 by t-paul