openscad/openscad #2065: Feature Request: Allow for(… in …) as alias for for(… = …)

mk-pmb, 2017-07-05:
Reading the manual about list comprehensions, the `=` struck my eye. Why use syntax that resembles an assignment (both sides on equal level) for a selection (left side on deeper level)?
Example script:

```openscad
words = [ ["f","o","o"], ["b","a","r"], ["b","a","z"] ];
lastLetters = [for (w = words) w[2] ];
echo(lastLetters);
```

I'd consider it easier to read if I could use `in` instead of `=` there, so could we allow that as an alias? (Also in the other "for"s.)
Even more so if we continue the example with some flattening:

```openscad
allLetters = [ for (w = words, letter = w) letter ];
echo(allLetters);
```

When I take a second to think about it, I can perfectly understand that it's not really assignments. On the first casual read however, my mind parses it as being either an assumption about equality relation, or as two consecutive variable assignments to the effect of `letter = (w = words)`, so I wonder for a second why the "w" is repeated, as in both cases there's an expectation of transitivity for the equality relation. (In the assignment case, the expectation is about the state resulting from the assignments.)

Let's aim for a language where we can express programs in a way that makes them appear obviously correct on first read even to novices. As a side effect, it's much easier to teach a language if you can reduce the list of "Remember, in these cases it's not what it looks like". We could keep the `=` for compatibility and start using `in` in the docs (or my tutorials at least).

RonaldoCMP, 2017-07-05:
In your proposal, what would mean the following?

a = [for(x in 3) x];

If "in" is an alias, "x in 3" would be an odd form of assignement.

nophead, 2017-07-05:
for(x = 3) is indeed an assignment in OpenSCAD. I use it to replace assign() which is deprecated but still needed to force a new scope.

mk-pmb, 2017-07-06:
> for(x = 3) is indeed an assignment in OpenSCAD.

Thanks, didn't know that yet. Then it shouldn't be an alias, rather an alternate notation that only works with lists/vectors.

RonaldoCMP, 2017-07-06:
Em 5 de jul de 2017 22:01, "M.K." <notifications@github.com> escreveu:

for(x = 3) is indeed an assignment in OpenSCAD.

Thanks, didn't know that yet. Then it shouldn't be an alias, rather an
alternate notation that only works with lists/vectors.

Sorry but this doesn't answer my question and I don't know what *allletters*
in your code will be when variable *words* is a flat list.

mk-pmb, 2017-07-06:
Indeed, "works" was ambiguous. I meant that the "in" notation shall be invalid for variables that are not lists or vectors, so an attempt to use it like

> a = [for(x in 3) x];

shall fail. I don't know at which stage the type error can be detected and thus what kinds of failure are available. Since the invalid attempt is most probably a bug in the script, feedback should be as early as possible, best even fail the attempt to compile, 2nd best a runtime exception. If it's not feasible to stop execution and some value has to be assigned to x, it should be set to `undef`.

> what *allletters* in your code will be when variable *words* is a flat list.

I hope I understand "flat list" correctly, as in this example:

```openscad
words = [ "foo", "bar", "baz" ];
allLetters = [ for (w in words, letter in w) letter ];
echo(allLetters);
```

There will be up to 3 iterations, each with a `w` value that is neither a list nor a vector. If we can make this fail early, there shall be no assignment to `allLetters` because execution stops instead, or hasn't even started. If we can't prevent the assignment, the resulting value in `allLetters`  shall be `[ undef, undef, undef ]`.

mk-pmb, 2017-07-06:
I'll better make a new post for the 2nd update so you can see it via email, too:

While making up the example I discovered that it's actually valid to access items "inside" an integer value by indexes, as if the integers were lists that just happen to be empty. If `3[4]` is just as valid as `"foo"[4]`, it would be inconsistent to forbid item access via `for (… in …)`, and we should instead just make it always work as if the right-side variable is something with items in it. If it has no items, that just makes it empty, like empty lists or empty strings.

```openscad
mixed = [ "foo", "", 3 ];

flat_eq = [ for (outer =  mixed, inner =  outer) inner ];

// flat_in = [ for (outer in mixed, inner in outer) inner ];
// ^-- not yet available, thus use a stub:
flat_in = [
  mixed[0][0],
  mixed[0][1],
  mixed[0][2],
  // mixed[1] has no items inside it.
  // mixed[2] has no items inside it, either.
  ];

outOfBounds = [
  mixed[0][3],  mixed[0][4],
  mixed[1][0],  mixed[1][1],
  mixed[2][0],  mixed[2][1],
  ];

echo(flat_eq);      // ECHO: ["foo", "", 3]
echo(flat_in);      // ECHO: ["f", "o", "o"]
echo(outOfBounds);  // ECHO: [undef, undef, undef, undef, undef, undef]
```

mk-pmb, 2017-07-07:
Better stub:

```openscad
function each(x) = let (n = len(x)) (
  (n > 0)
  ? [for (i = [ 0 : n-1 ]) x[i] ]
  : []
  );

// flat_in = [ for (outer in mixed, inner in outer) inner ];
flat_in = [ for (outer = each(mixed), inner = each(outer)) inner ];
```

A reader who doesn't know the details could even assume the tertiary conditional operator to be just a performance tweak.

t-paul, 2019-11-21:
It does look lightly nicer in some cases, however I believe that would mean we would need to make `in` a keyword which would complicate things.

closed 2019-11-21 by t-paul

mk-pmb, 2019-11-21:
> we would need to make in a keyword which would complicate things.

Are there existing keywords or reserved stuff we could use instead? Maybe `<<`?

t-paul, 2019-11-21:
No, but the sort-of obvious choice would be `:` as quite a number of other languages do that for various reasons.

However there is currently no special case to handle `for()`, from language perspective the statement version is exactly the same as `cube(size = [1, 2, 3])`. This is also the reason that the slightly strange `for([1,2,3]) echo("hallo");` is valid.

mk-pmb, 2019-11-21:
I'd prefer something that depicts the direction of extraction.