openscad/openscad #2066: Export to path with spaces

darksnakezero, 2017-07-06:
When using the export function under windows and you export to a path that contains spaces, nothing happens.

Export to a simple path without spaces does work.

"Save as" works as intended. If it is too complicated to fix, at least a warning\error would be nice.

MichaelAtOz, 2017-07-06:
Which version of OpenSCAD?
What type of export?
Can you give a specific example of the path & filename?
Did you use the dialogue box to select the directory, or did you type the directory/filename?

MichaelAtOz, 2017-07-06:
FIXME: [Export CSG](https://github.com/openscad/openscad/blob/a84da9a110b46543520f7e2e73e69b0f071c90db/src/mainwin.cc#L2228) needs to use [completeBaseName()](http://doc.qt.io/qt-4.8/qfileinfo.html#completeBaseName), not baseName().

MichaelAtOz, 2017-07-06:
I can't reproduce this with 2015.03. or a recent snapshot. Windows 7/64
@darksnakezero Also what version of Windows?

darksnakezero, 2017-07-06:
Windows: Windows 10 Pro 64 bit
Openscad: 2015.03

working path U:\output.svg
non-working: C:\Benutzer\L ö\Desktop\output.svg.

German version I use:
From the menu: Datei->"Speichern Unter" works
Datei->Exportieren->Select any option
Then "choose Filename and path" window opens, I select said path and it closes without doing the export.

English translations could be:
File ->"Save As" working
File -> Export-> Any option -> "choose filepath"   not working, filepath window closes without error, but doesnt save the file

MichaelAtOz, 2017-07-07:
a. Check the console for any error messages.
b. can you try it with a directory with a space, but no unicode chars.

darksnakezero, 2017-07-07:
a) No error in console
b)
Path "U:\a space\output.svg" working
Path "U:\örk\output.svg" NOT working
So the problem really is the 'ö'. It is not really needed to support those paths I guess, but I would expect some warning popup in that case :-)
Again, I am sorry that I didnt test enough before writing that bug report.

MichaelAtOz, 2017-07-07:
Reproduce unicode export issue.
Export as CSG works, others don't

https://github.com/openscad/openscad/blob/a84da9a110b46543520f7e2e73e69b0f071c90db/src/mainwin.cc#L2263 
Seems to be the way to go.

kintel, 2017-08-04:
This sounds very much like a duplicate of #113, which is addressed in #1296. I'm not too happy with the solution in #1296 though. Any ideas on how to improve this would be appreciated. Now that NoWide has been accepted into Boost it might be a more viable solution than before though..

kintel, 2017-09-17:
Closing as duplicate of #113

closed 2017-09-17 by kintel