openscad/openscad #2067: Bad geometry, invalid triangles after exporting to stl

thehans, 2017-07-10:
I have a script, attached which is creating triangle faces in incorrect places, but the problem is not visibe in preview or render mode, until the file is exported to STL.

This happens for me on Linux Mint 18, using the openscad nightly, and also with version 2015.03-1

[spurious_flaps.scad.txt](https://github.com/openscad/openscad/files/1134206/spurious_flaps.scad.txt)
There is a specific rotation done in this script so that the cube points are perfectly diagonal, without the rotation I don't have any problem with the exported stl.

Link to stl that was exported
https://gist.github.com/thehans/3aff8ae94518baba513d776d2fc3196a

kintel, 2017-08-04:
An observation: This appears to _only_ be an issue if you preview first, then render. Clearing caches and rendering directly gives me a clean STL file.

It still puzzling why it doesn't show up when rendering though and should definitely be looked into.

thehans, 2017-08-04:
I just tried flushing caches (as well as going straight to render from a freshly opened openscad, with no preview first) and it makes no difference for me, the errors are always in the saved STL.

thehans, 2017-10-19:
I looked a little bit at the code, and I see that exporting to STL calls
CGALUtils::createPolySetFromNefPolyhedron3 (because this guarantees faces will be triangles i think?)

Whereas CGALRenderer converts CGAL_Nef_Polyhedron to CGAL_OGL_Polyhedron, and does not convert to polyset.  

So, since the render displays correctly, and error only shows after export, I'm guessing the fault is somewhere in CGALUtils::createPolySetFromNefPolyhedron3

thehans, 2017-10-19:
OK, so I see that ```CGALUtils::createPolySetFromNefPolyhedron3``` calls out to 
```GeometryUtils::tessellatePolygonWithHoles(verts, faces, triangles, nullptr);``` for the tesselation, and I think this is where its going wrong.

This comment above ```GeometryUtils::tessellatePolygonWithHoles``` caught my eye:

	One requirement: The input vertices must be distinct
	(i.e. duplicated must resolve to the same index).

When this object renders, the Console says it has 44 vertices, whereas I know that a level 1 menger sponge should have only 40 vertices, so 4 of them seem to be duplicated:
```
Top level object is a 3D object:
   Simple:        yes
   Vertices:       44
   Halfedges:     152
   Edges:          76
   Halffacets:     56
   Facets:         28
   Volumes:         2
```

I uncommented some debug statements in the code to list the vertices before tesselation, and they are as follows
```
0: -3.8551888792426325381e-07, 6.6773827711585909128e-07, 86.6025390625
1: 81.649658203125, 6.6773827711585909128e-07, 28.867513656616210938
2: 40.8248291015625, 70.7106781005859375, -28.867513656616210938
3: -40.8248291015625, 70.7106781005859375, 28.867513656616210938
4: 1.2462565791793167591e-05, 47.140445709228515625, 28.867513656616210938
5: 27.216548919677734375, 47.140445709228515625, 9.622516632080078125
6: 40.824817657470703125, 23.570234298706054688, 28.867513656616210938
7: 13.608280181884765625, 23.570234298706054688, 48.11251068115234375
8: 40.8248291015625, -70.7106781005859375, -28.867513656616210938
9: 27.216548919677734375, -47.140445709228515625, 9.6225156784057617188
10: 1.220046397065743804e-05, -47.140445709228515625, 28.867513656616210938
11: 13.608280181884765625, -23.570232391357421875, 48.11251068115234375
12: -40.8248291015625, -70.7106781005859375, 28.867513656616210938
13: 3.8551888792426325381e-07, -6.6773827711585909128e-07, -86.6025390625
14: -81.649658203125, -6.6773827711585909128e-07, -28.867513656616210938
15: -27.216548919677734375, -47.140445709228515625, -9.622516632080078125
16: -1.2462565791793167591e-05, -47.140445709228515625, -28.867513656616210938
17: -13.608280181884765625, -23.570234298706054688, -48.11251068115234375
18: -40.824817657470703125, -23.570234298706054688, -28.867513656616210938
19: -13.608280181884765625, 23.570232391357421875, -48.11251068115234375
20: -1.220046397065743804e-05, 47.140445709228515625, -28.867513656616210938
21: -27.216548919677734375, 47.140445709228515625, -9.6225156784057617188
22: -27.216562271118164062, 6.6773827711585909128e-07, 48.11251068115234375
23: -2.5828489924606401473e-07, 4.4736256654687167611e-07, 28.867494583129882812
24: -13.60826873779296875, -23.570211410522460938, 9.6224985122680664062
25: -40.8248291015625, -23.570211410522460938, 28.867513656616210938
26: 27.216535568237304688, 2.2698701229728612816e-07, 9.6224985122680664062
27: 54.43309783935546875, 6.6112635366266658821e-09, -9.622516632080078125
28: 40.8248291015625, -23.570211410522460938, -28.867513656616210938
29: 13.608267784118652344, -23.570211410522460938, -9.6224985122680664062
30: 13.608267784118652344, 23.570211410522460938, -9.6224985122680664062
31: -13.608267784118652344, 23.570211410522460938, 9.6224985122680664062
32: -40.8248291015625, 23.570211410522460938, 28.867513656616210938
33: 27.216535568237304688, 4.4736256654687167611e-07, 9.6224985122680664062
34: 13.60826873779296875, 23.570211410522460938, -9.6224985122680664062
35: 40.8248291015625, 23.570211410522460938, -28.867513656616210938
36: 40.824817657470703125, -23.570232391357421875, 28.867513656616210938
37: -27.216535568237304688, -2.2698701229728612816e-07, -9.6224985122680664062
38: -54.43309783935546875, -6.6112635366266658821e-09, 9.622516632080078125
39: 27.216562271118164062, -6.6773827711585909128e-07, -48.11251068115234375
40: 2.5828489924606401473e-07, -4.4736256654687167611e-07, -28.867494583129882812
41: -27.216535568237304688, -4.4736256654687167611e-07, -9.6224985122680664062
42: -40.824817657470703125, 23.570232391357421875, -28.867513656616210938
43: -13.608267784118652344, -23.570211410522460938, 9.6224985122680664062
```

I analyzed these and found that indeed 4 pairs of vertices are *very nearly* equal
```
24~=43
24: -13.60826873779296875, -23.570211410522460938, 9.6224985122680664062
43: -13.608267784118652344, -23.570211410522460938, 9.6224985122680664062
26~=33
26: 27.216535568237304688, 2.2698701229728612816e-07, 9.6224985122680664062
33: 27.216535568237304688, 4.4736256654687167611e-07, 9.6224985122680664062
30~=34
30: 13.608267784118652344, 23.570211410522460938, -9.6224985122680664062
34: 13.60826873779296875, 23.570211410522460938, -9.6224985122680664062
37~=41
37: -27.216535568237304688, -2.2698701229728612816e-07, -9.6224985122680664062
41: -27.216535568237304688, -4.4736256654687167611e-07, -9.6224985122680664062
```

@kintel Do you think this would trip up the tesselator?

thehans, 2019-12-23:
I just revisited this error and found that it is no longer reproducible using the latest nightly.  Yay!

closed 2019-12-23 by thehans