openscad/openscad #2071: Problem with closing a hollow form STL import

john-davies, 2017-07-22:
I'm not sure if this is an issue but it's similar to issue #495 which has been marked as fixed.

The scenario is:

1. Import an STL file which is a hollow form. Think cube with one face removed.
2. Attempt to close the hollow form with another cube, overlapping the original STL file so that there are no problems with just edges touching.
3. Rendering that shape gives an error - "WARNING: Object may not be a valid 2-manifold and may need repair!"
4. Exporting the object as STL and checking in Meshlab it seems that some of the normals on the original STL file are now facing inwards rather than outwards.
5. Exporting just the original STL file without the extra cube from OpenSCAD gives the correct result with all normals pointing in the correct direction.

Should it be possible to create a manifold model in this way with OpenSCAD? I'm using OpenSCAD version 2015.03-2 on Ubuntu Mate. 

Source file:

```
translate([-0.5,-0.5,0.2])
import("/home/john/Desktop/cube2.stl", convexity=10 );
cube([2,2,0.5],true);

```
STL file ( cube2.stl ) attached as a zip because GitHub won't allow me to upload an stl file directly
[cube2.stl.zip](https://github.com/openscad/openscad/files/1167291/cube2.stl.zip)

nophead, 2017-07-22:
>Should it be possible to create a manifold model in this way with OpenSCAD?

It isn't because OpenSCAD can only operate on manifold objects.

The issue you reference was exporting objects with internal cavities that were manifold, not objects with missing faces.

john-davies, 2017-07-22:
OK, thanks for the information, that's clear now. I'm closing this.

closed 2017-07-22 by john-davies

john-davies, 2017-07-23:
A small post script - I used Blender's "Solidify" option on the original imported STL file to create a valid manifold mesh and then everything worked as expected with OpenSCAD. Thanks for your help.

MichaelAtOz, 2017-07-23:
Thought bubble: What is Blender doing to achieve that?