openscad/openscad #2075: wrong height with minkowski function

Gurkengewuerz, 2017-07-27:
If you use the minkowski() function the height is also added.
I used the example code from the wiki. I expected a height of 1 but there was a height of 2.
```
minkowski()
{
  cube([10,10,1]);
  cylinder(r=2,h=1);
}
```
OpenSCAD Version 2015.03-2

If the function is right the wiki should be updated with a hint.

nophead, 2017-07-27:
Minkowski is short for Minkowski sum, which means all dimensions get added. The resulting shape will be 14 x 14 x 2

t-paul, 2017-07-27:
That's exactly what the [documentation](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#minkowski) also explains.

> (note that the outer dimensions of the box are now 10+2+2 = 14 units by 14 units by 2 units high as the heights of the objects are summed)

closed 2017-07-27 by t-paul