openscad/openscad #2076: Render() exposed as a function

TLC123, 2017-07-29:
I thought i posted this before but i cant find it.

If render() were exposed as a function, points and faces returned as lists could be subjected to some arbitrary operation before drawn as polyhedron or polygon in 2D case.
Related:  #1324  #301 

Benefit of exposing geometry this way is that a hole bunch of issues can safely be resolved in user space. with little commitment to future support.

```
My_polymesh = render_to_polyhedron(){ sphere(1) ; }
returns [points = [ [x, y, z], ... ], faces = [ [p1, p2, p3..], ... ]] for polyhedron.

My_polymesh = render_to_polyhedron(){ import("example012.stl", convexity=3); }
returns [points = [ [x, y, z], ... ], faces = [ [p1, p2, p3..], ... ]] for polyhedron.

My_polygon = render_to_polygon (){ circle(1) ; }
returns [points = [ [x, y, z], ... ], paths= [ [p1, p2, p3..], ... ]] polygon .

My_polygon = render_to_polygon (){import("example013.dxf" }
returns [points = [ [x, y, z], ... ], paths= [ [p1, p2, p3..], ... ]] polygon .
```

nophead, 2017-07-29:
Yes it would be nice but the problem is OpenSCAD evaluates expressions before it creates the geometry, so it would be a major redesign.  You might have geometry that depends on expressions that depend on geometry, etc.

TLC123, 2017-07-29:
Sure, pre-sub-evaluation of anything  under render treated as a separate model and successiv replacement with returned geometry as an expression. Finally evaluate parent model. It could be several recursive evaluations. Cyclic references wouldn't be an issue though. I specifically proposing naming the feature render() since a time penalty is already expected of render().

I reintroduced this issue because i wanted to add a specific bounty, a bounty i'm willing to raise considerably considering its a major challenge if there is serious support.

I think the feature carries way more value than it seams at first glance. 
virtually every mesh algorithm in the book becomes low hanging fruit to solve in user space.
usage example: mesh post processing like 
subdivision, edge rounding, decimation, import repairs, object distribution over surfaces, measurements, collision tests.
In a singel blow one feature expands OpenScads usefulness considerably without breaking backwards compatibility. 

cheers

nophead, 2017-07-29:
I think there was a working PR submitted a while ago that did nearly the same thing called probe but it was a module rather than a function.

TLC123, 2017-08-01:
Disclaimer: Thinking out loud readings ahead.

I imagine  if one of openscads export modes were a text file of this polyhedralist [[points],[faces]]
a second instance of openscad could be called during expression evaluation and passed the sub program as a .scad file with the argument to export resulting geometry as this polyhedra expression file, that in turn are read back to parent file.  or piped back.

A bit hackish detour of course  and I'm sure there can be some internal shortcut around actual files , this without major changes to the entire thing.  One new export mode and one new expression that queries the outside file system. Hardly a major redesign. 

Parsing might be an issue that force child scad to be passed as a third parameter of the function , i don't know, but not very pretty. And obviously child scad cant see any global variables from parent or call any functions. Just what is passed by the function. Unless all that is passed too by some, to me unfamiliar, magic spell.

for illustration i wrote render() as call_openscad_command_line() 

resultingpolygonlist =
call_openscad_command_line ( command_line_:arguments, passed_params_from parent ) 
{ 
the_program_to_be_evaluated; 
}

side note, a CSV and txt read expression could also be useful for many.
mydata=import("myfile.csv");
mystrings=import(mytext.txt);

MichaelAtOz, 2017-08-02:
> mystrings=import(mytext.txt);

You can include <> a file, provided you can control it's layout, you just need to have variables, eg
```
t=["
first line\n
second line\n
etc
"];
u=[
"item one",
"item two",
"etc"
];
echo(t[0]);
echo(u);
```

```
Compiling design (CSG Tree generation)...
ECHO: "first line
second line
etc"
ECHO: ["item one", "item two", "etc"]
Compiling design (CSG Products generation)...
```
It would probably be straight forward to write a script to grab a text file and produce a .scad file in a similar format to include<> in your main program.

kintel, 2017-08-04:
As @nophead pointed out, the primary architectural challenge for this is that we need a mechanism for feeding back rendering results into the script evaluation. Having this would open for a bunch of interesting features (e.g. any module could be called as a function returning geometry). I don't see this happening very soon, but this is covered by the OpenSCAD2 proposal by @doug-moen (https://github.com/doug-moen/openscad2). Evolving by cherry-picking ideas or design from that document is possible, but at some point we'll break compatibility and need to think about consequences of that.

nophead, 2017-08-04:
@blobule's probe module seemed to get around the problem of getting values back from geometry somehow. Is there a reason why that wasn't merged?

MichaelAtOz, 2018-01-28:
> I don't see this happening very soon, but this is covered by the OpenSCAD2 proposal by @doug-moen (https://github.com/doug-moen/openscad2). Evolving by cherry-picking ideas or design from that [...]

@doug-moen @kintel, can you confirm my understanding that OpenSCAD2 is dead in the water, and unlikely to get any wind in the future?

If so, then where is the direction for the topic of this issue?
It keeps popping its head up in various guises and always circles back into a self-referential death spiral.

Having just gone through a long backlog of posts, it has come up in at least three areas lately.

There is Functional OpenSCAD, Parkinbot & Ronaldo's creative works ( & skin sweep etc) , and import/export as poly/gon/hedron format, and probably a few more in the dimmer history.

The main thing that is preventing that from coalescing into a very useful capability is the lack of a feature to take geometry and turn it into a list.
As someone said recently, in BASIC (and other languages equally dating me) [Peek & Poke](https://en.wikipedia.org/wiki/PEEK_and_POKE), in the right hands, turned an otherwise pedestrian language into something that could achieve astounding things.

While OpenSCAD is not pedestrian, there is much flare being shown in the above developments.
I think this Issue's proposal, ie `render()` as a function, is the most cohesive incarnation of the concept.
It implies that it is resource hungry (tho cache will assist), the name is right (unlike other proposals - read() etc). You don't need a 'special' import, as you can do `geom=render() import("my.stl");`.

So I think this should be adopted. [Probably after any upcoming release if there is one]

doug-moen, 2018-01-29:
> @doug-moen @kintel, can you confirm my understanding that OpenSCAD2 is
dead in the water, and unlikely to get any wind in the future?
> If so, then where is the direction for the topic of this issue?

I had a discussion with Marius about this a few weeks ago.

On my end, I can say that there is no agreed plan on the next steps, and
I'm not currently working on OpenSCAD2. I did write some code prototyping
my ideas, and that code has evolved into a separate open source project
called Curv. I'm putting all of my resources into Curv right now.

The key idea behind OpenSCAD2 is to make functions into first class values.
Everything else follows from that. If somebody wants to take leadership on
that, I can act as a consultant, since I understand all of the gotcha's
with language design and implementation.

The render() function provides access to the underlying representation of a
shape. It's a great idea. In an ideal world, it would be orthogonal to
OpenSCAD2, but actually, there is a connection. OpenSCAD has "shape
expressions", which compute shape values, but these are called
"statements", and they are parsed using a different grammar that is
incompatible with expressions. OpenSCAD2 solves the problem of writing
shape expressions that are parsed as ordinary expressions, which leads to
the ability to pass shapes as arguments to functions, return them as
results, and assign them to variables. And the proposed "render function"
is essentially a function that takes a shape as an argument.
In order to implement a "render() function", *something* tricky has to be
done in the parser. You could use the grammar I invented for OpenSCAD2 to
solve the problem, or you could find some other solution. Some of the
solutions are uglier than others.

NateTG, 2018-01-29:
>  ... OpenSCAD has "shape expressions", which compute shape values, but these are called "statements", and they are parsed using a different grammar that is incompatible with expressions. ...

Is that another feature of OpenSCAD that I'm not aware of, and if so, could you provide an example?

As mentioned, naively setting up `render()` as a function breaks the declarative paradigm of the language.   To make it work consistently with the way things are structured now basically requires passing the code that you want to have rendered to the render function as an argument.  So `render()` becomes OpenSCAD's version of eval(), and that's usually not a way you want to go.   From that perspective the 'render as module' makes a lot more sense since modules already have a suitable scope.

I'm brainstorming here - so this is probably not the best idea in the world - but how hard would it be to allow modules to have a return value in addition to a geometry side effect? (The interface might be a `return()` statement that can be used to pass data out of the module, and a special `$childdata` variable that is a vector of returned data from the modules in the scope of an operator module.)

Then something like `render()` could be implemented in userspace as a module by clobbering the OpenSCAD primitives and implementing a userspace geometry engine.

doug-moen, 2018-01-29:
The issue I'm talking about is only about the language syntax, and not
about the implementation during evaluation.

The OpenSCAD grammar distinguishes between "statements" and "expressions".
For example, '2+2' is an expression that adds two numbers and returns a
number, while 'union{cube(10);sphere(5.5);}' is a statement that unions two
shapes and returns a shape.

A variable definition has the form 'name = expression'.

The examples given for the use of the "render function" look like this:

My_polymesh = render_to_polyhedron(){ sphere(1) ; }

geom=render_to_polyhedron() import("my.stl");

It looks like 'render_to_polyhedron(){ sphere(1); }' and
'render_to_polyhedron() import("my.stl")' are expressions.

It looks like the syntax of a call to 'render_to_polyhedron' is:
  render_to_polyhedron() <statement>
and that this is an expression.

We can't actually implement this syntax, it would be ambiguous. We can't
embed the entire statement syntax in expressions like this.

The simplest fix is to require the statement to be surrounded by braces.
Eg, you would have to write

geom=render_to_polyhedron() {import("my.stl");};

doug-moen, 2018-01-30:
Once `render_to_polyhedron` is implemented, there will be a desire for other special functions that take shapes as arguments. We should create a general framework for defining such functions, and not just put special code into the Bison grammar that recognizes `render_to_polyhedron` as a keyword.

OpenSCAD currently has two kinds of functions (callable entities that take arguments and return a result). These are "functions" and "modules". Let's add a third kind of callable entity, and call it a "functor".

A functor call is an expression with the syntax
```
   functorName (argumentList) {statement}
```
This expression syntax can be recognized unambiguously by the parser, without the parser needing to know all of the functor names in advance. This syntax is also upward compatible with OpenSCAD2, in case that becomes important in the future.

Functor names belong to the function namespace.

We could support user defined functors:
```
   functor bounding_box() =
      let (mesh = render_to_polyhedron(){ children() })
      compute_bounding_box_of_mesh(mesh);
```

Or, maybe this is more user friendly, since the first line (the "function prototype") is a visual reminder of the calling syntax. Also, we avoid adding a new keyword:
```
   function bounding_box(){} =
      let (mesh = render_to_polyhedron(){ children() })
      compute_bounding_box_of_mesh(mesh);
```

doug-moen, 2018-01-30:
@TLC123 When a functor call is evaluated, the evaluator may convert the shape argument into a CSG tree, call the renderer, then convert the result into a list of triangles. So the evaluator can now invoke the renderer. Once evaluation is complete, then we proceed as usual: we construct the final CSG tree and execute the preview or render phase. Any geometry contained in functor calls will not appear in the final CSG tree.

I don't think there is any problem with "circular references". The evaluator can call the renderer, but not vice versa.

The only real problem with this is that rendering can be slow, and it can now happen inside the evaluator. This is the tradeoff for making OpenSCAD much more powerful. Suppose that the evaluator performs an expensive CGAL render that takes 2 hours. Then if you click Preview, you have to wait 2 hours for a CGAL render before the preview appears. The original design of OpenSCAD was to guarantee that Preview is fast by eliminating the possibility of doing renders during evaluation.

To put this in perspective, I bet that CGAL renders are much faster than renders performed entirely in the OpenSCAD language using the "Functional OpenSCAD" library. So I think this is an opportunity to speed up the kinds of programs people want to write in "Functional OpenSCAD".

nophead, 2018-01-31:
>I bet that CGAL renders are much faster than renders performed entirely
in the OpenSCAD language using the "Functional OpenSCAD" library.

I doubt it. CGAL is slower than a slow thing or as they would say here in
NZ "slow as".

doug-moen, 2018-01-31:
@doug-moen said: I bet that CGAL renders are much faster than renders performed entirely
in the OpenSCAD language using the "Functional OpenSCAD" library.

@nophead  said: I doubt it. CGAL is slower than a slow thing or as they would say here in
NZ "slow as".

Except that the OpenSCAD interpreter is slower than a slow thing which is itself slower than another slow thing. I think that beats CGAL slowness.

For example, I just wrote a tail-recursive function that iterates 1,000,000 times, and increments a counter on each iteration.
* In OpenSCAD, which has tail recursion optimization, the runtime is 7 seconds (timed using a stopwatch).
* In Curv, which does *not* have tail recursion optimization, and has a simple parse tree interpreter similar to OpenSCAD, the runtime is 1.1 seconds.
* In Python, I couldn't make the stack large enough without getting a segfault, but the iterative for loop version executed in .38 seconds. Python is faster than Curv because it has (a) an optimizing compiler that (b) compiles into byte code for (relatively) fast execution.
* Python is notoriously slow. Javascript is much faster.

The OpenSCAD interpreter is so extravagantly slow that I can't believe that it could beat CGAL in performance. Also, when I tried iterating 10,000,000 times in OpenSCAD, it aborted evaluation because it was taking too long, and that has got to be a problem for the functional OpenSCAD library. I ran this on the December snapshot.

NateTG, 2018-01-31:
> ... The OpenSCAD interpreter is so extravagantly slow that I can't believe that it could beat CGAL in performance. ...

There's no computer fast enough that it can't be made slow by silliness.  There are certainly scenarios where it's hard to believe that an interpreted renderer with bounding box logic wouldn't outperform the F6 render.

jordanbrown0, 2023-11-18:
PR #4478 includes the feature described.
```
My_polymesh = render({{ sphere(1) ; }});
echo(My_polymesh);
```
yields (reformatted for readability, and with lists of points and faces ellided):
```
ECHO: {
    min : [-0.809017, -0.951057, -0.866025],
    max : [1, 0.951057, 0.866025],
    center : [0.0954915, 0, 0],
    size : [1.80902, 1.90211, 1.73205],
    objects : [
        {
            points : [
                [0.154508, -0.475528, 0.866025],
               ...
            ],
            faces : [
                [0, 1, 2, 3, 4],
                ...
            ]
        }
    ]
}
```

coryrc, 2025-12-09:
> Just bear in mind, that in OpenSCAD the order, how functions and modules are called is not exaclty defined.

The context of this is Jordan's "immediate geometry" which is basically an scad snippet you fully evaluate (though I assume it inherits the current dynamic and lexical scope).

jordanbrown0, 2025-12-09:
> Provide a method to translate the structure into VNF.

It's naturally very similar to VNF.  Ignoring other fields and the actual contents, it's
```
{
    objects: [
        {
            points: [...],
            faces: [...]
        }
    ]
}
```
I haven't looked at the details of VNF, but I suspect that the conversion is `[ r.objects[0].points, r.objects[0].faces ]`.

VNF is a BOSL2-ism; it's not derived from core OpenSCAD.  Generating VNF would not be helpful to non-BOSL2 users.  It predates objects, so is necessarily less easily extensible than the structures possible today.  I would expect that BOSL2 would either move over to this format (or a subset of it), or would add a trivial conversion function.

>>We should not expose a "number of triangles" value, because although it's always possible to calculate and Manifold probably always calculates it, a different back end that is willing to generate polygons with more than three sides wouldn't.

>That wouldn't make the count inaccurate.

But it might mean that it's not something that's already calculated.  Also, it's not clear how useful a "number of triangles" is, if the results don't consist of triangles.

>> isEmpty() is len(faces)>0, perhaps summed across all objects.

> So? Should we not have "false" because we can write "!true"?

I'm kind of a minimalist where it comes to the core; anything that's trivial to do in userspace should be done in userspace.

But we're a long ways from that sort of decision.

>The context of this is Jordan's "immediate geometry" which is basically an scad snippet you fully evaluate (though I assume it inherits the current dynamic and lexical scope).

Yes, what I call "geometry literals", which yield "geometry values".  See [OEP8](https://github.com/openscad/openscad/wiki/OEP8%3A-Objects-%28dictionaries%3F%29%2C-Geometry-as-data%2C-and-Module-References), in the section "Geometry as data".

Geometry literals are evaluated in the current context (and their own nested contexts, of course), and internally yield a CSG tree - that's what a geometry value is, under the covers.  render(g) renders the CSG tree into a data structure of points and faces and other metadata.

`objects` is an array so that a future multi-color multi-material world can have more than one object represented, each with its own color/material/whatever metadata.

jordanbrown0, 2025-12-10:
> VNF is just a two-item list [points, indexed faces] that polyhedron(points=vnf[0], faces=vnf[1]) is useful.

The [OEP8](https://github.com/openscad/openscad/wiki/OEP8%3A-Objects-%28dictionaries%3F%29%2C-Geometry-as-data%2C-and-Module-References) / PR #4478 structure includes those two pieces of data.  (Actually, in theory it includes N instances of them, in support of hypothetical future multi-material/multi-color/multi-whatever features.)  Those two items, by themselves, are defined by `polyhedron()`.  It's putting them together into one array in that order that makes it a VNF.  Generating a VNF *per se* has no additional value to a non-BOSL2 user, because the data is already there in the existing data structure.  (It would have value to a BOSL2 user, because BOSL2 has functions and modules that operate on VNFs - but BOSL2 can supply a function that consumes the more general data structure and yields a VNF.)

> My point wasn't "VNF" is was to use a method to return the points and faces; don't go creating OpenSCAD lists that might not be needed and could be huge. If someone just wants volume or bounding box, it's completely unnecessary to do make those lists.

That's a fair point, though my instinctive reaction is that you can't compute the volume or bounding box *without* computing the full list of points and faces.  You might be able to avoid copying them.

But, hmm.  We're blithely talking about offering a method.  I'd assumed that we were talking about a method on the object returned by `render()`.  Such a method can only base its results on the contents of the object.  So what would that object contain?  If it contains the list of vertexes and faces, then ... it contains them.  If it doesn't contain them, then if you want to be able to calculate them it would need to contain the CSG tree, which would bring in a host of other issues.  It would also make me wonder exactly what the `render()` function itself does.

So maybe we are really talking methods on geometry values, that would **replace** the `render()` function.  (And, since it's an opaque data structure, it can hide C++ mechanisms that OpenSCAD can't handle, like lazily calculating things and caching the results.)  That is, for a geometry value `g`, we might have `g.render()` that returns a list of { points, faces, other per-object metadata } pairs, and a `g.box()` that returns a bounding box, and so on.

But:  looking at the methods you mention, I think the only one that doesn't require fully calculating the list of vertexes and faces is `getTolerance()`, and if I understand that one correctly it's just returning an input parameter.

Calculation-wise, as soon as you ask for anything it has to do basically all of the calculation.  (Theoretically it might not do the bounding box, volume, or area calculations, but since I believe those are linear the savings is pretty low.)

Storage-wise, as soon as you ask for anything it has to store the vertex and face lists.  Internally, it could store them more efficiently - roughly 24 bytes per vertex plus 12 bytes times the number of triangles, versus maybe (educated guesses) 72 and 72.  But:  if it directly returns the vertexes and faces, that's the only copy that needs to be maintained, while if it stores them internally then it has to retain that list **even if** it's been returned as an OpenSCAD data structure.

> > Also, it's not clear how useful a "number of triangles" is, if the results don't consist of triangles.
> So don't call it on things without triangles.

How would you know?  Do cubes have triangles?  Do spheres have triangles?

> Maybe, but isEmpty() isn't trivial; you even were unsure how to do it because you wrote "perhaps".

Actually, come to think of it, it's probably simpler than I thought:  it's probably `len(r.objects)>0`.  I can't immediately come up with any rationale for having an `objects[]` entry that's empty.  (Though, hmm, that's a caution point:  you wouldn't be able to assume that there's always an `objects[0]`.)

jordanbrown0, 2025-12-11:
> I'm assuming geometry literal does create geometry and not just CSG (it's not obvious),

Just like any other OpenSCAD block, a geometry literal generates a CSG tree and the resulting geometry value is that CSG tree.

> so a "render" function isn't doing even kind-of the same thing.

It consumes a CSG tree and emits data describing a mesh.  It does exactly the same thing that render() and F6 Render do.  (Or, rather, that distinguishes F6 Render from F5 Preview.)

> Or "mesh()" which is self-documenting and literally the word you needed to describe what using the wrong word does...

Except for all of that other existing usage.

> [methods on built-in types are] pretty trivial...

Maybe.  They'd really be helpful on strings, so that we'd have a place to put all of the string operations that wouldn't consume global namespace.

>>I don't know what this is: NumProp, NumPropVert

>These are how Manifold applies colors and arbitrary information to faces and vertices. I guess this duplicates OpenSCAD, but you would want the equivalent or plan to switch OpenSCAD to using it after dropping CGAL.

Manifold and I have a disagreement:  it does face coloring, and I think we need volumetric color (or, more generally, volumetric properties).  You need to know what kind of filament to make the infill out of.

I do have a space reserved in the schema for that:  objects is an array of objects.  Each color would get one object, and would add (in addition to points and faces) a color specification.  (Or material specification or "this is a support blocker" specification or...)

> It might be better to hoist [Status] to the warning/error system instead. Well, hmm, that raises some questions about warnings and errors in geometry literally. This probably will be obvious when that's decided.

I think they're ... warnings and errors.  As with any other warnings and errors in OpenSCAD, depending on your settings either your run is aborted or you'll get something incomplete.

> Given there's not even a consistent theoretical foundation for how [volumetric color] would work, I don't think it's worth making the current implementation worse for something that is at least a decade off of even possible (beyond colorscad-type simplicity).

I don't think it's nearly that difficult.  I think there are straightforward boolean-math definitions for all of the operations.  There are a couple where you have to flip a coin to decide how it should behave (should overlapping union take color A or color B?) but I don't think it matters which side of the coin comes up.  I should really do an OpenSCAD-language POC.

> It's not just storage space, but operations on the data. Volume() written in OpenSCAD would be dog slow (orders of magnitude) 

I did say that I would be happy to have those supplied as built-in features.

> and probably have unfixable floating point issues.

Balderdash.  Everybody fixates on floating point issues.  I'm pretty sure there would be zero point zero floating point issues in calculating volume.  Yes, there would be an error of one part in ten to the minus 15 or so, and nobody would care in the slightest.  What do you think a C++ implementation would do differently?  It's not like C++ uses some magical data type that's immune to those problems.  With 64-bit floats, floating point inaccuracy is an issue **only** when comparing values.

> We're caching the Manifold object anyway, so there's no case where that data instantly goes away.

Hmm.  Good point.

coryrc, 2025-12-11:
> ```
> csg = {{cube(5);}};
> r = csg.render(); // r is an object with many methods
> if( !r.is_empty() ) {
> ```
> 
> What type is `r`?
> 
> If it's an OpenSCAD ObjectType, then its methods can only access data that's in the object,

I think you're forgetting every method is a closure, and these closures are being created in C++ so they can retain shared pointer to the `Manifold` object.

> If we went this way, with methods, I would think in terms of putting the methods all on the geometry value, and having it cache results internally.
> 
> 
> Thus, you would say `csg.mesh()`, `csg.volume()`, et cetera, and that value would cache the CSG-to-mesh results.

I'm fine with that too. I wouldn't refer to it as "csg" anymore.

```
g = {{cube(5);}}; // g is an object with many methods
echo(g.csg()); // returns a string which gets printed
if( !g.is_empty() ) {
  m = g.mesh();
  v = g.volume();
  a = g.surface_area();
  g2 = g.warp(function (vertex) return [vertex.x*2, vertex.y/2, vertex.z^2]);
  show(g2); // ?? I think have the same feelings as you regarding just `g2;` versus something else like this. PythonSCAD uses show though, so it kinda has precedent.
}
```

jordanbrown0, 2025-12-12:
> I think you're forgetting every method is a closure

You're right, but eww.

> I wouldn't refer to it as "csg" anymore.

I was just using the variable name from the example.  I'd call it a geometry value and my canonical variable name for those is `g`.  (But I'd say that conceptually it's still a CSG tree, with methods to turn it into other things and behind-the-scenes caching of those results.)

jordanbrown0, 2025-12-12:
Data points on my performance comment, to about one significant figure:

```
points = [ for (i=[0:999], j=[0:999]) rands(0,100,3) ];
```
Generate a million random points:  1100ms.
```
warped = [ for (p=points) 0 ];
```
Trivially warp that list: 1300ms, less the 1100ms to generate the original, is 200ms.
```
warped = [ for (p=points) [ p.x*2, p.y/2, p.z^2 ] ];
```
Warp a million points: 1900ms, less 1100 to generate the original is 800ms for the warp, less 200ms to for a trivial warp, is 600ms for this actual warp expression.

A built-in warp operation like you show is going to have to iterate across the point list, creating a vec3 for each point (to pass to the function), then calling the function, and turning the resulting vec3 back into internal form.  The function call will add a bit of overhead to the expression in the pure-OpenSCAD list comprehension above.  If the iteration, function call, and vec3-from/to-internal are **free**, the built-in will speed up that operation by 200ms out of 800ms, a 25% speedup.  It's dominated by the expression, and that's the same in both implementations.

And note that with a million vertexes we're still under two seconds.

That's on an i7-13700, which is not a slouch but is not top of the line either.

jordanbrown0, 2025-12-12:
Amplifying a bit on my "not general enough" comment... warping the vertexes alone isn't enough.  You also need to take the line segments that cross the bend and break them up and warp them.

coryrc, 2025-12-12:
I pretty much agree with everything you said in the former two posts.

Now on to benchmarking. To establish a baseline, I have these runnable examples.

tl;dr:

* Unlike what I expected, the sqrt lookup does not seem to effect runtime significantly.
*  C++ is unsuprisingly 2 OOM better. 
* My setup is 1100ms. Memory allocations add 650ms. Math and lookups add 400ms. 1e6 funcalls add 500ms.

```
$ time ./openscad -o /tmp/out.echo ~/tmp/just-echo.scad 

real	0m0.021s
user	0m0.014s
sys	0m0.007s
```

Now repro your findings:

```
$ cat > /tmp/jordans_warp.scad
points = [ for (i=[0:999], j=[0:999]) rands(0,100,3) ];
warped = [ for (p=points) [ p.x*2, p.y/2, p.z^2 ] ];
echo(len(warped));
$ time ./openscad -o /tmp/out.echo /tmp/jordans_warp.scad 

real	0m2.121s
user	0m2.025s
sys	0m0.096s
```

My claim is that the lookup makes things more costly, so I wrote and ran this:

```
$cat ~/tmp/benchmarking.scad 
points = [ for (i=[0:999], j=[0:999]) rands(0,100,3) ];
//warped = [ for (p=points) 0 ];
//warped = [ for (p=points) [ p.x*2, p.y/2, sqrt(p.z) ] ];

module run() {
  let(my_sqrt = function(x) sqrt(x)) {
    warped = [ for (p=points) [ p.x*2, p.y/2, my_sqrt(p.z) ] ];
    echo(len(warped));
  }
}
run();$
$time ./openscad -o /tmp/out.echo ~/tmp/benchmarking.scad 

real	0m2.221s
user	0m2.128s
sys	0m0.092s
```

Well, at least for this example, I was wrong. It is not significantly worse. And probably most functions you call in a hot loop won't be nested much deeper than this.

Lets compare against C++ using vec3:

```
$cat ~/tmp/just_vec3.cpp
#include <fstream>
#include <iostream>
#include <random>
#include <manifold/manifold.h>
#include <manifold/meshIO.h> // Includes ExportMesh and MeshGL

using namespace manifold;

#include <iostream>

double my_sqrt(double in) { return std::sqrt(in); }

void run(std::vector<vec3> &myvec) {
    std::vector<vec3> out;
    out.reserve(1000000);
    for (int i = 0; i<1000000; ++i) {
	auto old = myvec[i];
	out.push_back({old[0]*2.0, old[1]/2.0, my_sqrt(old[2])});
    }
    std::cout << out.size() << "\n";
}
	    
 
int main()
{
    std::random_device rd;  // Will be used to obtain a seed for the random number engine
    std::minstd_rand gen(rd());
    std::uniform_real_distribution<> dis(0.0, 100.0);

    std::vector<vec3> million;
    million.reserve(1000000);
    for (int n = 0; n < 1000000; ++n)
	million.push_back({dis(gen),dis(gen),dis(gen)});
    run(million);
    return 0;
}

$g++ -o /tmp/just_vec3 -O2 ~/tmp/just_vec3.cpp -I/opt/prg/manifold/include -L/opt/prg/manifold/build/src -lmanifold
$time /tmp/just_vec3 
1000000

real	0m0.040s
user	0m0.035s
sys	0m0.004s
```

That's 2 orders of magnitude better than OpenSCAD language. Not surprising.

Let's shift the OpenSCAD example to using a function:

```
$cat ~/tmp/benchmarking2.scad
points = [ for (i=[0:999], j=[0:999]) rands(0,100,3) ];

function converter(p) = [ p.x*2, p.y/2, sqrt(p.z) ];

warped = [ for (p=points) converter(p) ];
echo(len(warped));
$time ./openscad -o /tmp/out.echo ~/tmp/benchmarking2.scad && cat /tmp/out.echo 

real	0m2.775s
user	0m2.665s
sys	0m0.108s
ECHO: 1e+6
```

versus the exact same thing without function:

```
$cat ~/tmp/benchmarking3.scad
points = [ for (i=[0:999], j=[0:999]) rands(0,100,3) ];

warped = [ for (p=points) [ p.x*2, p.y/2, sqrt(p.z) ]];
echo(len(warped));
$time ./openscad -o /tmp/out.echo ~/tmp/benchmarking3.scad && cat /tmp/out.echo 

real	0m2.284s
user	0m2.160s
sys	0m0.124s
ECHO: 1e+6
```

and compare to a more-appropriate trivial case:

```
$cat ~/tmp/benchmarking4.scad
points = [ for (i=[0:999], j=[0:999]) rands(0,100,3) ];

warped = [ for (p=points) [ 0, 0, 0 ] ];
echo(len(warped));
$time ./openscad -o /tmp/out.echo ~/tmp/benchmarking4.scad && cat /tmp/out.echo 

real	0m1.937s
user	0m1.773s
sys	0m0.164s
ECHO: 1e+6
```

and just the setup:

```
$cat ~/tmp/benchmarking5.scad
points = [ for (i=[0:999], j=[0:999]) rands(0,100,3) ];

echo(len(points));
$time ./openscad -o /tmp/out.echo ~/tmp/benchmarking5.scad && cat /tmp/out.echo 

real	0m1.198s
user	0m1.117s
sys	0m0.080s
ECHO: 1e+6
```

jordanbrown0, 2025-12-12:
Adrian implementing a feature in BOSL2 is probably two orders of magnitude easier than implementing the same feature in core OpenSCAD in C++.  (Heck, I don't know whether Adrian even knows C++.)

Picking a random example, it is absolutely implausible that we would ever allow your `[ p.x*2, p.y/2, p.z^2 ]` warp function to be added to core OpenSCAD, so either it isn't possible, or we let somebody implement it in OpenSCAD-language.

coryrc, 2025-12-12:
> Picking a random example, it is absolutely implausible that we would ever allow your `[ p.x*2, p.y/2, p.z^2 ]` warp function to be added to core OpenSCAD, so either it isn't possible, or we let somebody implement it in OpenSCAD-language.

The point is said function is passed to the `warp()` method rather than applying it yourself to the matrix.

All of https://github.com/BelfrySCAD/BOSL2/blob/master/transforms.scad is written how I'm suggesting: it sets up the transformations and does the work in C++ (using `multmatrix`).

coryrc, 2025-12-12:
Similarly, on first glance, I think https://github.com/BelfrySCAD/BOSL2/blob/961ed6d8ff70da10861d873b06df628554fa1e8f/vnf.scad#L388 could be rewritten to be lambda for some transformation and greatly speed up things?

[function vnf_triangulate](https://github.com/BelfrySCAD/BOSL2/blob/961ed6d8ff70da10861d873b06df628554fa1e8f/vnf.scad#L1184) could simply call the Manifold triangulate and be OOM faster.

I can't be the only one who gives up on preferred precision because models take too long to iterate on.

jordanbrown0, 2025-12-12:
> The point is said function is passed to the warp() method rather than applying it yourself to the matrix.

And what does the warp() method do with it?

It must call the OpenSCAD interpreter to evaluate it.

So, for the warp method, given an internal representation of the mesh, it must:

- for each point
  - create a VectorType containing the point
  - evaluate the function (which, note, creates a vec3)
  - add the result to an internal representation of the result

A DIY method must:
- Convert the internal mesh to OpenSCAD data
- for each point
  - evaluate the function (creating a vec3)
  - add the vec3 to an OpenSCAD representation of the result.

So the question is how the loop overhead, everything *but* the function, compares.  And my results suggest that (for your sample function) the cost of the function evaluation is 3x as high as the cost of the loop overhead, so even if the loop overhead is reduced to **zero** the overall performance only improves 25%.

coryrc, 2025-12-12:
This is a lot like Python and numpy, if you've ever used it.

Sure, you *can* do anything with nested lists, and it lets you. But it's so, so much slower than setting up your data properly and calling the optimized functions.

coryrc, 2025-12-12:
> Can we agree that we'll want a way to extract the mesh out to OpenSCAD data for whatever processing isn't in the core?

I acknowledge it'll be present. :-)

> And can we agree that once we have that, we can benchmark a process using that versus an experimental implementation of a warp function, and see how much the warp function actually gains?

I can think of better examples it would be nice to compare, like `vnf_triangulate` vs Manifold's triangulate. But generally yes.

coryrc, 2025-12-14:
So my problem with supplying the mesh (vertices and faces) instead of allowing transformations on the data, is exactly the problem we discuss in [4478](https://github.com/openscad/openscad/pull/4478#issuecomment-3650207368): transformations on mesh vertices and faces are not meaningfully cached; they will run every time you press F6.

`warp` is actually the worst (or best?) example to use, since it can't (always/trivially) be represented by a transform in CSG format and so its CSG would need to be `polyhedron`, just like you need to when operating on a list of vertices in OpenSCAD language. For `warp` though, we don't need to actually generate the text `polyhedron(...)` (unless and until exporting CSG) -- we can use a shared pointer to the Manifold object created by warp, and use a hash of the Manifold as its CSG key? i.e. I'm thinking about this scenario:

```
function tf(p) = [ p.x*2, p.y/2, p.z^2 ];
s = {{my_shape();}};
module with_warp()
  difference() { s; s.warp(tf); }
module with_vnf() {
  warped = [ for(p=s.points()) tf(p) ];
  difference() { s; {{ polyhedron(warped, s.faces()); }} }
}
```

Naively we still must still do the warp every time we hit F6 (unless we traced the entire environment used by `tf` and any possible dependencies and compared them) but we can have the CSG internal representation of `warp` avoid the very large string encoding of `polyhedron()` and use a non-exportable "warp(#hash_of_Manifold)" as a lookup key for the actual Manifold object when render time comes. If exporting CSG it'd, of course, need to generate the `polyhedron(...)` string.

But, I supposed both `difference`s can be cached, so even if the key of `with_vnf`'s warped mesh is huge, it still can save running `difference` again.

coryrc, 2025-12-14:
Any transformation which doesn't take a lambda can be encoded into small CSG. Notably I'm thinking of Manifold's triangulate, but since CSG is actually a strict subset of OpenSCAD language, adding a new transform must pick one of these three options:

1. It's equivalent to an existing transform, so write it out as that.
2. Add a new OpenSCAD builtin supporting it
3. Export CSG as `polyhedron(...)` (always possible)

coryrc, 2025-12-14:
> It is currently the case that all script calculations are run on F5 or F6. For example, all the screw threads on every screw in my library are generated even if many are the same. The mesh parameters to polyhedron are used as the cache key.

It could be a union of a few simple CSG transforms instead, right? Threads should be able to be `linear_extrude` with twist, cylinder for the main body, and maybe a couple odd shapes for start and end?

nophead, 2025-12-14:
The ends spiral inwards in some cases.

jordanbrown0, 2025-12-14:
I think that one of the essential points is that we don't **know** all of the use cases, and we don't know all of the ways to implement those use cases.

If you're talking about which cases we should try to implement as builtins, I think we should get the basic feature in place and then figure out what special cases to implement as builtins.

But here's a few more:
* Wireframes
* Dynamic sizing and positioning
* Extrusion (sweep, dynamic transformation)
* Analysis (do these parts overlap?)

coryrc, 2025-12-14:
> I think that one of the essential points is that we don't **know** all of the use cases, and we don't know all of the ways to implement those use cases.

If I close my eyes, I cannot see.

> If you're talking about which cases we should try to implement as builtins, I think we should get the basic feature in place and then figure out what special cases to implement as builtins.
> 
> But here's a few more:
> 
>     * Wireframes

This means edges becomes cylinders, right?

>     * Dynamic sizing and positioning

rescale, scale and measurements already covers this?

>     * Extrusion (sweep, dynamic transformation)

So everything in here currently in OpenSCAD-language: https://github.com/BelfrySCAD/BOSL2/blob/master/skin.scad
      
>     * Analysis (do these parts overlap?)

`{{ intersection() { a; b; } }}.Volume() == 0` uses the cache and already does this.

I think if the API is going to include vertices and faces, it should encourage the user to not use them when an alternative exists. But I can't really think of a way to do so that's not stupid (i.e. get_mesh_but_be_sure_you_really_need_this() is dumb).

jordanbrown0, 2025-12-14:
> If I close my eyes, I cannot see.

Returning points and faces is a very general-purpose tool.  It is in the nature of general-purpose tools that it is difficult or impossible to know all of the uses to which they might be put.  We can give examples of cases that we can think of today, but we have no idea what we might think of tomorrow, or what other people might think of.

jordanbrown0, 2025-12-14:
But let's ask the community...

coryrc, 2025-12-16:
> You were just arguing that people would find the DIY "for" loop more obvious than the `{{ scale(...) ...}}`.
> 
> So, in your first example, where did "points" come from? 

BOSL2 generic_threaded_rod function.

> Or, in your second example, where did "other_shape" come from?

It's something holding the shape before scaling:

```
function bent_bolt() = let (
vnf = generic_threaded_rod(...),
vnf2 = vnf_bend(vnf, ...),
vnf3 = [[for(p=vnf2[0]) [2*p.x, p.y, p.z]], vnf[1]]
) vnf3;
```
instead could be: 
```
function bent_bolt() = let (
vnf = generic_threaded_rod(...),
vnf2 = vnf_bend(vnf, ...),
s = {{scale([2,1,1]) vnf_polyhedron(vnf2);}
) s.points_and_faces();
```

Now, obviously scale is the worst example because it's only a constant 100 times faster, but it's triviality makes for a small-but-complete example with minimal boilerplate.

coryrc, 2025-12-31:
[Email list message](https://lists.openscad.org/empathy/thread/VC2DF6UVMLY7AZMBJVC4FK4G5L3F5AZV) makes me think of more methods to have:

`num_meshes()` and `mesh(int)`, like `$children`/`children(int)`, but for distinct (non-overlapping) meshes in the geometry.

jordanbrown0, 2026-01-01:
(I've been trying to use "shape", to distinguish terminology from the "object" data structure concept.)

Yes, a void is a closed and connected set of faces with the normals pointing in, versus pointing out for a solid.  The way to determine that is, I believe, to calculate the volume of the shape, and that when you calculate the volume a solid will come out with a positive volume and a void will come out with a negative volume.  (Using "volume" here in its three-dimensional-measurement, length/area/volume, sense, not in its "disjoint chunk of space" sense.)

But my point is that if you've separated sets of faces so that you know that ABCD, EFGH, and IJKL are disconnected sets of faces, and you've done the volume calculation to determine that EFGH is a void, you still don't know whether EFGH is inside ABCD or inside IJKL, and cannot trivially determine which it is.

Though, come to think of it, it isn't as hard as I thought.  I thought you were going to have to calculate the intersections of EFGH and ABCD, and EFGH and IJKL, to determine which one had the non-empty intersection.  But if your geometry is all well-formed, with no shapes that cross other shapes' boundaries, you don't have to do that.  All you have to do is take one vertex from EFGH and determine whether it's inside ABCD or IJKL.  I don't know how to do that, but it's got to be far easier than calculating the intersection.

coryrc, 2026-01-01:
> So rather than render() just getting the polyhedron equivalent set of all faces, can't some topology come with it.

It *could*, but it *doesn't* ATM.