openscad/openscad #2077: Feature Request: add "offset" as option to linear_extrude

mrkmg, 2017-07-31:
Add the ability to define an "offset" to the linear_extrude. For non-symmetrical shapes, it is currently not possible to create a smooth transition to a slightly smaller size. Scale will scale from origin, which only works for symmetrical object.

mrkmg, 2017-07-31:
I am not well versed in C/C++ (Minor experience in school). I was looking to see if this is something I could implement. I do not think I have enough understand of how this was written or how to deal with 3d geometry to work on this.

I started by looking at https://github.com/openscad/openscad/blob/master/src/linearextrude.cc to see if that is where I would start work on that. Looking though, and I have no idea what is going on there. Any pointers on where I would even start?

I am willing to learn to make this a reality. It would make many of my creations much better!

thehans, 2017-07-31:
That code just handles the parsing of the parameters, so you'd need to add support for this "offset" parameter there.  But the real work on the geometry happens here:
https://github.com/openscad/openscad/blob/b48b525adeeaa5a320c5d81ec575a9ae83cf0987/src/GeometryEvaluator.cc#L678-L730

thehans, 2017-07-31:
I think the difficult part with this is going to be figuring out which vertices in one layer connect with vertices in the next, since the number of vertices may change as rounded corners/chamfers are added.

mrkmg, 2017-08-01:
I will give it a go. 

My naive idea would be to connect from the smaller side to the larger side, and just search for the closest vertex. I have gone through in my head and it sounds like it could be a good starting point.

Now, I need to go peak at the offset code, and work from there.

TLC123, 2017-08-01:
A working offset feature could possibly be further exploited to generate rounded end caps of extrusion.
for simple shape i don't se any major issues.
One hard nut probably is complex shapes with small islands and holes that might vanish in an offset before
 the last level is reached how is those cases handled.

mrkmg, 2017-08-01:
Okay. So a couple hours in and I am making some progress.

I have a question though on how an Outline2d actually represents an outline.

I have the following scad file:
```
linear_extrude(10)
square([10, 10]);
```

I added in some debugs, and am looping through the outline passed into `static Geometry *extrudePolygon(const LinearExtrudeNode &node, const Polygon2d &poly)`;

specifically, I am doing the following:
```
std::vector<Outline2d> outlines = poly.outlines();
for (unsigned int j = 0; j < outlines.size(); j++) {
    for (unsigned int k = 0; k < outlines[j].size(); k++) {
          Eigen::Matrix<double, 2, 1> v1 = outlines[j].vertices[k];
          qDebug() << j << ":" << k << ": " <<  v1(0, 0) << "," << v2(1, 0);
    }
}
```

When I run that, I get the following output:
```
0 : 0 :  0 , 0 
0 : 1 :  10 , 10 
0 : 2 :  10 , 10 
0 : 3 :  0 , 3.21143e-322 
```

Normalizing the rounding errors, I get:

```
0 : 0 :  0 , 0 
0 : 1 :  10 , 10 
0 : 2 :  10 , 10 
0 : 3 :  0 , 0
```
Making some assumptions, I assume it's making two triangles.

[0, 0], [10, 10], [0, 10]
and
[10, 10], [0, 0], [10, 0]

My question is how do I know which ones are the proper outside faces to be able to work up from there? In my ideal world, I would have the following verticies for a square:

[0, 0], [0, 10], [10, 10], [10, 0]

Are all polygons always triangles? Are they always *right* triangles? If so, is there something else down the rendering path that detects they are joined together and should be rendered as a solid square? A little insight here could help me get closer to this.

I am trying to look at `static void add_slice` to understand what is going. Without the understanding of PolySet, Polygon2d, and how the engine uses them I feel lost. At this point, I doubt my ability to implement even the simplest of functions.

Again, totally willing to learn!

mrkmg, 2017-08-02:
So, after a night of no sleep, I have something sort of working. It only work for very simple shapes, and fails and even segfaults for MANY shapes. I know the method I am using is not going to work for many cases, but after getting nothing for 10 hours, this is a huge breakthrough for me. I kind of wish I had committed at every compile so I could go back and look at all the failures. Some of them had some pretty funny results.

This is where I am at right now:

https://github.com/mrkmg/openscad/tree/linear_extrude_offset

The following file shows it working:

```
$fn=21;
linear_extrude(10, offset=-1)
difference() {
    square([20, 20], center = true);
    translate([10, 0])
    circle(7);
}
```

mrkmg, 2017-08-02:
Some of the basics I still need to think about:

1) Allows for the different types of offset (delta, r, chamfer)
2) How to handle a different number of edges from the bottom to the top layers
3) How to handle a different number of outlines from the bottom to the top layers
4) How to handle disappearing holes or islands.

I think this would all become much easier if I could somehow map all existing lines between the two faces. My original naive idea was to use "closest". I realized quickly how that would fail though. At this point I am at a logic standstill, but I will keep going on it.

TLC123, 2017-08-02:
simplest just test for simple shape and throw an error on complex shapes

2. probably the easiest. Find the layers with most edges. and re-sample all other layers to that number by breaking  some edges in two. If a shape has many parts repeat once for each.
 find the skin module in list-comprehension-demos for reference.

1.3.4 is trickier. Maybe a custom offset function that attaches an identity to each edge and tracks connectivity throughout extrusion and offsetting?

mrkmg, 2017-08-02:
I did some research on ClipperLib to see if it were possible to "track" which vertex maps to what. It does not seem possible in that lib. I will look to see what they do and see if I can implement something that does track them.

mrkmg, 2017-08-02:
@TLC123, Okay, so looking at the skin demo is pretty confusing. I've no idea what actually going on there. Man, 3d space stuff is hard... lol.

Corona688, 2017-08-02:
People have been pining for this and related features for at least **four years**.  This is still the best it has:  https://github.com/openscad/list-comprehension-demos

I will note that what you're looking for, in this specific case, looks a lot like the "convex hull" operator restricted to one axis.

t-paul, 2017-08-02:
I wonder if giving a function as parameter would help (basically looking at what implicitcad can do). Problem is of cause that feature does not exist at this point either. It might be possible to add this without total redesign though.

mrkmg, 2017-08-02:
So, I had an idea, and I am very well aware it may be stupid and not work.

Take a difference of the smaller side from the larger side (positive or negative offset). Take any vertex which lies on a point of the bottom side, and assign it a z of h1. Take any vertex which lies on a point of the top side and assign it a z of h2.

Here is what I am thinking, but without the z changes. 

[image]

Not quite sure how to accomplish assigning the z values yet as the Polygon2d does not have z values.

mrkmg, 2017-08-03:
I have made some good progress (I think). I put an update in PR #2079

your-friend-alice, 2021-11-07:
workaround (for some cases):
```
minkowski() {
    linear_extrude(0.1) text("test");
    cylinder(d1=4, d2=0, h=4);
}
```