openscad/openscad #2078: Bug? ECHO doesn't escape strings in a vector

MichaelAtOz, 2017-08-02:
```
t=["first line\nsecond line"];
echo(t);
echo(t[0]);
```
Produces
```
Compiling design (CSG Tree generation)...
ECHO: ["first line\nsecond line"]
ECHO: "first line
second line"
Compiling design (CSG Products generation)...
```
Issue #838 may relate.

thehans, 2017-10-18:
@MichaelAtOz I commited a fix for this, can you try a nightly build?

MichaelAtOz, 2017-10-19:
@t-paul is a ludite like me likely to have the the ability to do a nightly build for Windows?

t-paul, 2017-10-28:
Fixed by #2126.

closed 2017-10-28 by t-paul

t-paul, 2017-10-28:
Building Windows snapshots is mostly automated but manually started, but uses a Linux based environment ([mxe.cc](http://mxe.cc/)). I wonder if there's a way to set the build up on Travis for really automatic build and upload to the snapshot folder.