openscad/openscad #2082: Cmake Build does not work for distros that install FindEigen3.cmake

ckelloug, 2017-08-08:
Open Suse Leap 42.3 installs a FindEigen3.cmake file that is provided by Eigen3 but not installed.  The openscad CmakeLists.txt file spells variables related to Eigen3 differently than the find script does and as a result does not work even though the find script finds Eigen3 fine.

The following patch against commit 9b0378836bd7af24fa67f93fa456c57cc3fdbbcb shows the changes to make openscad work with Eigen3's FindEigen3.cmake script

--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -29,8 +29,8 @@ set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_SOURCE_DIR}/cmake/Modules/")
 set(CMAKE_PREFIX_PATH "$ENV{OPENSCAD_LIBRARIES}")
 
 find_package(Eigen3 REQUIRED QUIET)
-message(STATUS "Eigen3: ${EIGEN3_VERSION_STRING}")
-include_directories(${EIGEN3_INCLUDE_DIRS})
+message(STATUS "Eigen3: ${EIGEN3_VERSION}")
+include_directories(${EIGEN3_INCLUDE_DIR})
 
 find_package(Boost 1.36 REQUIRED COMPONENTS filesystem system thread regex program_options QUIET)
 message(STATUS "Boost: ${Boost_MAJOR_VERSION}.${Boost_MINOR_VERSION}.${Boost_SUBMINOR_VERSION}")

kintel, 2017-08-08:
The challenge here seems to be that Eigen provides both an `Eigen3Config.cmake` file, as recommended by CMake (https://cmake.org/cmake/help/git-master/manual/cmake-packages.7.html#creating-packages) and a `FindEigen3.cmake`. The former is installed by default but the latter is not, except that some Linux distros do install and package it.

Our build system should probably handle both, or at least have a mechanism for picking up the correct file or using the correct variables.

donbright, 2017-08-18:
```
"Requirements

Eigen doesn't have any dependencies other than the C++ standard library.

We use the CMake build system, but only to build the documentation and unit-tests, and to automate installation. If you just want to use Eigen, you can use the header files right away. There is no binary library to link to, and no configured header file. Eigen is a pure template library defined in the headers. "
```
https://eigen.tuxfamily.org/

```
Eigen exports a CMake target called Eigen3::Eigen which can be imported using the find_package CMake command and used by calling target_link_libraries as in the following example:
cmake_minimum_required (VERSION 3.0)
project (myproject)
find_package (Eigen3 3.3 REQUIRED NO_MODULE)
add_executable (example example.cpp)
target_link_libraries (example Eigen3::Eigen)
```

https://eigen.tuxfamily.org/dox/TopicCMakeGuide.html

yeah i am very confused as to why a header-only template library needs a build system. we have spent a lot of time over the years dealing with eigen compatability and installation problems. is it worth it? how much faster is it than other free 4x4 matrix codes out there on the web? 

i would vote to replace eigen with some other free matrix math and get rid of it.

unfortunately it looks like 'libsvg' is now inside of openscad and it depends on eigen too.

donbright, 2017-08-18:
i am currently downloading a 4 gigabyte opensuse distro so that i can install to VM and create what is probably a 4 line patch.

ckelloug, 2017-08-18:
I just sent the following message to the Eigen team who have disabled their bug tracker account creation feature.

Dear Eigen Team,

I would have reported this as a bug but you have disabled bug tracker account creation.

I noticed that your 3.3.4 distribution provides cmake/FindEigen3.cmake  It however does not seem to install it when you install eigen from the sources.

Some distros like Open Suse Leap 42.3 do provide this package as part of their eigen packages.  Is it an omission that this file is not installed by the source eigen install?  It seems

like one because packages that use cmake and eigen cannot find eigen without it.

This query is part of an attempt to help the openscad maintainers resolve a bug that I reported.

Thanks in advance.

kintel, 2017-08-18:
@donbright If we find an alternative library to eigen we can try that. We do, however, use eigen for more than just matrix multiplication. The main other use is encapsulating all vector math and vector type encapsulation across the datatypes being used.

donbright, 2017-08-19:
i just installed OpenSUSE leap 42.3 inside of a virt-manager qemu vm, built openscad with qmake-qt5 then build the cmake tests, and there were no problems with eigen3

if you have some kind of shell log that can show the error, i can keep digging otherwise i dont know what im trying to fix here.

donbright, 2017-08-19:
as long as libsvg is needed we have to keep eigen around.

also i am using openscad master branch from github. in case that makes a difference

donbright, 2017-08-19:
also, eigen provides a pkg-config file ( eigen3.pc ) so we should probably just use that instead of dealing with cmake headaches. (i think i had a dead branch somewhere that did that)

kintel, 2017-08-19:
CMake should be able to hide which underlying mechanism is used to find Eigen3 (FindEigen3, eigen3.pc or  Eigen3Config.cmake). I guess we just need to read up on docs + source code of the 3 alternatives and implement this in a clean way.

The cmake build system is still experimental, so it's likely that we need to apply some conventions on top of this to manage builds if we want to move on using the cmake build system.

donbright, 2017-08-20:
Well like i said, i can't reproduce this error. I installed opensuse 42.3 with openscad git master and it worked fine, cmake found eigen3 without any problems. What other distro has this problem? 

For the long term, however, it is possible to replace a bunch of the CMakeLists library-finding code with very short pkg-config calls using cmake's own pkg-config interface. When the OpenSCAD CMakeLists were originally written, a lot of that infrastructure was not available.

ckelloug, 2017-08-20:
Sorry for the wild goose chase.  I must have slightly buggered something in my leap 42.3 install dealing with all the different software I have to build for the various work I'm doing.  A number of them seem to have mutually exclusive eigen requirements and as a result, lots of installing eigen, uninstalling eigen, reinistalling egien. . .  I do think it's true that the cmake stuff in openscad is not consistent with eigen3's find script but apparently it doesn't matter unless you also broke whatever else it is that I broke.

donbright, 2017-08-20:
did you install eigen3 using it's own source code? or did you always use OpenSUSE package manager? what mutually exclusive eigen requirements are there? between different versions of eigen3 ( 3.0 vs 3.3? ) was anything using eigen2?

ckelloug, 2017-08-22:
I had trouble with the openscad build with cmake and Eigen and decided it might be the distro's install of Eigen since I've had trouble with eigen before on other packages.  I remembered seeing that the version and location of eigen in the cmake configure for openscad didn't show up right. Since Distro versions of things like Eigen often don't quite work as expected, I uninstalled the distro's version with the package manager and built and installed Eigen 3.3.4 from source.  It looked like the find script installed as part of the distro's version didn't get uninstalled and cmake wanted to use it when I tried to build openscad with the eigen I installed from source.  I think whatever problem I had is probably a corner case.  It just happened to highlight that the openscad build and eigen's find script call variables by different names.  Since I had done a clean install of the distro a week previous, I didn't think my install was unclean but apparently it was.  I remember there was some quantum chemistry software that needed eigen 2 that caused a problem with an earlier Suse version and some other non openscad thing I was working on had a complaint about needing an earlier eigen 3 version but I don't remember what the package was.

donbright, 2017-09-03:
@ckelloug @kintel thanks,

i want to apologize profusely for screwing this up so bad. . . i was testing the tests CMakeLists.txt not the gui CMakeLists.txt ... too quick and did not read what you were saying! my bad. 

yes it is broken. extremely broken.

we are not the only people with this problem, see for example https://github.com/pybind/pybind11/blob/master/tests/CMakeLists.txt

patch is on issue2082fix branch, tested on ubuntu 16, mint 18, suse 423, solus

t-paul, 2022-01-19:
Change seems to be in `master` and builds are `cmake` only for a while now. OBS builds 15.3, 15.4 and Tumbleweed without issues.

closed 2022-01-19 by t-paul