openscad/openscad #2083: Torus primitive

Corona688, 2017-08-09:
As part of learning the ropes with OpenSCAD I've built a new primitive, the torus.  It accepts major and minor radii (or diameters) as inputs.

[torus.diff.zip](https://github.com/openscad/openscad/files/1213176/torus.diff.zip)

It plays nice with $fs, etc, and produces a good mesh with no duplicates, non-manifold vertices, or other garbage so far as Teepee and I can tell.

Nonetheless it misbehaves when you put it inside a difference() and I'm not sure why.

difference() { torus(10,2); cylinder(d=5, h=10, center=true); }

It acts like you put the torus through a hull() before subtracting the cylinder, i.e. filling the donut hole.

Since the mesh is OK, I have no idea why.

thehans, 2017-08-10:
Commenting from my phone so cant see the example but you could have inverted faces causing problems.  Have you tried View -> "Thrown together" on the plain torus?

This will highlight backwards faces as magenta.  Then you can flip the order of vertices to fix.

Corona688, 2017-08-10:
I did check this first.  No backwards faces are apparent in "thrown together".  Blender doesn't see any backwards faces either, nor any non-manifold vertexes.  I sent Teepee an STL to check and he found nothing.

http://burningsmell.org/3d/junk/torus.stl

MichaelAtOz, 2017-08-10:
[image]

There is a clue. 
I frequently cut problematic STLs with Netfabb to have a look inside (with 'triangulate cut' un-checked), I couldn't see anything.
The above is with it checked, where is should close the cut with a face, on each piece.

That looks like what you describe, filling in the donut.

I have no idea what's happening, tho...
When cut on other axis, there is not problem.

This could be a red herring, but the STL has lots of coords with n.ne-16 etc, ie very small numbers.

kintel, 2017-08-11:
I haven't looked at the code, but is there need for a C++ implementation when we have torus in MCAD?
 https://github.com/openscad/MCAD/blob/5ff8ce964c71f59db463a8adf81bb22a2a48dee3/regular_shapes.scad#L207

Corona688, 2017-08-11:
I think my version plays better with $fs, rotate_extrude() produces seriously chunky torii sometimes, but strictly speaking no, this was a learning exercise.

MichaelAtOz, 2017-08-11:
I do like the look of these tori (toruses?).

Corona688, 2017-08-11:
michaelatoz:  Actually what I get in CGAL is completely filled, not just a flat polygon at the center.  Kind of a tablet shape.  It cuts the cylinder out of that.

Corona688, 2017-08-12:
```
render() {
difference() {
    import("torus.stl");
    cylinder(d=5, h=10, center=true);
}

translate([55,0,0]) difference() {
    torus(20, 5);
    cylinder(d=10, h=20, center=true);
}
}
```
[image]

Exact same mesh has different results when imported vs generated.

Corona688, 2017-08-12:
I FOUND IT!  CGAL was making bad assumptions because I was telling it a bad assumption.

```
//auto p = new PolySet(3,true); // true assumes convex shape
auto p = new PolySet(3,false); // nothing with a hole is convex!
```
Now it works great.
[working-torus.zip](https://github.com/openscad/openscad/files/1219390/working-torus.zip)

MichaelAtOz, 2017-08-13:
Do you want to produce another .stl so I can see what happens in Netfabb?

Corona688, 2017-08-13:
The mesh hasn't changed at all.

MichaelAtOz, 2017-08-13:
Well something is wrong with it, so presumably a generic OpenSCAD STL error?
Yes, I did the same netfabb cut with a differenced cube, and it had the same filled-in hole surface (with one 'inside' face showing - the red) as the above.

Doing the same cut, on a netfabb primative torus has no problems...

Corona688, 2017-08-13:
I've been over that torus so many times with a fine-toothed comb that I really **don't** think there's anything wrong with the mesh at this point.  All the vertices are attached to each other, all normals face the right direction, there's no apparent duplicates or degenerate triangles, according to Blender and according to OpenSCAD.

You're cutting right down the middle, where there's a line of vertexes, try offsetting the torus a tiny bit on Z before splitting so the vertexes aren't fighting for which side they end up upon.

MichaelAtOz, 2017-08-13:
Yep, that was it.

m-bob, 2018-08-23:
I'm confused about how to use this. I found 3 relevant downloadable files here. [1] torus.stl is a specific torus model, and not the generator. [2] torus.diff.zip contains a diff file. I expected to find a torus generator script. The downloadable torus.stl is not for the torus generator, but for a specific torus. I don't know how to use the torus.diff file or the working-torus.patch files.

I tried git apply torus.diff but that failed. Can you help? How do I get the generator script?

Thanks!

m-bob, 2018-08-23:
Sorry! Still can't use this, but I now understand that the code in for a modification to OpenScad itself ad not what I thought it was.

t-paul, 2019-11-21:
Closing as there seems to be good enough option to do in user space.

closed 2019-11-21 by t-paul