openscad/openscad #2084: Feature Request:  Vector Slicing

Corona688, 2017-08-12:
Here's my try at at a flexible array-slicer which dovetails with OpenSCAD's existing types and syntax:  cut(vector,template), where where template can be an array index, a range, or a vector - a vector containing ranges, array indexes, and vectors.

I've built and torture-tested a crude implementation:
```
in=["a","b","c","d","e","f","g","h","i"];

a=cut(in,3);
b=cut(in,[3]);
c=cut(in, [ [1,2], [3,4] ]);
d=cut(in, [1:3]);
e=cut(in, [ 0, [1:3] ]);
f=cut(in, [ 0, [[1:3]] ]);
g=cut(in, undef);
h=cut(in, []);
i=cut(in, -1);
j=cut(in, [-1]);
k=cut(in, [-1,-1,-1], "default");

echo("cut([...], undef)=", g); // should be undef
echo("cut([...], [])=", h); // should be []
echo("cut([...], -1)=", i); // should be undef
echo("cut([...], [-1])=", j); // should be [undef]
echo("cut([...], [-1,-1,-1],\"default\")=", k); // should be ["default","default","default"]

echo("cut([...], 3 )=",a); // should be d
echo("cut([...], [3] )=",b); // should be [d]
echo("cut([...], [[1,2],[3,4]] )=",c); // should be [[b,c],[d,e]]
echo("cut([...], [1:3] )=",d); // should be [b,c,d]
echo("cut([...], [0, [1:3] ] )=", e); // should be [a,b,c,d]
echo("cut([...], [0, [[1:3]] ] )=", f); // should be [a,[b,c,d]]
```
[cut.zip](https://github.com/openscad/openscad/files/1220140/cut.zip)

OpenSCAD's range type isn't currently able to specify "all possible elements" right now, but I have half an idea how to do that on the backend and am wondering how to add that to the language.

t-paul, 2017-08-13:
This feels quite complex, a bit like `search()` which has a complicated implementation and is hard to understand.

What do you think about pulling in the slicing and indexing ideas from [Sequences.md](https://github.com/doug-moen/openscad2/blob/master/rfc/Sequences.md) (not necessarily with that exact notation) which is a more comparable to other languages? That would not have the template feature, but for most cases that could be replaced by the declarative access to the array elements, I think.

Corona688, 2017-08-13:
Wouldn't that be nice.

That's not going to materialize out of nowhere.  We have to start somewhere.  Consider this a step towards it.

runsun, 2017-09-14:
To me it's indeed too complicated, for the reason that it intends to do too much work that requires complicate parameter structure. For example, 

`cut( arr, [[1,2],[3,4]])`

What is the usage percentage of this? 1%? 2%? A range slicer can be widely used in the code, not a good idea to ask 98% usage carry the burden to make that 2% work. 

If this burden is lifted, the api can be just:

`arr = [ a, b, c ]`
`cut( arr, 1)` // [ b, c ]
`cut( arr, 1,2)` // [ b ]
`cut( arr, 0,-1) // [ a, b ]
`cut( arr, -1 ) // [ c ]

Example 2: in your line:

`cut([...], 3 )=",a); // should be d`

is not a good ideal for the same reason: packing too many features into it, especially when it can be easily achieved by either arr[3], or, if the api is simplified, cut(arr,3,4)[0].

I haven't got a chance to see your code yet. In my experience, doing array slicing in OpenSCAD is pretty expensive, even with the "simplified version" that I suggested above. I once have a function `range(arr, I, j,k)` that I used widely, but later on removed it from my code due to low efficiency. I'll check your code later to see if the algorithm is more efficiency than mine.

Corona688, 2017-09-14:
Thank you for your constructive comments.

On thinking about it, I agree that some of its syntax is redundant.  I can reduce it to cut(a,1,2) without changing a whole lot or eliminating any of its more useful abilities.

My cut's implementation is actually extremely simple - nothing but a tree walk.  Its usage is also extremely simple, so simple that "simplifying" it means removing features from OpenSCAD or throwing in pointless if(input = valid) crash_anyway() clauses.  My **examples** are awful.  They're **hand-picked** for the sole reason of being awful.  I was showing off that it never crashed, and confused you all instead.  :P 

I'll be back with something a little better.

runsun, 2017-09-14:
I didn't realize that your cut.zip is written in C. Thought it's written in the OpenSCAD language itself, thus all that "expensive" talk. Please ignore that.

doug-moen, 2017-09-16:
In most scientific/linear algebra/array languages, the array indexing
operator a[i] is generalized so that i can either be a single index, or a
list of indices.

So instead of writing cut(a, [1,2,3]) or cut(a, 1, 2, 3) you write
a[[1,2,3]].

For example, Python's numpy library uses exactly this syntax, and Matlab
(which OpenSCAD borrows from) uses a([1,2,3]).

Why not use standard syntax, rather than inventing something new?

MichaelAtOz, 2017-09-17:
+1 for standard syntax. The good thing about standards is that there are so many to choose from.

We also have Doug's argument elsewhere about core features and user space libraries. It is relatively straight forward to write any vector manipulation function. However the balancing arguments are; not having to reinvent the wheel every time, and consistency of coding standards; better to have an easily remembered syntax that everyone uses and means the same thing. So I vote for adding to the core (in most cases - at least until V2 library enhancements - when the core can be 'purified' and move things out into a std library)

Re standards, in general the Sequences.md that @t-paul linked above looks good. I'm agnostic [a..b] v's [a:b], but I think long term ranges and vectors should be aligned, the increment feature of ranges is equally handy for vectors. Anyway +1 for reductionism.

I'd also like to include negative offsets as 'from the end', given that OpenSCAD supports -0 we need to carefully choose the end reference. Many languages (which don't support -0) use -1 as the last element. However I suggest that -0 MAY be more consistent. {arguments welcome}. I suppose [-a,-b] -a<=-b? Unless supporting the reversal syntax mentioned in the link?

Are we differentiating vectors from strings? I've wanted a substring() for a while...so whatever slicing we get should work on both. (eventually)

As usual, these discussions don't help @Corona688 's desire for a feature. Perhaps we should code cut() into an OpenSCAD function for now?

doug-moen, 2017-09-17:
I'm proposing that `a[i]` be generalized so that `i` can be a list of indices.

Note that `children` already works this way. You can write `children(0)`, or `children([1,2,3])`, or `children([1:3])`.

@Corona688 has already provided a C++ implementation of `cut`. All that would be needed is to move this code into ArrayLookup::evaluate() in expr.cc.

doug-moen, 2017-09-17:
@MichaelAtOz said:
> We also have Doug's argument elsewhere about core features and user space libraries. It is relatively straight forward to write any vector manipulation function. However the balancing arguments are; [...]

In my Curv language, I have implemented most of the core features in a script file that is loaded at startup. Any core feature that can be implemented in Curv goes there, instead of being coded in C++. I highly recommend this: it makes the code smaller and easier to maintain, and makes it much easier to add features to the core.

Here are some OpenSCAD core language features that could be moved out of C++ and into a startup script:
* PI
* translate, rotate, scale, mirror -- implement using multmatrix
* square, cube -- implement using polygon, polyhedron
* search, lookup

doug-moen, 2017-09-17:
@MichaelAtOz wrote
> Re standards, in general the Sequences.md that @t-paul linked above looks good. I'm agnostic [a..b] v's [a:b], but I think long term ranges and vectors should be aligned, the increment feature of ranges is equally handy for vectors.

If `a[i]` is generalized so that `i` can be a range, then you can write `a[[1:3]]` to specify a slice. This can be implemented without changing the parser.

If we change the parser and implement `a[i:step:j]` as an alternate slice notation, then we create a useability problem, because in Fortran and Python, `a[i:j:step]` is slice notation in those languages. I see from the mailing list that lots of people with Python experience come to OpenSCAD.

In the OpenSCAD 2 design, I was trying to avoid this pitfall for Python programmers.

By the way, my syntax for unified ranges and slices has changed. In the new design,
`i..j` or `i..j by step` is a range value, equivalent to `[i:j]` or `[i:step:j]` in OpenSCAD. Note that there are no brackets in my new syntax. This means that you write `a[0..10]` or `a[0..10 by 2]` to create a slice. It's equivalent to `a[[0:10]]` or `a[[0:2:10]]` in OpenSCAD, but with no double brackets, and no confusion for Python programmers.

nophead, 2017-09-18:
A major pitfall is Python ranges don't include the final value but OpenSCAD
ranges do, requiring lots of -1s.

Corona688, 2017-09-18:
Good discussion.  I implemented this as a function because it doesn't require changes to the parser.  Changing what valid **syntax** is could mean digging into lexx/yacc/bison stuff and hoping I didn't add any unintended side-effects.

@doug-moen sees what I'm going for, I think.  OpenSCAD isn't very orthogonal yet -- i.e. we could pass ranges around in variables but generally don't since they're pretty opaque to the programmer.  I will take a look at ArrayLookup as suggested, maybe it's less hard than I thought.

@nophead makes an excellent point.  OpenSCAD ranges have no way to encode "empty" right now.  Certain kinds of loops will always generate off-by-one errors.  Nested loops could conceivably **grow** instead of terminating once they go beyond bounds.

Since it's been brought up:  Why not denote empty ranges with backwards numbers?  Backwards ranges have already been depreciated.  If we really need a backwards count we can always just do for(X=[0:10]) for(Y=-X).

t-paul, 2019-11-21:
Closing as #3115 will bring vector/array/string slicing eventually. This one does include parser changes, so it's a bit more complex, but I think it's worth the more concise syntax of just `v[1:]` to skip the first element.

closed 2019-11-21 by t-paul

kdomanski, 2021-02-26:
Pleas reopen, since #3115 to be stalled for over a year now.

t-paul, 2021-02-26:
No, sorry, the specific API given in the issue description is not going to be integrated, so reopening is not an option.