openscad/openscad #2085: Variables not referenced correctly in for loop

markroworth, 2017-08-13:
Using the following code, one would expect the values 0 to 50 in steps of 5:

```
a=0;
b=5;

for(i=[0:1:10])
{
    echo(a, b);
    a = a + b;
}
```

However, the following is output:

```
Saved backup file: C:/Users/Mark/Documents/OpenSCAD/backups/unsaved-backup-ypwV5684.scad
Compiling design (CSG Tree generation)...
ECHO: 5, 5
ECHO: 5, 5
ECHO: 5, 5
ECHO: 5, 5
ECHO: 5, 5
ECHO: 5, 5
ECHO: 5, 5
ECHO: 5, 5
ECHO: 5, 5
ECHO: 5, 5
ECHO: 5, 5
Compiling design (CSG Products generation)...
Geometries in cache: 109
Geometry cache size in bytes: 58600264
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 0 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds
```

Not quite sure what is going on here. Any thoughts?

t-paul, 2017-08-13:
The `a = a + b` is not a valid construct.

See FAQ: [Why am I getting an error when writing a = a + 1?](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_am_I_getting_an_error_when_writing_a_.3D_a_.2B_1.3F)

closed 2017-08-13 by t-paul

markroworth, 2017-08-15:
Thanks Paul. I think I understand the reasoning behind this (OpenSCAD is effectively declarative, although appearing to be procedural).
The example I posted was a boiled down version from something I was trying to achieve, which was to wrap a piece of non-equispaced text around a cylinder. The number of degrees per character depends on the width of the character, for which I had a look up, so was adding a different amount per character to an angle. Hence the angle for character x depends on the cummulative angles of all the prior characters. Am I right in thinking that I need to calculate these in another programming language and generate the AutoSCAD code? It wasn't clear from the explanation whether I can use a module to recurse back along the string being rendered to calculate the cummulative size.
Many thanks,
Mark 
Mark Roworthroworthm@yahoo.co.uk07870 575707

RonaldoCMP, 2017-08-15:
This function may help:

    // The accumulated sum of a list
    // Generates the list of partial sums of list l, that is, the i-th
element
    // of the list accum_sum(l) is the sum of all elements of l with
indices less
    // than or equal to i. The offset offs is added to each element of the
list.
    // It works with uniform lists of numbers, vectors, matrices, etc.
    function accum_sum(l, offs=0) = len(l) > 0 ? _accum_sum(l, offs): undef;
    function _accum_sum(l, offs=0, res=[]) =
        len(res) == len(l) ?
            res :
            _accum_sum(l, offs+l[len(res)], concat(res, [ offs+l[len(res)]
] ));
/*
echo(accum_sum([1,1,1,1,2], offs=1));
//ECHO: [2, 3, 4, 5, 7]
*/

t-paul, 2017-08-15:
Depending on the specific model, there's probably a number of ways to solve the problem. Another way could be recursive modules like shown on the "[Tips & Tricks](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Tips_and_Tricks#Stack_cylinders_on_top_of_each_other)" page.

I'd suggest to post the question on the [forum](http://forum.openscad.org/), that's a good place to discuss options.

laird, 2017-08-16:
markroworth, instead of incrementally adding to the angle, compute the angle from the loop variable.

E.g. rotate([0,0,i*b]) .... ;

Or to simplify a bit more, you can make a loop of ‘angle’ from 0 to 359 step b, then rotate([0,0,angle])...

markroworth, 2017-08-17:
Hi Laird,
I did try that, but because it's not an equispace font, I need to change the angle a different amount depending on the character being rendered, i.e. an 'i' takes a far smaller angle than a 'w'.
Thanks,
Mark 
Mark Roworthroworthm@yahoo.co.uk07870 575707