openscad/openscad #2089: OSX Snapshots Fail to Open

deckar01, 2017-08-18:
Every time I download a new developer snapshot on OSX it fails to launch from finder. I have tried allowing the unidentified developer in the security settings, but it just doesn't open.

The work around is always to launch it from the command line:

```sh
open -a /Applications/OpenSCAD
```

Once I do that I can open it from finder/spotlight just fine.

kintel, 2017-08-18:
A few questions which would help us find the cause:
* Which version of OS X are you using?
* Has this behaviour changed? (i.e. can you download old working snapshots
from http://files.openscad.org/snapshots/)
* Are there any traces of this in Console.app ?

 -Marius

deckar01, 2017-08-18:
MacOS Sierra 10.12.6 (on both a 2011 Air and a 2013 Pro)

The first time (last month) I installed a snapshot this happened. I tried the latest snapshot, then a snapshot from February and they both had the same issue. Now that I have the latest snapshot installed correctly, I tried to open the February snapshot again and it still has the problem, so this happens to each fresh snapshot.

<details>

<summary>Logs filtered by "OpenSCAD":</summary>

```
Aug 18 12:07:15 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000026.OpenSCAD[25855]): Service exited due to signal: Killed: 9 sent by CoreServicesUIAgent[25820]
Aug 18 12:07:34 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000027.OpenSCAD[25874]): Service exited with abnormal code: 1
Aug 18 12:08:09 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000028.OpenSCAD[25888]): Service exited due to signal: Killed: 9 sent by CoreServicesUIAgent[25820]
Aug 18 12:08:21 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000029.OpenSCAD[25899]): Service exited with abnormal code: 1
Aug 18 12:08:39 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x1000002a.OpenSCAD[25906]): Service exited due to signal: Killed: 9 sent by CoreServicesUIAgent[25820]
Aug 18 12:08:59 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x1000002b.OpenSCAD[25908]): Service exited with abnormal code: 1
Aug 18 12:10:27 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x1000002d.OpenSCAD[25944]): Service exited due to signal: Killed: 9 sent by CoreServicesUIAgent[25820]
Aug 18 12:10:32 jair OpenSCAD[25945]: Customizer feature not activated
Aug 18 12:10:32 jair OpenSCAD[25945]: Usage: /private/var/folders/l3/bf5bw49x5_567xd5m5nc1b8r0000gp/T/AppTranslocation/5C6C0AD4-9910-47AA-822F-1B869583D436/d/OpenSCAD.app/Contents/MacOS/OpenSCAD [ -o output_file [ -d deps_file ] ]\
Aug 18 12:10:32 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x1000002e.OpenSCAD[25945]): Service exited with abnormal code: 1
Aug 18 12:11:24 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000030.OpenSCAD[25957]): Service exited due to signal: Killed: 9 sent by CoreServicesUIAgent[25820]
Aug 18 12:11:57 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000031.OpenSCAD[25966]): Service exited with abnormal code: 1
Aug 18 12:35:22 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000006.OpenSCAD[798]): Service exited due to signal: Killed: 9 sent by CoreServicesUIAgent[678]
Aug 18 12:35:52 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000007.OpenSCAD[812]): Service exited with abnormal code: 1
Aug 18 12:37:29 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000008.OpenSCAD[857]): Service exited with abnormal code: 1
Aug 18 12:40:16 jair OpenSCAD[991]: assertion failed: 16G29: libxpc.dylib + 74307 [BF896DF0-D8E9-31A8-A4B3-01120BFEEE52]: 0x89
Aug 18 12:40:20 jair OpenSCAD[991]: BUG in libdispatch client: kevent[EVFILT_MACHPORT] monitored resource vanished before the source cancel handler was invoked
Aug 18 13:12:35 jair OpenSCAD[1255]: assertion failed: 16G29: libxpc.dylib + 74307 [BF896DF0-D8E9-31A8-A4B3-01120BFEEE52]: 0x89
Aug 18 13:12:42 jair OpenSCAD[1255]: BUG in libdispatch client: kevent[EVFILT_MACHPORT] monitored resource vanished before the source cancel handler was invoked
Aug 18 13:19:04 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x1000000b.OpenSCAD[1351]): Service exited due to signal: Killed: 9 sent by CoreServicesUIAgent[678]
Aug 18 13:19:16 jair OpenSCAD[1365]: Customizer feature not activated
Aug 18 13:19:16 jair OpenSCAD[1365]: Usage: /private/var/folders/l3/bf5bw49x5_567xd5m5nc1b8r0000gp/T/AppTranslocation/188FDD26-12C9-4FE1-93BB-37484A5AB797/d/OpenSCAD.app/Contents/MacOS/OpenSCAD [ -o output_file [ -d deps_file ] ]\
Aug 18 13:19:16 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x1000000c.OpenSCAD[1365]): Service exited with abnormal code: 1
```

</details>

deckar01, 2017-08-19:
I built master and the app file opened without issue. I tried building the dmg, but was blocked on a missing Sparkle header (https://github.com/openscad/openscad/issues/283).

kintel, 2017-08-19:
If you want to build dmgs from source you need to install Sparkle (`macosx-build-dependencies.sh -d`).

deckar01, 2017-08-19:
That worked. Thank you.

Running the dmg directly after building runs the program without any security warnings.

Uploading, downloading, then running the dmg on the development machine shows the security warning, but correctly opens.

Downloading and running the dmg on another machine shows the security warning, but it fails to open.

The relevant portion of the log seems to be:

```
Aug 18 23:34:33 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000013.OpenSCAD[2999]): Service exited due to signal: Killed: 9 sent by CoreServicesUIAgent[2863]
Aug 18 23:34:43 jair com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000014.OpenSCAD[3005]): Service exited with abnormal code: 1
```

"Killed: 9 sent by CoreServicesUIAgent" might suggest that the gatekeeper is rejecting the certificate.

kintel, 2017-08-19:
There shouldn't be any certificates involved in this process.
Does your dmg still work fine on the cmd-line on the other machine?
Could you try running both `open -a OpenSCAD` and `./OpenSCAD.app/Contents/MacOS/OpenSCAD` ?

deckar01, 2017-08-19:
`open -a OpenSCAD` and `./OpenSCAD.app/Contents/MacOS/OpenSCAD` work and they seem to completely bypass developer verification. Once either of those methods are used, running the app file normally starts working as expected.

I updated the description with screenshots of the security warnings.

Neither of the Macs I tested have this issue opening other unidentified apps from the internet.

I thought that `macdeployqt` generated a signature even if you didn't give it your own certificate for codesign.

I tried using my self signed certificate to build the dmg, but it failed when verifying the Sparkle objects.

```sh
$ macdeployqt OpenSCAD.app -dmg -no-strip -codesign="Jared Deckard"
...
ERROR: "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/objdump: '/Users/jared/openscad/OpenSCAD.app/Contents/Frameworks/Sparkle.framework/Versions/A/Resources/zh_TW.lproj/SUUpdatePermissionPrompt.nib': The file was not recognized as a valid object file\n"
Segmentation fault: 11
```

kintel, 2017-09-09:
About the original issue, looking at your logs, I noticed this:
```
Aug 18 12:10:32 jair OpenSCAD[25945]: Customizer feature not activated
Aug 18 12:10:32 jair OpenSCAD[25945]: Usage: /private/var/folders/l3/bf5bw49x5_567xd5m5nc1b8r0000gp/T/AppTranslocation/5C6C0AD4-9910-47AA-822F-1B869583D436/d/OpenSCAD.app/Contents/MacOS/OpenSCAD [ -o output_file [ -d deps_file ] ]\
```

-> this means that the following code path was chosen: https://github.com/openscad/openscad/blob/master/src/openscad.cc#L942

i.e., someone passed a `-p` or `-P` to the OpenSCAD cmd-line.
If you could figure out who did that, it would be very helpful.

deckar01, 2017-09-10:
It looks like Finder appends a process ID argument.

https://stackoverflow.com/questions/10242115/os-x-strange-psn-command-line-parameter-when-launched-from-finder

FreeSCAD had the same problem and posted a workaround in this thread.

https://forum.freecadweb.org/viewtopic.php?t=4447

kintel, 2017-09-11:
I know, but that's not supposed to happen post OS X 10.8..

deckar01, 2017-09-11:
It happens on 2 different machines with MacOS 10.12. I don't have an earlier version to test it on. Maybe it was reintroduced in Sierra?

deckar01, 2017-09-11:
https://apple.stackexchange.com/questions/207895/app-gets-in-commandline-parameter-psn-0-nnnnnn-why

I'm not convinced by the discussion on that thread that it was actually removed in 10.9. There are some API changes related to the PSN, but nothing to indicate it was removed entirely.

Since OpenSCAD targets 10.8, then it doesn't surprise me that the PSN argument is present. I suspect it would still be present when targeting 10.9 and it never actually was removed.

closed 2017-09-12 by kintel

doug-moen, 2018-01-31:
I'm running MacOS 10.11.6 (El Capitan), and I have the same issue.

I installed the latest snapshot directly into my Applications directory, I started OpenSCAD from the Dock, and I get a security error, due to my security settings.

I don't want to change my security settings to let unknown applications run. Instead, my procedure is to control-click the OpenSCAD icon in the dock, and select "Open" from the menu. A dialog box asks me if I really want to run an unknown application, and I click yes.

I'm documenting this procedure here in case somebody else runs into the problem and looks up this issue.

deckar01, 2018-01-31:
@doug-moen That sounds like standard OSX behavior when the developer is unverified. This issue was for a bug that prevented the app from opening without starting it from the command line.

> To override your security settings and open the app anyway:
> 
> 1. In the Finder, locate the app you want to open.
> 2. Don’t use Launchpad to do this. Launchpad doesn’t allow you to access the shortcut menu.
> 3. Control-click the app icon, then choose Open from the shortcut menu.
> 4. Click Open.
> 
> The app is saved as an exception to your security settings, and you can open it in the future by double-clicking it just as you can any registered app.
> 
> Note: You can also grant an exception for a blocked app by clicking the “Open Anyway” button in the General pane of Security & Privacy preferences. This button is available for about an hour after you try to open the app.

https://support.apple.com/kb/PH25088

kintel, 2018-01-31:
@doug-moen FYI: This issue is about OpenSCAD not opening at all, even when given the permissions (due to the `-psn` argument being sent by Finder.

PR #2107 (2017-09-12): Fix MacOS First Run
Handle the `-psn_*` argument that Finder on OSX sends to apps.

> It looks like Finder appends a process ID argument.
> 
> https://stackoverflow.com/questions/10242115/os-x-strange-psn-command-line-parameter-when-launched-from-finder
> 
> FreeSCAD had the same problem and posted a workaround in this thread.
> 
> https://forum.freecadweb.org/viewtopic.php?t=4447

https://github.com/openscad/openscad/issues/2089#issuecomment-328371812

Fixes #2089.