openscad/openscad #2090: Minkowski of cube and polyhedron crashes OpenSCAD

Chais, 2017-08-19:
This relatively simple polyhedron reliably segfaults OpenSCAD 2015.03.03 on Arch.
```
points = [
	[0,0,0],
	[0.5,0,0.5],
	[0,0.5,0.5],
	[-0.5,0,0.5],
	[0,-0.5,0.5],
	[0,0,1]
];
faces = [
	[0,1,2],
	[0,2,3],
	[0,3,4],
	[0,4,1],
	[5,1,2],
	[5,2,3],
	[5,3,4],
	[5,4,1]
];

translate([0,0,3.5]) {
	minkowski() {
		cube([29,29,6], center=true);
		polyhedron(points,faces);
		*sphere(0.5);
	}
}
```
The polyhedron alone compiles and renders just fine, as does the minkowski of the cube with a sphere. I can provide a coredump if it's any use.

donbright, 2017-08-19:
it doesnt crash current github master but it does throw an error and produce null result

`Saved backup file: /home/don/.local/share/OpenSCAD/backups/unsaved-backup-SLh14633.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe != 0 File: /usr/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 212 
CGAL Cache insert: minkowski(convexity=0){cube(size=[29,29, (0 bytes)
CGAL Cache insert: multmatrix([[1,0,0,0],[0,1,0,0],[0,0,1,3 (0 bytes)
Geometries in cache: 2
Geometry cache size in bytes: 1456
CGAL Polyhedrons in cache: 2
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 0 seconds
Rendering finished.`

Chais, 2017-08-19:
That's somewhat better, but still doesn't produce the desired result. And I don't see any hint in the error that would indicate that the above code is actually incorrect.

gadgetmind, 2017-08-19:
On 19/08/17 13:36, Chais wrote:
> 	[5,1,2],
> 	[5,2,3],
> 	[5,3,4],
> 	[5,4,1]

These four faces are the wrong way around. Swap the last two points for each. The "thrown together" view is good for checking this.

I tested it with polyhedron and sphere with this fixed and it worked.

Of course, the seg fault isn't the best way of reporting this error. :-)

Chais, 2017-08-19:
Oh, so I *do* have to keep track of my surface normals?

gadgetmind, 2017-08-19:
Yes, the polyhedron module is very unforgiving. You need to get all the faces the right way around, avoid holes, and avoid non-planar faces. The only reliable way that I've found to achieve the latter is to divide everything into triangles and avoid any faces with more vertices than this. 

The docs say "All faces must have points ordered in the same direction . OpenSCAD prefers clockwise when looking at each face from outside inwards. The back is viewed from the back, the bottom from the bottom, etc.."

They also say "Points which describe a single face must all be on the same plane." but floating point inaccuracies mean that you can have them all in the same plane but rotations etc. mean that you get CGAL warnings/errors when rendering.

gadgetmind, 2017-08-19:
BTW, you may find a hull of six spheres to be a faster way of drawing that object, but this is perhaps just one step along the way to something more complex that you're making. (And hull isn't stunning fast either!)

Chais, 2017-08-19:
Sorry, I must have overlooked the paragraph about ordering. The spheres don't produce the kind of edge I want. I want a bevelled edge, not a rounded one.

gadgetmind, 2017-08-19:
You could do sphere ($fn=8) or similar, but if minkowski works for you, and is fast enough, that's great.

I had to read right to the end to find that paragraph, and was almost starting to think I'd imagined it. It maybe needs to be in the faces section rather than further down where it's easy to miss.

gadgetmind, 2017-08-19:
Interestingly, the section on Debugging polyhedra is exactly what I did, even though I don't recall ever reading it!

t-paul, 2017-08-19:
I'm not getting a crash with both nightly build (2017.08.16.nightly (git 7274bbd)) and 2015.03 release version on Debian. Maybe report that issue in the Arch bugtracker.

kintel, 2024-03-27:
This does actually crash OpenSCAD when using the Manifold backend

kintel, 2024-07-21:
Manifold crash is now fixed - the initial example works as expected with Manifold. CGAL still yields an empty result.