openscad/openscad #2091: [TODO] Wireframe Width

deckar01, 2017-08-19:
Add a preference for customizing the wireframe line thickness.

- [ ] [MVP] Use a 1px line thickness for wireframes
- [ ] [PR] Add a preference for setting the line thickness for wireframes

deckar01, 2017-08-19:
I would like to get a hard coded value of 1px working in my local build since I have been using wireframes heavily when checking if union operations had any precision problems. The chunky lines make it hard to spot errors without zooming in closely.

I would be open to finishing out the TODO with a preference so I can start getting my feet wet with this code base.

kintel, 2017-08-20:
1px may also be a better default value

deckar01, 2017-08-20:
There are comments already in `Polyhedron::draw()` for making the lines and nodes smaller. After comparing a high density display to an standard display I realized that there is not a one size fits all pixel value. The current default actually looks OK on a retina display, but on an old Air it is way too thick.

I added a 3rd item to the description for calculating the default using the display density.

deckar01, 2017-08-20:
I changed the line width to 1px and the point size to 3px, then multiplied them by `QScreen::devicePixelRatio()`.

Retina:

[image]

Air:

[image]

I think this looks good on x1 and x2 devices, and it appears to match the weight of the axis lines.

It looks like I should probably be using `QGLView::getDPI()` since it is behind a version flag, but I'm not quite sure how to use it in `OGL_helper.h` yet.

deckar01, 2017-08-20:
I tried passing the DPI from `GLView` down to `Polyhedron::draw()`, but got stuck at `polyset->render_*()`. I suspect `polyset` contains a collection of `Polyhedron`, but relevant implementation was external.

It was a lot simpler to just probe the `QScreen` in `OGL::Polyhedron::get_dpi()` than try to pass it through the stack anyway. Does this sound like an acceptable solution?

There are a few other places that are using hard coded line widths instead of DPI adjusted values. I might investigate those and create new issues for noticeably thick lines on standard displays.

MichaelAtOz, 2017-08-22:
Thanks, I always found the wireframe ugly.

deckar01, 2017-08-28:
@MichaelAtOz It looks like #2099 is lingering in review. Would you mind testing out the changes and see how it looks on your device(s)? If you need me to build any binaries let me know.

MichaelAtOz, 2017-08-28:
@deckar01 happy to. I'll need a windows binary. 
I'll have to check the DPI's on my monitors.
That last comment in #2099 'desired behavior for free' means command line output, yes?

deckar01, 2017-08-28:
The command line renderer is display independent, so it will start rendering the line width at 1 pixel regardless of the device. Unless you're running Windows 10 on a 4k display it is probably going to have a DPR of 1, so it should look a lot more detailed on the GUI for standard HD displays and the command line than it did before.

I will let you know when the binaries have been released on my fork.

t-paul, 2017-08-31:
http://files.openscad.org/snapshots/OpenSCAD-2017.08.31-x86-32_wireframe-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2017.08.31-x86-32_wireframe.zip
http://files.openscad.org/snapshots/OpenSCAD-2017.08.31-x86-64_wireframe-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2017.08.31-x86-64_wireframe.zip

deckar01, 2017-08-31:
Thanks @t-paul! I'll work up to a full build eventually. 😊

deckar01, 2017-08-31:
I don't see any settings for exporting wireframes from the command line and the source code seems to indicate that there isn't one.

MichaelAtOz, 2017-09-01:
Looks much better.
First test:
[image]

Are they meant to be dashed?

deckar01, 2017-09-01:
They are not supposed to be dashed. It looks like rendering the surface view with edges turned on causes the polygons to intersect with the lines. Since OpenSCAD doesn't (can't?) enable antialiasing it produces gaps when trying to render 0.5px lines. It would be possible to double the line width when edges are present to compensate. The simplest solution would be to just use a 2px line width and ignore the problem. 2px is still better than the 5px it is using now.

Side note: I really wish OpenSCAD had antialiasing. I tried enabling some of the commented lines of code, that hint at it, but never got it to work right.

nophead, 2017-09-01:
On my previous computer I think preview was anti-aliased but F6 wasn't. I can't remember what I did to turn it on.

MichaelAtOz, 2017-09-01:
I turned on anti-aliasing (via NVIDA control panel), you need to restart OpenSCAD.
[image]

I actually like the thinner dashed look, particularly with fonts which are vertex heavy.
Perhaps sticking with 1px, let users enable/disable via their settings.

The NVIDIA control can be set to 'let app decide' (somehow?) or override it.

However, I just turned all those options off (to my best recollection - there were not that many), and now it won't go back to the dashed lines...will try reboot...

p.s. Above two pics are on my 1080p monitor 96ppi, I have a new UWQHD monitor which I'm adjusting DPI settings for best eye comfort, I'll do some tests there next.

(also will report minor splash screen size issue when changing DPI)

MichaelAtOz, 2017-09-02:
<reinsert brain> of course the above is 'show edges' not wireframe....

MichaelAtOz, 2017-09-04:
So a little further testing.
I presume the same code does the 'show edges' in rendered 'view/surfaces'.
On my other PC I get the dashed lined.
I did the same NVIDIA control change, and it didn't affect the display, still dashed. ?

Having looked at actual wireframes now I think they are too thin.

I'm having browser issues, can't do images here...standby

MichaelAtOz, 2017-09-04:
[image]

For me I can't get a good feel for the solidity. It is even worse for complex objects

[image]

p.s. Reboot got dashed lined back BTW

[image]

MichaelAtOz, 2017-09-04:
Note, the viewability varies with the various 3D View Color Scheme Preferences, Cornfield is one of the worst.

deckar01, 2017-09-04:
2px might be a safer default. I sounds like it also might be necessary to add a setting for the line width, because use cases can vary.

closed 2026-03-01 by github-actions[bot]