openscad/openscad #2100: Accept HTML Color code

kuldeepdhaka, 2017-08-24:
Please accept \#RRGGBB (and \#RRGGBBAA will not be a big either if the non alpha is implemented)

People (including me) who have worked with the HTML color code will find it very handy.

I dont think this is supported because nothing mentioned in documentation
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#color

and if i try to use, result in
```
WARNING: Color name "#b5a642" unknown. Please see
WARNING: http://en.wikipedia.org/wiki/Web_colors
```

MichaelAtOz, 2017-08-25:
Thanks, I needed a challenge.
[rgb.txt](https://github.com/openscad/openscad/files/1250754/rgb.txt)

Corona688, 2017-09-18:
This is a terrific idea.

justbuchanan, 2017-10-22:
Closed as of #2134.

closed 2017-10-28 by t-paul