openscad/openscad #2102: Can you cut a 2D object into 2 children along a line?

highfellow, 2017-09-04:
I am looking for a way, given a 2D object and a cut line that is known to cut the object into two parts, to generate those two parts as 2D objects. I would like the object to be any 2D object, and the cut line to be specified as a vector. The only way I can see of doing it at the moment is to specify the cut as a region that is known to enclose one of the parts, but this is a bit clunky for what I want. Can anyone help?

(The application is a sheet material folding module I am working on: https://github.com/highfellow/openscad-sheet-folder )

highfellow, 2017-09-04:
Sorry I just realised this isn't the place for support requests.  I am reposting this on the forum and closing the issue.

closed 2017-09-04 by highfellow

sbts, 2017-09-04:
@highfellow yes, the forum is a better place to ask questions like that, alternatively visit the IRC channel
either
- #openscad @ irc.freenode.net
- [#openscad via Riot web client]](https://riot.im/app/#/room/#openscad:matrix.org)