openscad/openscad #2103: Errors when loading file in 2017 windows snapshots

runsun, 2017-09-07:
Got a new pc (windows 10, 64bit) and found some errors when loading files in different window snapshots. To locate when the errors introduced, I did some thorough checks. Here is a report. 

There are two errors found: 

ERROR-1: 

A popup window says: 

``Popup up Critical Error: A critical error was caught. The application may have become unstable: 
 vector:: _MN_range_check: __n(which is 0) >= this -> size() (which is 0)
``

This is lethal. After it, every file, even a file containing as simple as cube([1,2,3]); failed to draw. 

It seems to be happening on more complicated files (like files that import large files) but I am not sure. 

This error was introduced since OpenSCAD-2016.11.11-x86-64-Installer.exe 

ERROR-2: 

An error message about "Parameter Set" shows up in pink in the console for a sec then disappears. Supposed we are loading "c:/myOpenScad/file1.scad", the error message is : 

``
ERROR: Cannot open Parameter Set:'c:/myOpenScad/file1.json': c:/myOpenScad/file1.json: cannot open file 
Loaded design: c:/myOpenScad/file1.scad
``

That is, when loading whatever.scad, it complains it can't find whatever.json. 

This appears to be a non-lethal error and everything proceeds as usual. 

This error was introduced since OpenSCAD-2017.01.20-x86-64-Installer.exe 

Here are some snapshots I tested: 

2015.03-2 (This ver doesn’t have “each” ): OK 
OpenSCAD-2015.10.17-x86-64-Installer.exe : OK 
OpenSCAD-2016.01.26-x86-64-Installer.exe : OK 
OpenSCAD-2016.02.01-x86-64-Installer.exe : OK 
OpenSCAD-2016.02.09-x86-64-Installer.exe : OK 
OpenSCAD-2016.04.06-x86-64-Installer.exe : OK 
OpenSCAD-2016.05.12-x86-64-Installer.exe : OK 
OpenSCAD-2016.07.20-x86-64-Installer.exe : OK 
OpenSCAD-2016.09.18-x86-64-Installer.exe : OK 
OpenSCAD-2016.10.04-x86-64-Installer.exe : OK 
OpenSCAD-2016.11.11-x86-64-Installer.exe : ERROR-1 
OpenSCAD-2017.01.20-x86-64-Installer.exe : ERROR-1 + ERROR-2 
OpenSCAD-2017.08.30-x86-64-Installer.exe : ERROR-1 + ERROR-2

t-paul, 2017-09-09:
Can you please create separate issues for different errors?

t-paul, 2017-09-09:
@kintel I guess that would need to go into the "next release" bucket?

runsun, 2017-09-12:
As you guys would expect, no more warning after turning off the Customizer

MichaelPFrey, 2018-02-24:
isn´t this issue a duplicate of #1927?

kintel, 2018-02-24:
Closing as duplicate of #1927

closed 2018-02-24 by kintel