openscad/openscad #2104: Can't pass special vars to children

thehans, 2017-09-09:
If you try to set special vars in a call to children, the parameter is interpreted as passing an index to children.  It would be nice if children discriminated that a named variable beginning with '$' is not meant to represent an index variable.

Here is a short scad that hopefully demonstrates the issue and one workaround I'm currently using.
```
module foo_broken(x) {
  children($special=x);
}

module foo_workaround(x) {
  noop($special=x) children();
}

module noop() { children(); }

module bar() { echo($special); }

$special = 0;
bar();
foo_workaround(1) bar();
foo_broken(2) bar(); 
```
this outputs:
```
ECHO: 0
ECHO: 1
WARNING: Children index (2) out of bounds (1 children) 
```

sbts, 2017-09-09:
Hi
Looking at (the Children section](https://en.m.wikibooks.org/wiki/OpenSCAD_User_Manual/User-Defined_Functions_and_Modules#children) of the manual I don't understand what you are hoping to achieve here.
`children ()` takes a single argument that specifies the index to a specific child (or range of children)
By definition the argument resolves to an index to an array.
Parameters can't be passes to each child via the `children ()` function 

The wrapper you describe seems rational to me. It could also be written as....

```
module foo(x) {
  $special=x;
  children();
}
```
Which keeps the change to `$special` contained in the same scope as the call to `children ()`

nophead, 2017-09-09:
> wikibooks.org/wiki/OpenSCAD_User_Manual/User-Defined_
> Functions_and_Modules#children) of the manual I don't understand what you

Or you can just use for($special = x) children();

t-paul, 2017-09-09:
I don't think that is even possible, as that would produce 2 different evaluation levels for the same sub-tree (the one represented by `children()`)?
```
$special = 3; children($special = 5); // which value would be effective?
```

nophead, 2017-09-09:
$fn = 3; circle($fn = 5);

Answer 5.

t-paul, 2017-09-09:
Looking at the code, the difference seems that `children()` does not introduce a new context for evaluation. Adding that might make this case work. I don't know the context stuff well enough to judge if that would break other cases though.

thehans, 2017-09-09:
@nophead your `for` loop trick is a nice workaround that did not occur to me to apply for this; I prefer that over my inital `noop` workaround.

@sbts I try to give simplified examples, but I may have oversimplified in this case.  The reason I don't set it like that is because I also care about the value of `$special` from within `foo`, I just want it to be able to apply a different value to `$special` for its children.

So for example add a `echo` statement to `foo`, I want `foo` to see the `0` value that was set from the outermost scope.  If you set it right there, then you overwrite and lose that information:
```
$special = 0;

module foo_nophead(x) {
  echo($special);
  for($special=x) children();
}

module foo_sbts(x) {
  $special=x;
  echo($special);
  children();
}

foo_nophead(1);
foo_sbts(2);
```

nophead, 2017-09-09:
I use ```for``` because ```assign``` is deprecated but it does still work with a warning. If you enable experimental features then you can use ```let```. I think both will work with vectors.

MichaelAtOz, 2017-09-09:
```
union(a=5,$b=6) // I would not expect a or b to be in scope of children of union()
  echo(a=a,b=b);  // undef, undef
```

thehans, 2017-09-17:
I feel this was not given full consideration.

@MichaelAtOz  Do you expect user-defined modules to treat special variables differently than builtin modules?  Because that's what's happenening.

The fact that it also doesn't work for union should be its own issue.

```
module user_defined() {
  cube();
  children();
}

user_defined(a=5, $b=6)
  echo(a=a,$b=$b);       // ECHO: a = undef, $b = 6
```

MichaelAtOz, 2017-09-17:
I was doing a few more examples and came across something.
I'll have a more detailed look.
Reopening.

thehans, 2018-02-05:
Since I just shared the same code on the mailing list, I'm adding the full use-case which prompted me to open this issue in the first place.

In the code below I use `noop` so that children can have a different value in special variables than their parent.  The commented out line below `noop` shows how I was expecting to be able to pass the special variables directly to `children`

```
// Pipe library demonstrating a concept of chained operations

$fn = 72; // define exact number of fragments in curves
// using $fs instead would cause some mismatched # of edges during tapers
echo(str("For best results, all angles should be multiples of ", 360/$fn, " degrees"));

// Example of how to use pipe
// Note that all calls are chained (there's only one semicolon, at the end)
// children are automatically transformed to the end of the "path"
pipe([4,6]) 
  elbow() 
  straight(10) 
  taper_to([8,10], length=10) 
  elbow(angle=180, heading=30) 
  straight(20)
  sphere(d=6); // we can even terminate the chain with a module that doesn't "know" anything about our pipes

// initializes the pipe size, but draws nothing
module pipe(size=[2,4]) {
  $pipe_size = size;
  children(); 
}

module straight(length=10) {
  linear_extrude(height=length) difference() {
    circle(d=$pipe_size[1]);
    circle(d=$pipe_size[0]);
  }
  translate([0,0,length]) children(); 
}

// no-operation.  This aids in passing data from parent to children
// cals to noop() could also be replaced by let() (experimental), if running a development snapshot.  
module noop() children();

module taper_to(size, length=10) {
  IR1 = $pipe_size[0]/2;
  OR1 = $pipe_size[1]/2;
  IR2 = size[0]/2;
  OR2 = size[1]/2;
  points = [
    [IR1, 0], [OR1, 0],
    [OR2, length], [IR2, length] 
  ];

  rotate_extrude() polygon(points);

  translate([0,0,length]) 
    noop($pipe_size=size) children(); 
   // children($pipe_size=size);  // I had originally expected this to work, in lieu of noop, or let
}

// R = center radius of elbow
// angle = the angle through which the elbow bends.  
// heading = which direction it bend towards
module elbow(R=$pipe_size[1]/2, angle=90, heading=0) {
  OR = $pipe_size[1]/2;

  if (angle != 0) {
    rotate(heading) translate([R+OR,0]) rotate([-90,0,0]) {
      rotate_extrude(angle=angle)
      translate([-R-OR,0]) difference() {
        circle(d=$pipe_size[1]);
        circle(d=$pipe_size[0]);
      }
    }
  }
  rotate(heading) translate([R+OR,0,0]) rotate([0,angle,0]) translate([-R-OR,0,0]) children();
}
```

revarbat, 2019-03-13:
This may be helpful to illuminate behavior.  Perhaps assign() shouldn't be deprecated?

    module copier(n,s) for ($i=[0:n]) translate([$i*s,0,0]) children();
    copier(5,20) sphere(d=1+$i);   // Works!  Each successive sphere is bigger.
    copier(5,20) {diam = 1+$i; sphere(d=diam);}  // Fails!  diam=undef
    copier(5,20) assign(diam=1+$i) sphere(d=diam);  // Works!  But deprecated!
    copier(5,20) for(diam=[1+$i]) sphere(d=diam);  // Works!  But ugly.

nophead, 2019-03-13:
Yes I have noticed that failure. I use let() to get around it.

revarbat, 2019-03-13:
`let()` just makes me twitchy regarding backwards compatibility of my code, since it's not in the latest *official* non-dev release from 2015.

jordanbrown0, 2020-03-30:
I noticed this too.  I don't know that I have a use case, but it's tantalizing.  It's almost, but not quite, like being able to put a reference to a module in a variable.  It does seem pretty wrong that a $ variable isn't visible to assignments in the body.  They *are* visible when used as parameters to module invocations, and to any modules invoked.

PR #3761 (2021-05-15): Use name-based argument parsing in children(), normalizing special variable passing.
Use name-based argument parsing in children(), normalizing special variable passing.

Fixes #2104.
commit 131e25d3 (2021-05-15): Use name-based argument parsing in children(), normalizing special variable passing.

Fixes #2104.
closed 2021-05-15 by t-paul