openscad/openscad #2105: File loading error in 2017 windows snapshots

runsun, 2017-09-09:
Errors when loading files in different window snapshots (including all 2017 versions). 
(Windows 10 standard edition, 64 bit)

Upon loading, a popup window says:

Popup up Critical Error: A critical error was caught. The application may have become unstable: vector:: _MN_range_check: __n(which is 0) >= this -> size() (which is 0)

This is lethal. After it, every file, even a file containing as simple as cube([1,2,3]); failed to draw.

It seems to be happening on more complicated files (like files that import large files) but I am not sure.

This error was introduced since OpenSCAD-2016.11.11-x86-64-Installer.exe

Snapshots tested:

2015.03-2 (lastest release): OK
OpenSCAD-2015.10.17-x86-64-Installer.exe : OK
OpenSCAD-2016.01.26-x86-64-Installer.exe : OK
OpenSCAD-2016.02.01-x86-64-Installer.exe : OK
OpenSCAD-2016.02.09-x86-64-Installer.exe : OK
OpenSCAD-2016.04.06-x86-64-Installer.exe : OK
OpenSCAD-2016.05.12-x86-64-Installer.exe : OK
OpenSCAD-2016.07.20-x86-64-Installer.exe : OK
OpenSCAD-2016.09.18-x86-64-Installer.exe : OK
OpenSCAD-2016.10.04-x86-64-Installer.exe : OK
OpenSCAD-2016.11.11-x86-64-Installer.exe : ERROR
OpenSCAD-2017.01.20-x86-64-Installer.exe : ERROR 
OpenSCAD-2017.08.30-x86-64-Installer.exe : ERROR

t-paul, 2017-09-09:
Hmm, that might be related to the Customizer. Can you reproduce that with some specific file? If yes, does it happen with the Customizer disabled (in Preferences) and could you share one of those files? So far I could not reproduce the issue.

runsun, 2017-09-11:
1) The files tested is a lib that imports more than 10 files and total having more than 21,000 lines. By itself, it runs ok.  The error appeared when it is imported by another file that is 2000 lines by itself. So I am suspecting file size problem. 

WIll try to make files of minimal size that still produced the error.
 
2) No error if Customizer turned off.

t-paul, 2017-09-11:
So maybe it's worth having a go with the full big bundle of files, if those can be shared somehow. For geometry it's often extremely helpful to minimize the data to search in. In this case there's a chance, it's not so important.

runsun, 2017-09-12:
Hi boss, I turned my private repo into public: https://bitbucket.org/runsun/scadx . Note that it still has a lot of junks that I haven't got time to clean up. 

Loading the main file, `scadx.scad`, is ok, but loading `scadx_make.scad`, which is a file including `scadx.scad`, produces the error.  

`scadx_make` is an attempt to use OpenSCAD in an OOP way, with that each object is represented by a set of data. Still work in progress but you can try the demos in the file.

t-paul, 2017-09-12:
It's not visible in the list of public repositories.

runsun, 2017-09-12:
sry forgot to save the setting. Should be ok now.

t-paul, 2017-09-12:
Yep, thanks, works now. Bug in Customizer is caused by:
```
Dim=[];
```

MichaelPFrey, 2018-02-25:
@runsun would you mind to test with the newest experimental builds on windows and update the issue?

I downloaded your project and tested it with freshly build OpenSCAD under Ubuntu 17.10, making me pretty certain that my fix for the empty vector has resolved your issue.
The current experimental build for Windows should also include the patch.

kintel, 2018-11-04:
Assuming this has been resolved

closed 2018-11-04 by kintel