openscad/openscad #2106: Renderer croaks "WARNING: No top level geometry to render" though preview works

thomaskilian, 2017-09-10:
I'm using the following code snippet to render a plate with holes. Running in preview it's fine, but when rendering it throws the above error message.

    w = 126;
    h = 60;
    module Side() {
      module Holes() {
        sin60 = sin(60);
        holeDist = floor(w/10);
        dw = w / (holeDist+1);
        dw2 = dw/2;
        dw3 = dw/3*2;
        yDist2 = sin60*dw;
        yDist = yDist2*2;
        nh = floor(h/yDist);
        y0 = (h - (nh-1)*yDist)/2;
        for (y = [0:nh-1]) 
          for (i = [1:holeDist-2]) 
            translate ([i*dw+dw2, y*yDist+y0, 0]) %cylinder(d=dw3, h=shellThickness);
        
        for (y = [0:nh-2]) 
          for (i = [1:holeDist-3]) 
            translate ([i*dw+dw2+dw/2, y*yDist+y0+yDist2, 0]) %cylinder(d=dw3, h=shellThickness);
      }
    //  difference() {
    //    cube([d, h, shellThickness]);
        Holes();
        
    //    cube();

    //  }
    }
    Side();

t-paul, 2017-09-10:
You are using the `%` modifier which specifically omits those parts from the render.

See https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Modifier_Characters#Background_Modifier

closed 2017-09-10 by t-paul

thomaskilian, 2017-09-10:
Shame on me :-(