openscad/openscad #2108: Minkowski on non-convex objects produces geometry errors

a-ludi, 2017-09-11:
I ran across some error with minkowski with non-convex objects as can be seen in this [example](https://gist.github.com/a-ludi/76632913dbd221769f9eb8f9ee749edc). The error is produced by a difference of some object with a minkoski sum of multiple unified convex objects and a single convex object. In some cases I was able to observe errors even without the difference operation.

I am working with commit 959b9c4b712ed5db6938fc6ebdf2c1cb58df6c4f.

kintel, 2017-10-10:
Closing as known behavior

closed 2017-10-10 by kintel

a-ludi, 2017-10-10:
@kintel Is there another issue on this topic where I can track the status?

nophead, 2017-10-10:
It isn't an issue with OpenSCAD, it is a problem in your example and I posted the solution there.