openscad/openscad #2110: Crash on OSX 10.11.6

robertrau, 2017-09-13:
I was using OpenSCAD and it was no longer showing most of the menu items, so I closed my window, quitting. I then restarted, re-opened my part, and started to rotate it in the preview window and crashed.

Thank you for all the hard work on OpenSCAD!

Crash log:

Process:               OpenSCAD [25156]
Path:                  /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:            org.openscad.OpenSCAD
Version:               2015.03-3 (2016.02.23)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OpenSCAD [25156]
User ID:               501

Date/Time:             2017-09-12 20:45:55.094 -0400
OS Version:            Mac OS X 10.11.6 (15G1611)
Report Version:        11
Anonymous UUID:        2C32644B-B144-0B71-1440-D9EC2C05D916

Sleep/Wake UUID:       CBD04A7A-AF19-4DF5-AB8C-6D34E9FC66CB

Time Awake Since Boot: 680000 seconds
Time Since Wake:       4800 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000010000006a
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x10000006a:
--> 
    __TEXT                 0000000104598000-0000000104b4c000 [ 5840K] r-x/rwx SM=COW  /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation      	0x00007fff8b499822 CFPasteboardGetGenerationCount + 66
1   com.apple.HIServices          	0x00007fff8b08f679 IndexToItem(OpaquePasteboardRef*, long) + 30
2   com.apple.HIServices          	0x00007fff8b0b24a0 PasteboardCopyDataProc(void*, __CFPasteboard*, long, long, __CFString const*) + 190
3   com.apple.CoreFoundation      	0x00007fff8b50db1b __CFPasteboardResolvePromiseForItemLocal + 539
4   com.apple.CoreFoundation      	0x00007fff8b50d725 CFPasteboardResolveAllPromisedData + 757
5   com.apple.CoreFoundation      	0x00007fff8b492b0c __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
6   com.apple.CoreFoundation      	0x00007fff8b492a9f ___CFXRegistrationPost_block_invoke + 63
7   com.apple.CoreFoundation      	0x00007fff8b492a17 _CFXRegistrationPost + 407
8   com.apple.CoreFoundation      	0x00007fff8b492782 ___CFXNotificationPost_block_invoke + 50
9   com.apple.CoreFoundation      	0x00007fff8b44f592 -[_CFXNotificationRegistrar find:object:observer:enumerator:] + 1922
10  com.apple.CoreFoundation      	0x00007fff8b44e7e5 _CFXNotificationPost + 693
11  com.apple.Foundation          	0x00007fff977e8f9a -[NSNotificationCenter postNotificationName:object:userInfo:] + 66
12  org.openscad.OpenSCAD         	0x00000001047453ec MainWindow::quit() + 76
13  org.openscad.OpenSCAD         	0x0000000104744c6a MainWindow::~MainWindow() + 362
14  org.openscad.OpenSCAD         	0x000000010474544f MainWindow::~MainWindow() + 15
15  org.qt-project.QtCore         	0x00000001067090bf QObject::event(QEvent*) + 815
16  org.qt-project.QtWidgets      	0x00000001059079ae QWidget::event(QEvent*) + 4110
17  org.qt-project.QtWidgets      	0x0000000105a0dd6d QMainWindow::event(QEvent*) + 1725
18  org.qt-project.QtWidgets      	0x00000001058c885b QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
19  org.qt-project.QtWidgets      	0x00000001058cb1be QApplication::notify(QObject*, QEvent*) + 5630
20  org.qt-project.QtCore         	0x00000001066dd840 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 976
21  org.qt-project.QtCore         	0x00000001066dd14c QCoreApplication::exec() + 428
22  org.openscad.OpenSCAD         	0x000000010472c490 gui(std::vector<std::string, std::allocator<std::string> >&, boost::filesystem::path const&, int, char**) + 3040
23  org.openscad.OpenSCAD         	0x000000010472f77f main + 10127
24  org.openscad.OpenSCAD         	0x00000001045a3564 start + 52

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff95e14efa kevent_qos + 10
1   libdispatch.dylib             	0x00007fff9a17d165 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff9a17cdcd _dispatch_mgr_thread + 52

Thread 2:: com.apple.NSEventThread
[... 387 lines cut ...]
shared memory                     16.4M       11 
===========                     =======  ======= 
TOTAL                              1.1G      868 

Model: MacBookPro6,1, BootROM MBP61.0057.B17, 2 processors, Intel Core i7, 2.66 GHz, 8 GB, SMC 1.57f18
Graphics: NVIDIA GeForce GT 330M, NVIDIA GeForce GT 330M, PCIe, 512 MB
Graphics: Intel HD Graphics, Intel HD Graphics, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1067 MHz, 0x04CD, 0x46332D3130363636434C392D344742535100
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1067 MHz, 0x04CD, 0x46332D3130363636434C392D344742535100
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x93), Broadcom BCM43xx 1.0 (5.106.98.100.24)
Bluetooth: Version 4.4.6f1 17910, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi 3, AirPort, en1
Serial ATA Device: ST2000LM003 HN-M201RAD, 2 TB
Serial ATA Device: HL-DT-ST DVDRW  GS23N
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: IR Receiver
USB Device: Built-in iSight
Thunderbolt Bus:

robertrau, 2017-09-13:
> From the log it shows that OpenSCAD crashes while trying to quit, but you wrote that it crashed mid-rotate.
> Some questions:
>
> Can you reproduce this behavior?
> Does it depend on what file you're showing?

Marius,

Is is possible that some processes of OpenSCAD had not yet quit from before my re-launch? I have seen that behavior on Windows where some applications are actually many processes. That is my only guess as to why the crash log would indicate I was quitting.

I have only been using OpenSCAD since Friday, this is the only crash I have had.

I have been designing an enclosure for a Raspberry Pi HAT board, attached, so I haven’t tried other files yet. I will work with some examples and see if I have other troubles.

Thank you for such a quick reply.

Bob Rau
www.rau-deaver.org

kintel, 2017-09-13:
That's possible; we have a bug which causes occasional crash on exit (#1251)

MichaelAtOz, 2017-09-13:
(On Windows) When I deal with complex models with a resulting large memory footprint, then exit OpenSCAD, the GUI disappears practically instantly, yet I have observed the .exe in Task Manager gradually reducing its working set, then exiting. I have seen this take several minutes with a 12GB working set. With trivial programs it is almost instant.
Presumably as destructors are unwound.
I wonder if there is a dirtier way...

MichaelAtOz, 2017-09-13:
p.s. Tho that has not prevented a subsequent OpenSCAD from starting while the other was still ending.

kintel, 2017-09-17:
@MichaelAtOz Yep, that sounds like #1251 - we don't have a safe way to kill a CGAL thread so we currently don't try.

kintel, 2017-09-17:
@robertrau If your symptoms are in-line with #1251, we should probably close this as a duplicate.

MichaelAtOz, 2017-09-17:
@kintel those times was after render had finished, likely after an export too.
If CGAL is running and is less responsive to the cancel button I use Task Manager to End the Process, practically instant.

kintel, 2017-09-17:
@MichaelAtOz In that case it's probably destructors, and unrelated to this issue.

t-paul, 2019-02-19:
Closing, assuming duplicate as discussed above.

closed 2019-02-19 by t-paul