openscad/openscad #2112: Silent installer for Windows?

ehmatthes, 2017-09-16:
Hi, I'm trying to install OpenSCAD on a set of laptops for a high school class I'm teaching. I was hoping to find an installer that I could run remotely, but all I'm finding is the .exe that has a few prompts when you run it.

Has anyone made an .msi installer out of the .exe? Is there a way to install OpenSCAD silently on Windows?

t-paul, 2017-09-16:
The *.exe installer is created by NSIS which should support:
```
/S - silent install
/D - set install folder

OpenSCAD-2017.08.30-x86-64-Installer.exe /S /D "C:\Program Files\OpenSCAD"
```
(Not sure about the quotes needed if the install folder has spaces, I can only try on Linux via Wine)

donbright, 2017-09-18:
curious what kind of hardware and software is your school running?  Does it not throw the "this code might be dangerous" warning if you try to run unsigned code? ( in other words, there is no way for me to test this on my copy of Windows 10 because it will throw up several warnings every time i install something thus mooting the point )

MichaelAtOz, 2017-09-19:
(I run with User Account Control OFF - ie running with scissors...)

ehmatthes, 2017-09-19:
@t-paul, that's really helpful. I had tried /silent but didn't know to use /S. I was able to do the installation automatically today, which was much easier than the one-by-one installations I've led students through in the past. I don't mind doing this with students because I like them to know how to install software, but it's also really good to push software out to all of our computers most of the time. Once people learn to use OpenSCAD, I want them to know it's already on most or all of our computers.

@donbright, we're running Windows 10 on a set of Lenovo Yoga 12 computers. The district manages the base image, but I can install software as needed. I don't get that warning; I think our district has decided to trust teachers with the ability to install software at their discretion.

Thanks everyone, I love using OpenSCAD with students and this install process makes it much easier to jump into modeling with them!

closed 2017-09-19 by ehmatthes