openscad/openscad #2113: New Feature: 'power of' syntax x^y, e.g x^2 etc.

MichaelAtOz, 2017-09-17:
I'm starting to need non trivial maths in some geometry calculations.
Have to do pow(x,y) is not as intuitive as x^y when transcribing equations such as
[image]

runsun, 2017-09-17:
I support this idea.

Corona688, 2017-09-18:
I wonder if it'd be easier to support the perl-style ** syntax.

doug-moen, 2017-09-18:
I guess that's a personal preference, depending on what languages you are
familiar with.

I'd prefer x^y, since all of the popular math and linear algebra languages
use this syntax. It's also what you use in Latex, or if you are editing
math equations on Wikipedia or in a math.stackexchange.com forum post. And
I do consider OpenSCAD to be a math language.

On the other hand, perl, python and ruby use x**y.

The R programming language supports both syntaxes: that's a potential
compromise in case this turns out to be a contentious issue.

kintel, 2017-10-15:
Challenge is that if you come from C/C++, `^` means XOR. Introducing this operator may actually make code _less_ readable at first glance..

nophead, 2017-10-15:
Can it be ** like Python instead.

thehans, 2017-10-15:
I implemented the operator as '^' for a few reasons.  First, since that was the original request, and second because I agree with @doug-moen that ^ is typical for general math syntax.  I also view OpenSCAD as more of math language than a C-family language.

Lastly this was my first time messing with yacc so I am still not sure if there's any special handling that would need to be done to differentiate between multiplication '*' and exponent '**' tokens.

MichaelAtOz, 2017-10-15:
I usually do my maths research in wikipedia initially. When you come across a relevant equasion you can edit that section and copy the maths bit, which as others have said uses `^`.
e.g. [Torus](https://en.wikipedia.org/wiki/Torus) 
[image]

`<math> f(x,y,z) = \left(R - \sqrt{x^2 + y^2}\right)^2 + z^2 - r^2.</math>`
So `^` would be really handy.

MichaelAtOz, 2018-04-28:
Added $25 bounty, for '^'.

MichaelAtOz, 2018-05-06:
> I implemented the operator as '^' for a few reasons. 

@thehans Did you 'implement' this in OpenSCAD or are you talking elsewhere?

thehans, 2018-05-06:
@MichaelAtOz I implemented it in PR #2125 but it could probably still use a test case.

seam345, 2019-10-25:
@thehans I have merged your update to the latest master and have added a test case for the '^' operator in PR #3104

closed 2019-11-24 by t-paul

t-paul, 2019-11-24:
@seam345 with this issue closed, the bounty should be claimable at [bounty source](https://www.bountysource.com/issues/49417555-new-feature-power-of-syntax-x-y-e-g-x-2-etc?utm_campaign=plugin&utm_content=tracker%2F52063&utm_medium=issues&utm_source=github).

commit f72dddad (2019-11-24): Merge pull request #3104 from seam345/exponentTest

New Feature: 'power of' syntax x^y, e.g x^2 (closes #2113).