openscad/openscad #2114: Excessive rendering time/memory

yschaeff, 2017-09-18:
I created the following Scad file where the preview works fine (in a couple of seconds) but the render takes a lot of time and memory. (more than 1Hr / 8 GB roughly) reported with OpenSCAD 2016.04.24 and 2015.03-2.

see also https://www.reddit.com/r/Ask3D/comments/6xkpud/rendermything_openscad/

[spheres.zip](https://github.com/openscad/openscad/files/1311031/spheres.zip)

Corona688, 2017-09-18:
The preview doesn't **compute** it, just **draws** it.  640,000 polygons is no sweat for any decent video card.  It doesn't have to worry about subtracting the inside parts from each other, it just lets the Z-buffer sort that out.

Exporting the STL, on the other hand, means locating and removing intersections between these 400+ spheres of 3200 polygons each.  That's 160,000 comparisons just to find out **which** objects are intersecting each other -- which turns out to be almost all of them.

You've also imposed things like $fn=40 which makes things very hard for OpenSCAD.  You had to do that because your spheres are all very tiny.  If you make them 10 or 100 times bigger, OpenSCAD will generate a sane number of fragments for them.

Ways you could make this faster:

* Multiply all the values your code generator creates by 10 or 100 to get your spheres above the minimum fragment size.  It will then be able to decide the number of fragments in an intelligent way without looking dumb.
* Shrink the smaller spheres so fewer of them touch each other.  The fewer intersections, the faster it is to calculate.

Corona688, 2017-09-18:
Actually, $fs=0.1 looks pretty good without modifying your model at all.  Setting fragment **size** is usually smarter than forcing $fn - that makes most of your tiny spheres ludicriously detailed.

It's still taking a while, but is using less than a hundred megs right now, not eight gigs.

nophead, 2017-09-18:
On my system it also took an hour but only seemed to use about 4GB.

MichaelAtOz, 2017-09-18:
@nophead  32/64bit OS & OpenSCAD?

nophead, 2017-09-18:
Win 7 64 bit. OpenSCAD 2017.01.20.

Corona688, 2017-09-19:
How do you intend to print this model?  Most spheres rest on each other unsupported, and the level of fine detail is implausible (unless your printer is SLA).  I suggest bumping the size of the spheres up a few percent so they **all** intersect with each other, if you want a functional print.

Anyway, I couldn't compute the model on $fs=0.1 yesterday, my anemic work computer cranked for 5 hours then out-of-mem-crashed at 1.3 gigs.  I'm trying with less detail today.

* No forced FS/FN:  3 minutes, 23 seconds.  Model very chunky.
* $fs=1:  5 minutes 23 seconds, The larger spheres look nicer, the little ones are still very chunky.
* $fs=0.75:  11 minutes, 4 seconds.  Memory growth approaches that of Firefox.  Much better-looking model.  Everything but the tiniest spheres subdivide reasonably.
* $fs=0.5:  Oh boy.  500 megs and counting.  Wish me luck.

nophead, 2017-09-19:
You should set $fa as well as $fs. This is what I use for 3D printing:

$fa = 6;
$fs = filament_width / 2;

yschaeff, 2017-09-20:
Thanks for the explanation. So the gist is that my model just sucks. Fair enough I agree with you there. I did actually print the model, with supports so that it touched the build plate. And it printed fine. https://i.imgur.com/P389CF2.jpg 
It was more an experiment than anything else really. The reddit thread seemed to indicate something more than just a stupendous model, but I totally accept that verdict. I'll read up on $f* and maybe someday improve my algorithm generating this.

yschaeff, 2017-09-20:
Fun fact: It was an experiment regarding infill. Filling a model (or rather gutting) with spheres to gain maximum strength with minimum infill. In theory you can reach an arbitrary sparse infill by just subtracting smaller and smaller spheres. In practice of course this is stupid and doesn't work as you can't print infinitely thin. But hey, had fun.

Corona688, 2017-09-20:
Ah, interesting.

If you remove the most tiniest spheres, the printed model won't change much at all and the generation time will be much reduced.

If your algorithm is recursive (and I bet it is) you may be able to implement it in OpenSCAD itself.

yschaeff, 2017-09-21:
Well not entirely recursive. In 2D you can do this. A circle inscribed in three circles partitions that space in three. But in 3D, inscribing 4 spheres with a sphere, this partitioning doesn't happen, it stays one (irregular shaped) space. Spheres from different branches will at some point overlap. So the recursive approach doesn't work all that well. I choose python to generate the OpenSCAD code and tried to eliminate these overlaps as much as possible.

Also, in 3D the math of finding an inscribed sphere is surprisingly hard and I couldn't do it. I used an iterative approximation to find the locations. I wouldn't know whether this is possible in OpenSCAD at all?

kintel, 2017-09-30:
Closing as discussion around known limitations. Please use the mailing list for such discussions.

closed 2017-09-30 by kintel