openscad/openscad #2118: Crash when rendering an imported stl and mix with primitives

FVivolo, 2017-10-02:
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /opt/mxe/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 

please let eme know to send you an example

t-paul, 2017-10-03:
Please check the [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_imported_STL_file_only_showing_up_with_F5_but_not_F6.3F) which has some tips.

Does "crash" mean the application actually crashes, or is it giving the error message above in the output window.

FVivolo, 2017-10-03:
sorry the application not crash this is a log in the console side when the render is requested

kintel, 2017-10-03:
We have a number of related issues, so this is most likely a duplicate, just don't know exactly which one yet: https://github.com/openscad/openscad/issues?q=is%3Aissue+is%3Aopen+label%3A%22Mesh+Repair%22

kintel, 2024-03-27:
Closing as duplicate of "some known existing issue". Also: Manifold is expected to behave better in the vast majority of cases.

closed 2024-03-27 by kintel