openscad/openscad #2120: QXcbConnection: XCB error, Could not initialize GLX

ProgramItUp, 2017-10-06:
Any thoughts on what is going wrong here?  Debugging suggestions welcomed.  Is this a bug or an OS configuration problem?

After installing OpenSCAD from the ppa it is unable to run.  While trying to start OpenSCAD on an Ubuntu 14.04 system I'm getting the following error:

```
sudo add-apt-repository -y ppa:openscad/releases
$ sudo apt-get update
$ sudo apt-get install openscad

$ openscad -v
OpenSCAD version 2015.03-1

$ openscad
Could not initialize localization.
Error reading examples.json: examples.json: cannot open file
QXcbConnection: XCB error: 145 (Unknown), sequence: 158, resource id: 0, major code: 139 (Unknown), minor code: 20
Could not initialize GLX
Aborted
```

kintel, 2017-10-10:
Are you running OpenSCAD locally on a native Ubuntu box, or are you attempting something fancy, like using a remote display?

ProgramItUp, 2017-10-10:
Remote display.  My localhost is a Windows 10 machine running MobaXterm to connect to the Ubuntu machine.

kintel, 2017-10-10:
In that case this looks like a known Qt bug: https://bugreports.qt.io/browse/QTBUG-31389 - should be fixed in Qt 5.5.

ProgramItUp, 2017-10-10:
Thanks. Do you happen to know the timing on the fix?

doug-moen, 2017-10-10:
QT 5.5 was released July 2015. You are running Ubuntu 14.04, which has Qt
5.2.1. Upgrading to Ubuntu 16.04, which has Qt 5.5.1, is one option for you.

t-paul, 2018-04-29:
Closing as the issue seems to be outside of OpenSCAD and no feedback for quite some time.

closed 2018-04-29 by t-paul