openscad/openscad #2121: Is your web secure? Every time I click over word "register" I got a spam popup of super resume!!

juanccilleruelo, 2017-10-11:
I want to subscribe to the forums on your web page.
I go to the first entry of the forum "Registration & Subscription Info....."
I read it and I click on the link in the word "register" I sometimes, I got a popup window, with a little size that shows me "super resume" and... a very strange behaviour.

The link in this word "resume" is http://forum.openscad.org/template/NamlServlet.jtp?macro=start_registration_page&nextUrl=http%3A%2F%2Fforum.openscad.org%2F

This happends to me at least two or three times. When I'm writing this issue, I check it and not.

It's secure the process of subscription at this moment?

t-paul, 2017-10-11:
This is hosted by [nabble](http://www.nabble.com/), we can't do much about that.

See also http://forum.openscad.org/Forum-login-not-protected-by-https-td22121.html

closed 2017-10-11 by t-paul

t-paul, 2017-10-11:
Also if you don't like registering there, it should be possible to just subscribe to the [mailing list](http://www.openscad.org/community.html#mailing-list). That should cover most of the posts.

MichaelAtOz, 2017-10-11:
>  I sometimes, I got a popup window, with a little size that shows me "super resume" and... a very strange behavior.

That is a nabble ad, a very clumsy technique. I believe you can turn off ads via a link on the footer. I'm not sure if you have to be registered for that to work.

juanccilleruelo, 2017-10-12:
But If there is security hole like this in the registration process, what can I wait to receive through the email? How many spam companies are going to have my electronic address?

juanccilleruelo, 2017-10-12:
Are you planning to create another service for forums?
Like Google+ or other.

I want to collaborate with the project if you want. 
We can initiate a discussion about this: 
What is the best option? 
Is it necessary to migrate current entries of the forums?
What is the cost and who finance it? (if applicable)

But do not ask me to subscribe first to the forums, please!!!  ;)

juanccilleruelo, 2017-10-12:
Looking for a new place for forums I've found this:

[http://discourse.org]http://www.discourse.org

[free-discourse-forum-hosting-for-community-friendly-github-projects/]https://blog.discourse.org/2016/03/free-discourse-forum-hosting-for-community-friendly-github-projects/

It's a real possibility, I think.

MichaelAtOz, 2017-10-12:
> But If there is security hole like this in the registration process, what can I wait to receive through the email? How many spam companies are going to have my electronic address?

What security hole?

The forum is hosted on Nabble, Nabble does a pop-up ad, it is NOT a hack or anything like it.

Until a few moths ago we did not have spam, when we got a 'manual' registering spammer who auto registered a userid and posted a 'spam' like forum post, which got sent to the mailing list. I introduced the new registration procedure where you need to be added by an Admin before you post. That stopped the (minor) spamming that was happening.

AFAIK there has NOT been leakage of email addresses that then get Spammed.
Any other forum/mailing-list users, please advise if you have had any spam since.

kintel, 2017-10-12:
@juanccilleruelo As hinted towards, the forums are simply a bi-directional interface to the mailing list. These forums are hosted by (the ad-financed) company Nabble, and the process is indeed a bit suboptimal. However, the forums are completely voluntary to use.
The mailing list itself can be found from our website or directly here: https://lists.openscad.org/mailman/listinfo/discuss_lists.openscad.org

One note about forums: Existing members of the mailing list have voiced a strong preference for maintaining an email interface to whatever future discussion forum we implement. That may guide our choice of technical solution going forward. Self-hosting is also an option (we've got a Wordpress site in the works).

In terms of way forward: If someone wants to volunteer to configure, run and maintain forums, that would be awesome. We don't have any financing, so any chosen solution should be close to free or self-financing.