openscad/openscad #2135: Initial release of 3D Mouse/joystick support

MichaelAtOz, 2017-10-17:
I'm raising a separate issue to facilitate a bounty.
The original request had my bounty which I needed to remove for various reasons.
As that shows the bounty paid-out I didn't want to risk a muddle.

Original request #1433, intermediate work #1807, current activity #2130.

The bounty covers basic functionality, including configuration and calibration, into Master (& working...).
($50 bounty pending)

MichaelAtOz, 2017-10-17:
@kintel is there something that needs to get added to the issue to link to bountysource?

kintel, 2017-10-17:
Not that I know of - their github integration stopped working at some point, so it's still a mystery what happens after closing..

MichaelPFrey, 2017-10-21:
Could you specify the operating system(s) you want/need?

MichaelAtOz, 2017-10-21:
@kintel is there an 'official' list of O.S.'s, I would expect Windows 7+ (32/64bit), macOS (versions?), and some Linuxi
@MichaelPFrey I need it for a SpaceMouse Wireless on Windows 7/64, but it should be multiplatform in design, for the bounty, W7/64, macOS (at least one version TBA) & at least one Linux (TBA). (?)
@t-paul you may want to comment on minimum support needed, particularly the various driver/other mechanisms?

t-paul, 2017-10-22:
In general it would be useful to have at least some kind of support for the 3 platforms we officially support. So yes, Windows, macOS, Linux would be nice and in theory there is already at least one driver for each platform which supports USB HID devices. I don't know if there are issues with the latest versions of Windows and macOS though.

t-paul, 2017-10-28:
http://files.openscad.org/snapshots/OpenSCAD-2017.10.28-x86-32_inputdriver5-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2017.10.28-x86-32_inputdriver5.zip
http://files.openscad.org/snapshots/OpenSCAD-2017.10.28-x86-64_inputdriver5-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2017.10.28-x86-64_inputdriver5.zip

MichaelPFrey, 2017-10-29:
Thank you for the windows binaries.

Note to other testers: The input drivers needs to be enabled under edit -> preferences -> Features -> (scroll down) -> input-driver and then openscad needs to be restarted.

sweet. I have tested it now with a SpaceExplorer under windows 10 64 bit, without any drivers installed.
The axis work fine and the default axis assigment makes sense.
The buttons do not work in my constelation.

MichaelAtOz, 2017-11-20:
@MichaelPFrey I think you have progressed some since there binaries? Worth a new one?

MichaelPFrey, 2017-11-20:
Yes, given that I have implemented deadzone and trim, it would be worth new binaries.

On the other hand: I would like to have #2166 merged and I have a gain/sensitivity in progress (will be ready soon).
That would be most features I need/would like to have.

t-paul, 2017-11-21:
http://files.openscad.org/snapshots/OpenSCAD-2017.11.20-x86-32_inputdriver5-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2017.11.20-x86-32_inputdriver5.zip
http://files.openscad.org/snapshots/OpenSCAD-2017.11.20-x86-64_inputdriver5-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2017.11.20-x86-64_inputdriver5.zip

MichaelPFrey, 2017-11-21:
tested with Space Explorer under Windows 10 x64

current issues:
[] default gain not set correctly - Workaround: goto preferences -> axis and set all gains to 1.00 (experiment with the value for zoom - even 9.99 still feels like not enough)
[] increase max. gain for zoom
[] default deathzone not set correctly - Workaround: goto preferences -> axis and set all deadzones to 0.1 (or more if required)
[] space explorer buttons do not work under windows 10 x64

For debuging the space explorer buttons I will have to at some debugging features to test under windows.
The rest are generall issues, that I can resolve under linux. (I should delete the openscad setting more often to test for this)

t-paul, 2017-11-21:
https://download.opensuse.org/repositories/home:/t-paul/AppImage/OpenSCAD-Branch-InputDriver-latest-x86_64.AppImage
https://download.opensuse.org/repositories/home:/t-paul/AppImage.arm/OpenSCAD-Branch-InputDriver-latest-armv7l.AppImage
https://download.opensuse.org/repositories/home:/t-paul/AppImage.arm/OpenSCAD-Branch-InputDriver-latest-aarch64.AppImage

MichaelPFrey, 2017-12-30:
Just a short deceleration: I don´t want to claim any part of the bounty.

First of all, what t-paul had developed was exactly the feature set I wanted to see and some more.
What I do is just using and polishing his infrastructure.

Second: I have declared my involvement here as hobby/educational project to my employer - receiving money would blur the line of just a hobby.

ofloveandhate, 2018-02-18:
i want this feature soooo bad, i'd pitch in on that bounty, ha.  i am a bit intimidated on the process of using the drivers posted.  they are for windows only, yeah?

t-paul, 2018-02-18:
Support is different per OS, best support is actually for Linux as there's the most options. GamePad and Connexion 3D mouse support should work for all platforms (although it may conflict with the native driver which we can't use).

Some more info in the [WIP](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/WIP#Inputdriver) section of the Manual.

What OS would you want to try it on? Getting some feedback for different devices would be quite helpful.

ofloveandhate, 2018-02-18:
my desired platforms are osx and linux -- i care only modestly about windows.  i use a 3dconnexion space mouse, and love it with blender.  would love to get that functionality in OpenSCAD, too.

when you say GamePad and Connexion 3D "should work for all platforms", do you mean as of some particular release number, or if this issue is resolved?

t-paul, 2018-02-18:
It's still all experimental and not yet offically integrated (although we are getting close). There are separate builds needed still.

Easiest way to try out would be using the Linux AppImage linked in the comment above from Nov 21. This link always points to the latest build for Linux and supports currently the Joystick based driver (which also works with the 3D Mouse, but needs the spacenavd driver to be not running).

With a recent Linux distribution it should be simply
* Download the AppImage
* `chmod 755 OpenSCAD-Branch-InputDriver-latest-x86_64.AppImage`
* Run with `./OpenSCAD-Branch-InputDriver-latest-x86_64.AppImage`

The feature needs to be enabled in the Preferences -> Features panel (check 'input-driver' box) and after restarting OpenSCAD the configuration panels should show up in Preferences.

MichaelAtOz, 2018-02-19:
> Just a short deceleration: I don´t want to claim any part of the bounty.

@t-paul I think you should claim the bounty, I had you in mind when I raised it. Due to [this](http://forum.openscad.org/Bountysource-developments-td22864.html#a22869) my paraoid side says you may want to do so soon.
Of course, kudos to @MichaelPFrey for your work, but as you explained you have employer issues.

Is it about time, feature wise, for another Windows exe? I've had my hand surgery so can do some testing again.

t-paul, 2018-11-07:
Closed via #2130.

closed 2018-11-07 by t-paul