openscad/openscad #2139: nonplanar tests fail with CGAL assertion violation

thehans, 2017-10-18:
I attempted to run the same tests that are run on travis, copying the list of tests from logs:
```
ctest -j8 -E 'opencsgtest_rotate_extrude-tests|opencsgtest_render-tests|opencsgtest_rotate_extrude-hole|opencsgtest_internal-cavity|opencsgtest_internal-cavity-polyhedron|opencsgtest_minkowski3-erosion|opencsgtest_issue835|opencsgtest_issue911|opencsgtest_issue913|opencsgtest_issue1215|opencsgtest_issue1105d|dxfpngtest_text-font-direction-tests|cgalpngtest_text-font-direction-tests|opencsgtest_text-font-direction-tests|csgpngtest_text-font-direction-tests|svgpngtest_text-font-direction-tests|throwntogethertest_text-font-direction-tests|throwntogethertest_issue964|opencsgtest_issue1165|opencsgtest_issue1258|throwntogethertest_issue1089|throwntogethertest_issue1215'
```

I ran this on master HEAD (which was on 0737990f12a7d9e0bbae195607af3f3710f03056 at the time) and 4 tests (all with nonplanar in the name) failed with CGAL assertion errors like the following:
```
stderr output: ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: ss_plane.has_on(sv_prev->point())
File: /usr/local/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h
Line: 294
CGAL Cache insert: polyhedron(points=[[0.0174497,-0.0174524 (284912 bytes)
```

The following tests FAILED:
        323 - cgalpngtest_polyhedron-nonplanar-tests (Failed)
        634 - csgpngtest_polyhedron-nonplanar-tests (Failed)
        875 - monotonepngtest_polyhedron-nonplanar-tests (Failed)
        898 - cgalstlcgalpngtest_polyhedron-nonplanar-tests (Failed)

In the generated output images, the object with nonplanar faces is not rendered, there is empty space there.  HTML report output attached (renamed to comply with github's dumb filename extension filter)
[linux_x86_64_quadro-m1000m_sjgi_report.html.txt](https://github.com/openscad/openscad/files/1395489/linux_x86_64_quadro-m1000m_sjgi_report.html.txt)

Library info for my system:
```
OpenSCAD Version: 2017.10.17
System information: Linux 4.4.0-87-generic #110-Ubuntu SMP Tue Jul 18 12:55:35 UTC 2017 x86_64 Linux Mint 18.1 8 CPUs 15.51 GB RAM
Compiler: GCC "5.4.0 20160609" 64bit
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.10-I-900, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.5.1
QScintilla version: 2.9.1
MingW build: No
GLib version: 2.48.2
libzip version: 1.0.1
Application Path: /home/hans/projects/openscad
Documents Path: /home/hans/.local/share
Resource Path: /home/hans/projects/openscad
User Library Path: /home/hans/.local/share/OpenSCAD/libraries
User Config Path: /home/hans/.config/OpenSCAD
Backup Path: /home/hans/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/hans/.local/share/OpenSCAD/libraries
/home/hans/projects/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.13.0
OpenGL Version: 4.5.0 NVIDIA 375.66
GL Renderer: Quadro M1000M/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8880), depth(24), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(24), stencil(0)

GL Extensions:
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
[... 302 lines cut ...]
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vdpau_interop
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
```

kintel, 2017-10-19:
FYI it works on Mac with the latest master and the following system info:
```
OpenSCAD Version: 2017.10.15
System information: Mac OS X Version 10.12.6 (Build 16G29) MacBookPro11,3 4 CPUs 16.00 GB RAM 
Compiler: Clang "9.0.0 (clang-900.0.37)" 64bit
Boost version: 1_61
Eigen version: 3.3.1
CGAL version, kernels: 4.8.1, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.7.0
QScintilla version: 2.9.3
MingW build: No
GLib version: 2.50.1
libzip version: 1.1.3
Application Path: /Users/kintel/code/OpenSCAD/openscad/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/kintel/Documents
Resource Path: /Users/kintel/code/OpenSCAD/openscad
User Library Path: /Users/kintel/Documents/OpenSCAD/libraries
User Config Path: /Users/kintel/Library/Application Support/OpenSCAD
Backup Path: /Users/kintel/Documents/OpenSCAD/backups
OPENSCADPATH: /Users/kintel/code/OpenSCAD/openscad/tests/../libraries
OpenSCAD library path:
  /Users/kintel/code/OpenSCAD/openscad/tests/../libraries
  /Users/kintel/Documents/OpenSCAD/libraries
  /Users/kintel/code/OpenSCAD/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:

GLEW version: 2.0.0
OpenGL Version: 2.1 NVIDIA-10.17.5 355.10.05.45f01
GL Renderer: NVIDIA GeForce GT 750M OpenGL Engine
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: Cocoa / CGL
PNG generator: Core Foundation
OS info: Mac OS X 10.12.6 (x86_64 kernel)
Machine: 64-bit
```

thehans, 2017-10-19:
@kintel That's because mine is built on CGAL 4.10, whereas yours is 4.8

It looks like CGAL 4.10 added some additional assertion checks, which affect the behavior of openscad for these sort of tests.

kintel, 2017-10-22:
Wops, didn't see the title..

supervacuus, 2017-10-22:
I can verify the failing assertions in CGAL with 4.10.1 with bc00974cdcbf0366123fe2dc38253b1ba345dee0  and the sysinfo at the end. According to git-blame the assertions have not been changed for 10 years: 

https://github.com/CGAL/cgal/blame/a4c29ed9aaba4a6f543888122bbb9205c29bc66a/Nef_3/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h#L294

So it might be that actual behaviour leading up to the assertions has changed in 4.10

```
OpenSCAD Version: 2017.10.22
System information: Linux 4.13.5-200.fc26.x86_64 #1 SMP Thu Oct 5 16:53:13 UTC 2017 x86_64 Fedora 26 (Workstation Edition) 4 CPUs 15.55 GB RAM
Compiler: GCC "7.2.1 20170915 (Red Hat 7.2.1-2)" 64bit
Boost version: 1_63
Eigen version: 3.3.4
CGAL version, kernels: 4.10.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.7.1
QScintilla version: 2.10
MingW build: No
GLib version: 2.52.3
libzip version: <not enabled>
Application Path: /home/mischan/devel/source/openscad
Documents Path: /home/mischan/.local/share
Resource Path: /home/mischan/devel/source/openscad
User Library Path: /home/mischan/.local/share/OpenSCAD/libraries
User Config Path: /home/mischan/.config/OpenSCAD
Backup Path: /home/mischan/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/mischan/.local/share/OpenSCAD/libraries
  /home/mischan/devel/source/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:

GLEW version: 2.0.0
OpenGL Version: 3.0 Mesa 17.2.2
GL Renderer: Mesa DRI Intel(R) Sandybridge Mobile 
GL Vendor: Intel Open Source Technology Center
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: GLX
PNG generator: lodepng
GLX version: 1.4
OS info: Linux 4.13.5-200.fc26.x86_64 #1 SMP Thu Oct 5 16:53:13 UTC 2017
Machine: x86_64
```

kintel, 2017-12-01:
Confirmed on Travis CI (Mac) using CGAL 4.11

commit c0ed9b57 (2017-12-02): Matched against changed exception in CGAL 4.10. Fixes #2139
closed 2017-12-02 by kintel