openscad/openscad #2146: Add units to SVG export width and height

JamesNewton, 2017-10-23:
I'd like to point out that /units/ are not added to height and width which was mentioned by the OP in #1597 but not addressed with the issue was closed. e.g. we now get:
```html
<svg width="10" height="10" ... />
```
but we really should get:
```html
<svg width="10mm" height="10mm" ... />
```
This makes it necessary to re-size the SVG files in e.g. laser cutter or other software in order to re-establish the correct size of the total drawing. This then requires /knowing/ the original size, and if you don't know that it's in the SVG file it can be very confusing to the operator. I.E. you must realize that the SVG file you just imported is NOT at the correct size because it is assuming px or other "user" units, and not mm, and then you need to know that you must go into the raw SVG file to find the height and width, and then you must re-size the drawing to that in the software. Often this requires manually converting units. e.g. if the software expects inches, then you convert mm to inches and re-size to that. Or course, at that point, you might as well just have edited the SVG to add "mm" after the width and height units.

It would be really nice to include the units so the program using the SVG file could just do that for you and the drawing would automatically be the correct size. 

Having said that, I worry that there may be people who depend on the SVG export being in "user" units (e.g. no unit of measure specified) as it currently is.  "Speak now or forever hold your peace"?

JamesNewton, 2017-10-23:
This is worth reading:
https://mpetroff.net/2013/08/analysis-of-svg-units/

t-paul, 2017-10-23:
Nightly builds already export with `mm`.

Which is not actually the correct solution, but the simple compromise. Mid term it could be a solution to add settings to the export dialog (#649).

Having units fully supported by the language might be an option but that's probably too huge a project at this point.

closed 2017-10-23 by t-paul

filipgoc, 2018-07-15:
Thanks for mentioning that nightly has this fixes @t-paul! Saved me.