openscad/openscad #2157: Preview stopped working properly since Ubuntu 17.10 update

theopensourcerer, 2017-10-30:
(Mentioned originally in #1409)

Since upgrading Ubuntu from 16.10 to 17.10 a few days ago the preview quite often simply disappears from the viewport. Pressing F6 to render it works fine. If I quit OpenSCAD and open it again with the same design file, the preview is visible again but once I change something and save it then (mostly) disappears and I have to do a F6 to get back a drawing.

I am not sure if this is something to do with the new Weyland graphics or Gnome desktop, an update to OpenSCAD (am using the Ubuntu packaged version of OpenSCAD) or something else?

t-paul, 2017-10-30:
Can you please add the output of `Help`->`Library Info`.

theopensourcerer, 2017-10-30:
No problem. Will do it tonight when I get home. Thanks.

theopensourcerer, 2017-10-30:
Here you go:

OpenSCAD Version: 2015.03-2
System information: Linux 4.13.0-16-generic #19-Ubuntu SMP Wed Oct 11 18:35:14 UTC 2017 x86_64 Ubuntu 17.10 4 CPUs 15.55 GB RAM
Compiler: GCC "6.2.1 20161124" 64bit
Boost version: 1_62
Eigen version: 3.3.0
CGAL version, kernels: 4.9, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.9.1
QScintilla version: 2.9.3
MingW build: No
GLib version: 2.51.0
Application Path: /usr/bin
Documents Path: /home/alanlord/.local/share
Resource Path: /usr/share/openscad
User Library Path: /home/alanlord/.local/share/OpenSCAD/libraries
User Config Path: /home/alanlord/.config/OpenSCAD
Backup Path: /home/alanlord/.local/share/OpenSCAD/backups
OPENSCADPATH: /home/alanlord/Personal/OpenSCAD_Libraries
OpenSCAD library path:
/home/alanlord/Personal/OpenSCAD_Libraries
/home/alanlord/.local/share/OpenSCAD/libraries
/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 2.0.0
OpenGL Version: 3.0 Mesa 17.2.2
GL Renderer: Mesa DRI Intel(R) HD Graphics 530 (Skylake GT2) 
GL Vendor: Intel Open Source Technology Center
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QOpenGLWidget

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
[... 184 lines cut ...]
GL_KHR_robustness
GL_KHR_texture_compression_astc_ldr
GL_KHR_texture_compression_astc_sliced_3d
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

thehans, 2017-10-31:
@theopensourcerer If you launch openscad from the terminal and reproduce the issue, do you see any error messages etc output there?

And no errors in the OpenSCAD GUI console?

theopensourcerer, 2017-10-31:
Correct. 

It prints just the normal Compile and preview finished, and Rendering statements at the end and no errors (unless I have something wrong in the design code of course).

theopensourcerer, 2017-10-31:
Ahh - sorry I missed the first part of your question re. the cli. I will try that tonight - good idea.

theopensourcerer, 2017-11-01:
@thehans Just tried it and nothing logged to the terminal after starting it from the command line. The problem happens very frequently.

theopensourcerer, 2017-11-01:
Here's a bit more info. I created a really simple new drawing.

`cube([20,30,40],true);`

Just as a single cube. the preview always works. It doesn't disappear.

If I then do something like this:

```
difference() {
    cube([20,30,40],true);
    cube([10,10,10]);
}
```

The the preview disappears all the time. I can only view the drawing using F6. F5 produces an empty viewport.

If I comment out either of the two cubes in the difference() function, the preview drawing reappears again on save and F5/F6 work as expected..

No errors are reported anywhere.

theopensourcerer, 2017-11-01:
Something else... I closed (exited) OpenSCAD and re-opened it and loaded this "test" file above.. F5 and F6 worked as expected. I couldn't get F5 to not work until I resized OpenSCAD. It was full screen. When I shrank the window a bit, the preview disappeared again. Putting it back to full-screen didn't help to resolve the issue.

freedriven, 2017-11-02:
I'm have a very similar/same problem on Ubuntu 17.10, but I'm a new user of OpenSCAD and haven't used it on previous versions of Ubuntu.

Basically, several/many previous shapes will disappear until I close OpenSCAD and reopen the file.

donbright, 2017-12-13:
can you try a newer version of OpenSCAD?

theopensourcerer, 2017-12-13:
@donbright There don't appear to have been any more recent releases than the one the Ubuntu package maintainers are providing [1]... Is there a beta release or something?

[1] https://packages.ubuntu.com/artful/openscad

freedriven, 2017-12-13:
I downloaded the latest appimage (2017.12.13), and it appears to have fixed several of the problems I was having with the version in the standard Ubuntu repositories (2015.03). I'll update if I still find issues.
THANKS!

freedriven, 2017-12-13:
It appears there are development snapshots, both appimages and repositories) available here (this is where I found the 2017.12.13 appimage):
http://www.openscad.org/downloads.html

theopensourcerer, 2017-12-13:
@freedriven Thanks for the cluebat - I never looked "below the fold" ;-)

I'll test when I get home and report back (it'll be a day or two)

theopensourcerer, 2017-12-14:
I downloaded the x64 appimage and it doesn't exhibit the same symptoms as the Ubuntu packaged version. It is easy to replicate the problem. Open any reasonably complex design in the Ubuntu packaged version and change the from full screen to a smaller window - the F5 quick render disappears or only some parts of the image are shown. In the appimage code - this does not happen. 

Thanks.

simon1tan, 2018-01-23:
I am having a similar issue where some of my components do not show in preview but renders fine. Also installed from Ubuntu 17.10 repository. Just installed nightly and the preview looks working properly.

kintel, 2018-01-23:
Closing as fixed since newer versions works as expected.

closed 2018-01-23 by kintel

markfinn, 2018-09-03:
Newer versions don't work as expected.  I'm trying 2018.09.03.nightly right now.  From the previous comments I wonder if it is the appimage that worked, not the new versions (see https://github.com/openscad/openscad/issues/2157#issuecomment-351675118)

please re-open.

donbright, 2018-09-03:
@markfinn do you have

output of help/library info

an example .scad file that shows the problem reliably?

t-paul, 2018-09-03:
If the LibraryInfo shows anything lower than OpenCSG 1.4.2, we can't do much about it. Maybe there's a backport available somewhere. Ubuntu 18.04 is the first to ship with OpenCSG 1.4.2.

donbright, 2018-09-03:
back in the old days we had to ask people to build OpenCSG and CGAL on just about every distro since most distros did not have them. i think it would be possible to go back to that, and just ban building with older OpenCSG.

markfinn, 2018-09-03:
I'm trying both the PPA and nightly, both exibit the problems.  And I just went from 17.10 to 18.04. none of these fixes it.

Yes it's OpenCSG 1.4.2:

OpenSCAD Version: 2018.09.03.nightly (git ce08fcd)
System information: Linux 4.15.0-33-generic #36-Ubuntu SMP Wed Aug 15 16:00:05 UTC 2018 x86_64 Ubuntu 18.04.1 LTS 12 CPUs 23.53 GB RAM
Compiler: GCC "7.3.0" 64bit
Boost version: 1_65_1
Eigen version: 3.3.4
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.9.5
QScintilla version: 2.10.2
MingW build: No
GLib version: 2.56.1
libzip version: 
Application Path: /usr/bin
Documents Path: /home/mark/.local/share
Resource Path: /usr/share/openscad-nightly
User Library Path: /home/mark/.local/share/OpenSCAD/libraries
User Config Path: /home/mark/.config/OpenSCAD
Backup Path: /home/mark/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/mark/.local/share/OpenSCAD/libraries
/home/mark

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 2.0.0
OpenGL Version: 4.6.0 NVIDIA 390.48
GL Renderer: GeForce GTX 1050 Ti/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8880), depth(24), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(24), stencil(0)

GL Extensions:
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
[... 321 lines cut ...]
GL_NV_texture_rectangle_compressed
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vdpau_interop
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum

markfinn, 2018-09-03:
oooh.  setting "turn off rendering at" to an absurdly high value fixes it.  

With that set at the default of 2000 I get and a "simplified" object which works I get:

`Compiling design (CSG Products normalization)...
WARNING: Normalized tree is growing past 4000 elements. Aborting normalization. 
Normalized CSG tree has 1424 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds`

but for a "complex" object (that fails) I get:

`Compiling design (CSG Products normalization)...
WARNING: Normalized tree is growing past 4000 elements. Aborting normalization. 
Normalized CSG tree has 1044 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds`

I find it odd that the normalized tree has fewer objects when the scene is more complex.

For reference, when I set the "rendering off" threshold really high I get no warnings followed by 2638 elements for the complex and 1508 for the simple one. both render correctly.

code follows.  my "simple" and "complex" quoted above comes from changing on and off the element at:
`//ring
if (0) `

Code:
`use <threads.scad>

$fn=120;

module cap(knurled=0)
{
    union(){
        difference(){
            union(){
                cylinder(r=29, h=20);
                if (knurled) {
                    for(r=[0:45:360])
                        rotate([0,0,r])
                        translate([0,27,0])
                        cylinder(r=6, h=20);
                    for(r=[0:5:360])
                        rotate([0,0,r])
                        translate([0,29,0])
                        cylinder(r=.6, h=20);
                }
            }
            translate([0,0,-1])
            metric_thread (50.8, 4, 17, thread_size=4.5, rectangle=.5, internal=true);
            translate([0,0,-1])
            cylinder(r2=24, r1=24+4, h=4);

        }

    cylinder(r2=18+7/32*20, r1=18, h=20);

    }
}

//cap
if(0) 
translate([60,0,20])
rotate([180,0,0])
cap(knurled=1);

translate([-60,0,20])
rotate([180,0,0])
difference(){
    cap(knurled=1);
    cylinder(r=14, h=100, center=true);
    translate([0,0,16-100])
    cylinder(r=23, h=100);
}

//plug
if (1) 
    translate([0,0,10])
    difference(){
    union(){
        intersection(){
            translate([0,0,4-20])
            cylinder(r2=18+7/32*20, r1=18, h=20);
            translate([0,0,-10])
            cylinder(r=50, h=10);
            translate([0,0,-15])
            cylinder(r2=32, r1=13.5, h=20);
        }
        translate([0,0,-1])
        english_thread (17/16, 11.5, .75, leadin=1);
    }

    cylinder(r=10, h=100, center=true);
       translate([0,0,-11])
         cylinder(r2=9, r1=12, h=5);
}`

t-paul, 2018-09-03:
The current default is actually 100000 and even higher values should be ok on most machines nowadays.

Anyway, this has nothing to do with the originally reported issue then.