openscad/openscad #2165: Missing expected images for tests

thehans, 2017-10-31:
I ran `ctest -C All` on Linux Mint 18, using a cmake build, and 45 of 1454 tests failed.  Of those failures about one third were simply due to missing test data.  I've cut out just the relevant error lines from my test report.  I can attach the full report also if wanted.
```
Error: test 'cgalpngtest' is missing expected output in /home/hans/projects/openscad/tests/regression/cgalpngtest/sign-expected.png
Error: test 'cgalpngtest' is missing expected output in /home/hans/projects/openscad/tests/regression/cgalpngtest/issue1529-expected.png
Error: test 'cgalpngtest' is missing expected output in /home/hans/projects/openscad/tests/regression/cgalpngtest/issue1455-expected.png
Error: test 'cgalpngtest' is missing expected output in /home/hans/projects/openscad/tests/regression/cgalpngtest/candleStand-expected.png
Error: test 'csgpngtest' is missing expected output in /home/hans/projects/openscad/tests/regression/cgalpngtest/sign-expected.png
Error: test 'csgpngtest' is missing expected output in /home/hans/projects/openscad/tests/regression/cgalpngtest/issue1529-expected.png
Error: test 'csgpngtest' is missing expected output in /home/hans/projects/openscad/tests/regression/cgalpngtest/issue1455-expected.png
Error: test 'csgpngtest' is missing expected output in /home/hans/projects/openscad/tests/regression/cgalpngtest/candleStand-expected.png
Error: test 'opencsgtest' is missing expected output in /home/hans/projects/openscad/tests/regression/opencsgtest/sign-expected.png
Error: test 'opencsgtest' is missing expected output in /home/hans/projects/openscad/tests/regression/opencsgtest/candleStand-expected.png
Error: test 'opencsgtest' is missing expected output in /home/hans/projects/openscad/tests/regression/opencsgtest/issue1455-expected.png
Error: test 'throwntogethertest' is missing expected output in /home/hans/projects/openscad/tests/regression/throwntogethertest/sign-expected.png
Error: test 'throwntogethertest' is missing expected output in /home/hans/projects/openscad/tests/regression/throwntogethertest/candleStand-expected.png
Error: test 'monotonepngtest' is missing expected output in /home/hans/projects/openscad/tests/regression/monotonepngtest/issue1455-expected.png
Error: test 'cgalstlcgalpngtest' is missing expected output in /home/hans/projects/openscad/tests/regression/monotonepngtest/issue1455-expected.png
stderr output: Can't open input file '/home/hans/projects/openscad/tests/../testdata/scad/bugs/issue1529.scad'!
```

kintel, 2017-10-31:
Some of these are tests of the examples - they should definitely be updated.
The other are known bugs - updating those with expected files are sometimes trickier, but often possible by working around the bugs to generate good expected images.

thehans, 2017-11-19:
Thanks for the merge.  As you mentioned the remaining ones are known bugs, which are still open, so they can continue to be tracked in their own issues:
#1455 
#1529

Closing this

closed 2017-11-19 by thehans