openscad/openscad #2167: CGAL 4.7 iterator assertion errors in tests

thehans, 2017-10-31:
I ran ctest -C All on Linux Mint 18, using a cmake build, and 45 of 1454 tests failed. Of those failures about one third were from CGAL assertion errors.  When one of these occurs, the output image is an empty scene with just background color.  I've cut out just the relevant error lines from my test report. I can attach the full report also if wanted.

4 tests failed with this assertion:
```
stderr output: ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation!
Expr: itl != it->second.end()
File: /usr/include/CGAL/Nef_3/SNC_external_structure.h
Line: 1106
```
those tests were:
```
cgalpngtest_issue802
csgpngtest_issue802
monotonepngtest_issue802
cgalstlcgalpngtest_issue802
```

And 10 tests failed with 
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: pe_prev->is_border() || !internal::Plane_constructor::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate()
File: /usr/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h
Line: 251
```
those tests were:
```
cgalstlcgalpngtest_issue945e
cgalstlcgalpngtest_issue13c
cgalstlcgalpngtest_issue945f
cgalstlcgalpngtest_issue1580-back-to-back
cgalstlcgalpngtest_issue1580-import-back-to-back
cgalstlcgalpngtest_issue1876
offcgalpngtest_issue1105b
offcgalpngtest_issue1105c
offcgalpngtest_issue1258
offcgalpngtest_issue1215b
```

thehans, 2017-11-19:
Of these failing issues, three are already considered "Closed", so maybe they should be reopened:
#1105
#1215
#1258

kintel, 2017-11-19:
I believe those issues reported were indeed fixed (e.g. crashes). What you experience are probably duplicates of existing issues related to CGAL.
I'll try to run this locally and see how that differs.

kintel, 2017-11-19:
I think the #1215 issue above is related to #910 as it only fails for the OFF export

kintel, 2017-11-19:
I think the correct resolution would be to move the OFF test cases back to Bugs, perhaps with a comment that they're there due to #910.

closed 2024-03-29 by kintel