openscad/openscad #2177: Feature Request: code completion in editor

thehans, 2017-11-03:
It would be nice if the OpenSCAD editor had ability to offer code completion.  

As the user begin typing a module/function name it could offer suggestions that fit the prefix, arrows to select between multiple options, tab or enter to complete.

I think having the ability to show function signature including potential parameter names, etc. would be a huge boon to scad productivity.  Even after years of experience with OpenSCAD, I find myself sometimes forgetting for example if the param name for linear_extrude is 'height' or 'h' or 'length', etc.  So I open a browser tab navigate to OpenSCAD wiki and find the module in the documenation.  If there was a quick reference at your fingertips while writing code it would be a lot more convenient.

I have now idea how this would work or how much effort would be involved: if qscintilla has any built in mechanisms, or plugins for doing this?

t-paul, 2017-11-03:
Old/obsolete prototype https://github.com/openscad/openscad/pull/905
There's also patches in http://forum.openscad.org/Abbreviations-td17583.html#a17635

Basically most of the feature is there, but it needs to be polished a bit to be included.

MichaelAtOz, 2017-11-04:
Also note you can use an [external editor](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_an_external_Editor_with_OpenSCAD) with that ability. Works quite well, I setup the 'save' function as F5, and hey-presto, press F5 and it previews.

VanLaser, 2019-04-11:
I want to add my small comment/input to this request: the implementation of this feature (completion, autocompletion) should really be handled by OpenSCAD in such a way that it will be usable for both the integrated editor *and* for any external editor. What I mean by that is that, even for an external editor, it is still OpenSCAD's task to have a way to export enough information, when asked (either using a shared library, or by querying the executable with some flags), so for example an external editor could simply ask: at this position, what completion solution(s) I have, and - in case we are in a module argument list -  what parameters are possible (names etc.). In a way, something similar to clang-based solutions for C/C++. Otherwise, any external editor completion implementation would have to invent its own parser for the OpenSCAD language and in practice will also always lag behind the new features implemented in OpenSCAD.

TLDR: offering code completion features for any editor *should be* a task internal to *OpenSCAD*.

t-paul, 2019-04-11:
@VanLaser What you are describing is covered by the #2629 feature request (https://langserver.org/).

VanLaser, 2019-04-11:
Indeed that is, thanks for the pointer!

thehans, 2019-08-22:
It seems that this functionality has been added by PR #2975
and further improvements are being tracked by issue #3047, so I'm going to close this one.

Thank you @RomitKumar !

closed 2019-08-22 by thehans

MarkJeronimus, 2021-04-23:
> and further improvements are being tracked by issue #3047, so I'm going to close this one.

That issue is contributor-only.

And I do have an issue with code completion. It's too eager. When I type variable name like `r1`, it starts suggesting at the `r`, blocking my view of the lines below. I know I can increase the character threshold, but I like autocomplete to happen with one letter, just either after a configurable delay (e.g. 400ms) or when I invoke the completion manually (my muscle-memory defaults to ctrl+space).

t-paul, 2021-04-23:
I don't see support for setting a delay, so that would need to be implemented first in the editor component before we can provide a configuration. Manual trigger via CTRL+Space already exists, so I'm not sure what the question is with that.