openscad/openscad #2180: Builds with -DNDEBUG flag crash on issue1672.scad with "stack smashing detected"

thehans, 2017-11-06:
This is an issue that was discovered while testing the cmake build system for openscad

 cmake supports 4 builtin build types: Debug,Release,RelWithDebInfo, and MinSizeRel.  
If any cmake build type *other than* `Debug` is specified: for ex `cmake -DCMAKE_BUILD_TYPE=Release`, then the default flags that cmake uses for this build type will include `-DNDEBUG` among others.

With this flag, the cmake and make processes complete without error, and the `openscad` executable runs fine for the most part, but there is one case where it causes a crash.  I found this by and running the test suite (`ctest -j8 -C All`) on both a build with and without `NDEBUG` defined, and these 4 additional tests failed(all related to issue 1672):
```
376 - cgalpngtest_issue1672 (Failed)
621 - opencsgtest_issue1672 (Failed)
863 - csgpngtest_issue1672 (Failed)
1105 - throwntogethertest_issue1672 (Failed)
```
the ctest console output shows the following error for each of these:
```
*** stack smashing detected ***: /home/hans/projects/openscad_test/openscad_cmake/tests/../openscad terminated
```

The crash can easily be reproduced manually by loading `testdata/scad/3D/issues/issue1672.scad` and hitting preview.

My vague understanding/assumption about the issue is that CGAL may have some assertions that would catch certain types of errors, but that these assertion get skipped over when `NDEBUG` is defined, causing some illegal operation.

Here is a backtrace from gdb when it crashed:
```
[New Thread 0x7fffd7239700 (LWP 12796)]
[Thread 0x7fffd7239700 (LWP 12796) exited]
*** stack smashing detected ***: /home/hans/projects/openscad/openscad terminated

Thread 1 "openscad" received signal SIGABRT, Aborted.
0x00007ffff2979428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff2979428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff297b02a in __GI_abort () at abort.c:89
#2  0x00007ffff29bb7ea in __libc_message (do_abort=do_abort@entry=1, fmt=fmt@entry=0x7ffff2ad345f "*** %s ***: %s terminated\n") at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff2a5d11c in __GI___fortify_fail (msg=<optimized out>, msg@entry=0x7ffff2ad3441 "stack smashing detected") at fortify_fail.c:37
#4  0x00007ffff2a5d0c0 in __stack_chk_fail () at stack_chk_fail.c:28
#5  0x000000000068fa7c in CGAL::convex_hull_3<std::_List_iterator<CGAL::Point_3<CGAL::Epick> >, CGAL::Polyhedron_3<CGAL::Epick, CGAL::Polyhedron_items_3, CGAL::HalfedgeDS_default, std::allocator<int> >, CGAL::Convex_hull_traits_3<CGAL::Epick, CGAL::Boolean_tag<true> > > (first=..., beyond=..., polyhedron=..., traits=...) at /usr/include/CGAL/convex_hull_3.h:941
#6  0x0000000000693692 in CGAL::convex_hull_3<std::_List_iterator<CGAL::Point_3<CGAL::Epick> >, CGAL::Polyhedron_3<CGAL::Epick, CGAL::Polyhedron_items_3, CGAL::HalfedgeDS_default, std::allocator<int> > > (polyhedron=..., beyond=..., 
    first=...) at /usr/include/CGAL/convex_hull_3.h:950
#7  CGALUtils::applyHull (children=..., result=...) at /home/hans/projects/openscad/src/cgalutils-applyops.cc:189
#8  0x000000000071a5aa in GeometryEvaluator::applyToChildren3D (this=this@entry=0x7fffffffaed0, node=..., op=op@entry=OpenSCADOperator::HULL) at /home/hans/projects/openscad/src/GeometryEvaluator.cc:113
#9  0x000000000071b1a6 in GeometryEvaluator::applyToChildren (this=this@entry=0x7fffffffaed0, node=..., op=op@entry=OpenSCADOperator::HULL) at /home/hans/projects/openscad/src/GeometryEvaluator.cc:96
#10 0x000000000071bdd8 in GeometryEvaluator::visit (this=0x7fffffffaed0, state=..., node=...) at /home/hans/projects/openscad/src/GeometryEvaluator.cc:1086
#11 0x00000000004aaf0e in NodeVisitor::traverse (this=this@entry=0x7fffffffaed0, node=..., state=...) at /home/hans/projects/openscad/src/NodeVisitor.cc:30
#12 0x000000000071968a in GeometryEvaluator::evaluateGeometry (this=0x7fffffffaed0, node=..., allownef=allownef@entry=false) at /home/hans/projects/openscad/src/GeometryEvaluator.cc:57
#13 0x00000000005a578d in CSGTreeEvaluator::visit (this=0x7fffffffafc0, state=..., node=...) at /home/hans/projects/openscad/src/CSGTreeEvaluator.cc:276
#14 0x00000000004aaf0e in NodeVisitor::traverse (this=this@entry=0x7fffffffafc0, node=..., state=...) at /home/hans/projects/openscad/src/NodeVisitor.cc:30
#15 0x00000000004aaf37 in NodeVisitor::traverse (this=this@entry=0x7fffffffafc0, node=..., state=...) at /home/hans/projects/openscad/src/NodeVisitor.cc:20
#16 0x00000000005a2b85 in CSGTreeEvaluator::buildCSGTree (this=this@entry=0x7fffffffafc0, node=...) at /home/hans/projects/openscad/src/CSGTreeEvaluator.cc:33
#17 0x0000000000742cce in MainWindow::compileCSG (this=this@entry=0xd847f0, procevents=<optimized out>) at /home/hans/projects/openscad/src/mainwin.cc:1163
#18 0x000000000074c62f in MainWindow::csgRender (this=0xd847f0) at /home/hans/projects/openscad/src/mainwin.cc:1872
#19 0x00000000007a60b6 in MainWindow::qt_static_metacall (_o=<optimized out>, _c=<optimized out>, _id=<optimized out>, _a=<optimized out>) at /home/hans/projects/openscad/moc_MainWindow.cpp:557
#20 0x00007ffff384084a in QMetaMethod::invoke(QObject*, Qt::ConnectionType, QGenericReturnArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument) const () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007ffff3845f0d in QMetaObject::invokeMethod(QObject*, char const*, Qt::ConnectionType, QGenericReturnArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x000000000074298d in QMetaObject::invokeMethod (val9=..., val8=..., val7=..., val6=..., val5=..., val4=..., val3=..., val2=..., val1=..., val0=..., member=<optimized out>, obj=0xd847f0)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs.h:425
#23 MainWindow::compileDone (this=this@entry=0xd847f0, didchange=<optimized out>) at /home/hans/projects/openscad/src/mainwin.cc:1064
#24 0x000000000074b57e in MainWindow::compile (this=this@entry=0xd847f0, reload=reload@entry=false, forcedone=forcedone@entry=false) at /home/hans/projects/openscad/src/mainwin.cc:992
#25 0x000000000074baf9 in MainWindow::actionRenderPreview (this=0xd847f0) at /home/hans/projects/openscad/src/mainwin.cc:1861
#26 0x00000000007a60ac in MainWindow::qt_static_metacall (_o=<optimized out>, _c=<optimized out>, _id=<optimized out>, _a=<optimized out>) at /home/hans/projects/openscad/moc_MainWindow.cpp:556
#27 0x00007ffff3863d2a in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x00007ffff411e412 in QAction::triggered(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#29 0x00007ffff4120898 in QAction::activate(QAction::ActionEvent) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#30 0x00007ffff4121204 in QAction::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#31 0x00007ffff412805c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#32 0x00007ffff412d516 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#33 0x00000000007590cf in OpenSCADApp::notify (this=<optimized out>, object=<optimized out>, event=<optimized out>) at /home/hans/projects/openscad/src/OpenSCADApp.cc:30
#34 0x00007ffff383538b in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#35 0x00007ffff3b9f888 in QShortcutMap::dispatchEvent(QKeyEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#36 0x00007ffff3b9f9c1 in QShortcutMap::tryShortcutEvent(QObject*, QKeyEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#37 0x00007ffff3b740ea in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#38 0x00007ffff3b79185 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#39 0x00007ffff3b5cf08 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
[... cut ...]
#45 0x00007ffff3832b4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#46 0x00007ffff383abec in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#47 0x0000000000453f2d in gui (inputFiles=std::vector of length 1, capacity 1 = {...}, original_path=..., argc=argc@entry=1, argv=argv@entry=0x7fffffffdc08) at /home/hans/projects/openscad/src/openscad.cc:774
#48 0x0000000000442a0e in main (argc=1, argv=0x7fffffffdc08) at /home/hans/projects/openscad/src/openscad.cc:988
```

donbright, 2017-12-13:
i think this is a good example of the relationship between the Simplicity of the build system and the Maintenance of the system as a whole. I notice Go and Rust have kind of moved away from 'selectable' asserts. 

Golang: https://golang.org/doc/faq

"Go doesn't provide assertions. They are undeniably convenient, but our experience has been that programmers use them as a crutch to avoid thinking about proper error handling and reporting. "

Rust: https://doc.rust-lang.org/std/macro.assert.html

"Assertions are always checked in both debug and release builds, and cannot be disabled."

It seems to be the argument for this complicated stuff is "performance" but i never see any performance numbers. In my opinion it is not worth it to improve speed by 10% if it causes a bunch of maintenance problems. 

Although, against my argument, Swift apparently has like 6 different kinds http://blog.krzyzanowskim.com/2015/03/09/swift-asserts-the-missing-manual/

Anyways I don't remember much about  NDEBUG, DEBUG, Openscad, and CGAL but my vote would be to see if we can force CGAL to always be compiled with either DEBUG or NDEBUG so that the behavior is consistent.  (wait... is that what we used to do back in the day? cant remember...)

thehans, 2019-03-05:
Closed by #2840

closed 2019-03-05 by thehans