openscad/openscad #2182: Segmentation fault when quitting application (QSettingsCached)

olnabe, 2017-11-06:
System: Gentoo x86-64 (up-to-date) with qtcore-5.7.1 (dev-qt/qtcore-5.7.1-r3)
Openscad version:  2017.11.05  https://github.com/openscad/openscad.git (media-gfx/openscad-9999)

Steps:
1. From a terminal run command: openscad
2. On the shown dialog, press the "New" button 
3. From the openscad mainwindow , click "File" and then click "Quit"

The problem:
1.Openscad quits but the terminal shows: "Segmentation fault (core dumped)"
2.And after the above steps, there are two files in  $HOME/.config/OpenSCAD/
OpenSCAD.conf
OpenSCAD.conf.lock

Backtrace of the coredump:
(gdb) bt
#0  0x00007fc13eee5670 in QDateTime::QDateTime() () from /usr/lib64/libQt5Core.so.5
#1  0x00007fc13eee9d7b in QDateTime::fromMSecsSinceEpoch(long long, Qt::TimeSpec, int) () from /usr/lib64/libQt5Core.so.5
#2  0x00007fc13eeea93a in QDateTime::fromTime_t(unsigned int) () from /usr/lib64/libQt5Core.so.5
#3  0x00007fc13ef86d46 in QFileInfo::lastModified() const () from /usr/lib64/libQt5Core.so.5
#4  0x00007fc13efc7312 in ?? () from /usr/lib64/libQt5Core.so.5
#5  0x00007fc13efc75ac in ?? () from /usr/lib64/libQt5Core.so.5
#6  0x00007fc13efbee26 in QSettings::~QSettings() () from /usr/lib64/libQt5Core.so.5
#7  0x00007fc13efbee49 in QSettings::~QSettings() () from /usr/lib64/libQt5Core.so.5
#8  0x00007fc13d70e330 in ?? () from /lib64/libc.so.6
#9  0x00007fc13d70e39a in exit () from /lib64/libc.so.6
#10 0x00007fc13d6f64f7 in __libc_start_main () from /lib64/libc.so.6
#11 0x00000000004507ea in _start ()

------------------------------------------------------------
A fix or probably workaround:
In the function "void MainWindow::closeEvent(QCloseEvent *event)" in  "./src/mainwin.cc",
change  "QSettingsCached settings;"  to  "QSettings settings;"  and no more Segmentation fault.

I am not sure if the problem is unique to my system (qt 5.7?? )or not. Or maybe the mutex thing in QSettingsCached causes some issue? It will be nice if the developer(s) can take a look into that.

kintel, 2017-11-06:
@DrCodeWizzard Does this sounds familiar to you?

DrCodeWizzard, 2017-11-07:
hmm... i hoped to avoid a global QSettings variable but it seems like this might have to be rewritten to do that. I wonder if this is related to internal clean up done by QT, i.e., we destroy the contents before they do (or vice versa).

MichaelPFrey, 2018-10-27:
could this be related to #2039?