openscad/openscad #2185: customizer mouse scrolling behavior

MichaelPFrey, 2017-11-07:
The customizer is using a scrollarea, meaning that when a lot of parameters are on display (or vertical space is at a premium), you have to scroll trough list.
Given that the spinboxes are sensitive to scrolling as well, there is little space to move.
With a slider or combox on screen, there is only the scrollbar left, where the user can scroll trough the entries.

Any ideas on how to solve that?

this issue originates from #1845

MichaelPFrey, 2017-11-07:
Some of my ideas:
  * fine tuning
    * fully disable the scrolling on the slider (-> the use can still scroll in the spin box)
    * fully disable scrolling on the comboxes 
    * right align the vector input
  * disable the scrolling in the widget, while the scrollarea is scrolling
  * disable the scrolling in the scrollarea, while the widgets are scrolling

MichaelAtOz, 2017-11-07:
I don't use any apps which have a similar layout.
Anyone come across this before in other apps?

MichaelPFrey, 2017-11-08:
Good observation. I mean, the layout (input elements like slider and combox in scroll area) is not uncommon, any website has to deal with that.

What is uncommon, that the input elements react to the scroll wheel.
We could simply disable the scroll wheel for most input elements and the user would not even realize that something is missing.
(yet: I do not want to do that - it really encourages users that think coding or 3d modeling is not theirs to use OpenSCAD. It lowers the bar of entry by giving these user imitate feedback with even using the keyboard)

As the scrollbar on the side only appears when there is not enough vertical space:
Discussing the general UI with a non-OpenSCAD user, a suggestion was to use tabs instead of folding boxes. Using tabs would also bring OpenSCAD more in line with http://customizer.makerbot.com/docs (and also give the Global Tab a reason for existing)

MichaelPFrey, 2018-01-05:
Working on a related problem in the preferences of the inputdriver (#2252), I found this: https://stackoverflow.com/questions/5821802/qspinbox-inside-a-qscrollarea-how-to-prevent-spin-box-from-stealing-focus-when

I specifically like the second answer, but I would not create our own class for spinbox, we can simply setFocusPolicy() and installEventFilter() (http://doc.qt.io/archives/qt-4.8/qobject.html#installEventFilter)

I will give that a try.

MichaelPFrey, 2018-03-10:
@MichaelAtOz Looking trough old issues: This issue originates from [this forum post](http://forum.openscad.org/GSoC-OpenSCAD-project-td18102i20.html) by you:

> There is conflict with the scroll wheel between scrolling the customizer pane, and changing slider values (and up/down, drop-down controls). If when scrolling the pane, the mouse happens to rest over a slider the scroll wheel then changes the slider instead of scrolling more. Not sure of best approach to this, as scrolling the controls could be useful, but when testing I had many unintentional changes to control values until I worked out what was going on. 

The my attempt of a fix should now be included in any "recent" build.
Would you mind testing the scroll wheel behavior so that we can update this issue?

t-paul, 2018-03-10:
New Windows builds on [files.openscad.com](http://files.openscad.org/snapshots/).

MichaelAtOz, 2018-03-10:
Ummm, hard to tell if scroll wheel is fixed...
[image]

Is there something I need to do?
This is the file used with Thingiverse Customizer.
Note I edited what was `/* [Hidden] */` & `/* [Global] */` in case they were involved.

MichaelAtOz, 2018-03-10:
Note #2337 may be related. When I uncheck the Automatic Preview box in the customizer panel, then select an item from the drop-down next to it, it still compiles/previews. I'm not sure whether it should?

MichaelAtOz, 2018-03-10:
Above source is [here](https://gist.github.com/MichaelAtOz/94ac481d66af361510371ccdeea6eb9b#file-sheriffsbadge-scad).

MichaelPFrey, 2018-03-10:
I now started my Windows 10. The customizer seems to be broken in the most recent build.
It is not related to the scad files beeing used, even the examples do not work.
Strange. I will look into it.

MichaelPFrey, 2018-03-11:
#2322 is the cause why it does not work under windows.

#2337 is unrelated, I was able to reproduce #2337 with the last build from 2017 (and I suspect the beviour is much older), while the costumizer was working with both december 2017 and january 2018 build and just broke with the march build.

MichaelPFrey, 2018-10-27:
I close this for now, as for half a year a workaround has been active and the same workaround now also in use in the preferences of the inputdriver.

closed 2018-10-27 by MichaelPFrey