openscad/openscad #2189: Crash when using mirror and using difference inside translate

cocsackie, 2017-11-09:
When working on a model I encountered crash. I managed to simplify it as much as it can be:

```
mirror([1,0,0]) union() {
    difference() {
        cube([60, 35, 14]);
        translate([0,31,0]) cube([45,3,3]);
    }
    
    translate([0,31,3]) {
        difference() {
            rotate([0,90,0]) cylinder(h=45, r=3);
            translate([0,-3,0]) cube([45,3,3]);
        }
    }
}
```

I cannot tell what this script is doing, but I hope its simple enough that you can debug it.

**What can be done, so that there will be no crash:**

- Removing mirroring
- Changing hierarchy so that `translate([0,31,3])` is inside `difference()` (other way around than it is now)
- Removing any line within difference (with or without removing difference itself)

**What still crashes:**

- Union replaced with module is crashing as well

This happend under Windows 7 (x64) on OpenSCAD 2015.03 as well as OpenSCAD 2017.01.20 commit 59df0d1

kintel, 2017-11-09:
Could you paste the result of Help->Library Info->Details for `OpenSCAD 2017.01.20`?

kintel, 2017-11-09:
My first hunch is that this is a duplicate of #1655

cocsackie, 2017-11-09:
Maybe it is duplicate, but I encountered the problem in stable version, and tried Development Snapshots and the bug occurs as well. Even if it's a bug in other artifact, the problem still exists here and by reading #1655 it was solved there. Then why newer version of that artifact isn't used?

```
OpenSCAD Version: 2017.01.20 (git 59df0d1)
System information: Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833 8 CPUs 7.98 GB RAM
Compiler: GCC "4.9.4" 64bit
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.6.3, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.7.0
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.44.1
libzip version: 
Application Path: C:/DOCUMENTS/openscad/openscad-2017.01.20
Documents Path: C:\Users\admin\Documents
Resource Path: C:/DOCUMENTS/openscad/openscad-2017.01.20
User Library Path: C:/Users/admin/Documents/OpenSCAD/libraries
User Config Path: C:\Users\admin\AppData\Local/OpenSCAD
Backup Path: C:/Users/admin/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/admin/Documents/OpenSCAD/libraries
C:/DOCUMENTS/openscad/openscad-2017.01.20\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 4.5.13399 Compatibility Profile Context 15.200.1062.1004
GL Renderer: AMD Radeon HD 6900 Series
GL Vendor: ATI Technologies Inc.
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QGLWidgetGL Extensions:
GL_AMDX_debug_output
GL_AMDX_vertex_shader_tessellator
GL_AMD_blend_minmax_factor
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_sample_positions
GL_AMD_multi_draw_indirect
GL_AMD_name_gen_delete
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_atomic_counter_ops
GL_AMD_shader_stencil_export
GL_AMD_shader_trace
GL_AMD_texture_cube_map_array
GL_AMD_texture_texture4
GL_AMD_transform_feedback3_lines_triangles
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_tessellator
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
[... 213 lines cut ...]
GL_IBM_texture_mirrored_repeat
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KTX_buffer_region
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
```

kintel, 2017-11-09:
#1655 was a bug in a 3rd party library - we had to wait until the fix was made available.
If you go to http://files.openscad.org/snapshots/, you can pick one of the later experimental Windows build. The main website hasn't been updated yet; we should probably do that.

cocsackie, 2017-11-09:
I downloaded OpenSCAD-2017.08.30-x86-64.zip, and it is not crashing. I was certain though, that the version from the website (developer snapshots) was the newest available.

kintel, 2017-11-09:
thx - we'll rebuild a new Windows version and update the website.

kintel, 2017-11-09:
Closing as duplicate of #1655

closed 2017-11-09 by kintel

cocsackie, 2017-11-09:
@kintel Sorry to address this again, but there still seems to be a problem, this is not a crash though. I opened the simplified model in the newest version of openscad and there was no crash. I returned to my original model and in console a CGAL error appeared.

Still using OpenSCAD-2017.08.30-x86-64.zip.

It seems to be more related to CGAL itself than openscad, but I am not aware of internals of openscad nor CGAL's, so I can't really tell. I hope if that is the case, that someone with knowledge of CGAL will be able to point the problem to CGAL developers.

Simplified again (similar to previous code):
```
module part() {
    difference() {
        cube([60, 35, 14]);
        translate([0,31,0]) cube([45,3,3]);
    }

    translate([0,31,3]) {
        difference() {
            rotate([0,90,0]) cylinder(h=45, r=3);
            translate([0,-3,0]) cube([45,3,3]);
        }
    }
}

part();
mirror([1,0,0]) part();
```

The error (above code):
```
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /opt/mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_external_structure.h Line: 1152
```

PS
Wasn't really sure if I should open a new issue or not... but the code above is very similar to previous one...

PS2
I am aware that this is developer snapshot version of openscad... though I am curios - why there was no stable build for a long time? or is it just because the Linux is primarily supported?

kintel, 2017-11-09:
PS We release simultaneously for all platforms, we just haven't gotten around to making a stable release in a while.

cocsackie, 2017-11-09:
Well, thank you for a quick answer.

I understand that current implementation or used libraries might have limitations which cause that workarounds are needed, but because error occurs instead of informing a user, that at this stage this program doesn't support this is causing confusion. Especially, that FAQ does mention only that coincident surfaces are going to make preview flicker (which can be easily be seen without any errors in CGAL console output occuring). AFAIK rendering is different from preview and works without flickering in those cases.

When there is an error in the source code, the user is notified, that he or she made a mistake. But in this case we get an error and there is no way to tell if it is user's fault and it even suggests that something that was not predicted occured (it's an assertion failure).

I suggest you consider this not only as a bug, but as an improvement potential as well.
Even if this is not possible to be done by openscad contributors, maybe you could persuade CGAL contributors to support it, so openscad could benefit from it.

kintel, 2017-11-09:
There's significant work to be done in this domain. For a taste, see https://github.com/openscad/openscad/labels/CGAL

Help is always welcome if you're interested in contributing towards improving  code, error messages, documentation, communication with CGAL staff etc.