openscad/openscad #2195: tan(90) is defined

thehans, 2017-11-14:
Run this code
`echo(tan(90));`

Result:  `1.63312e+16`
Expected result: `undef`

donbright, 2017-12-13:
this matches C++ because tan () is taking radians as input rather than the number 90. 

```
don@serebryanya:~/src/openscad/binmaster$ cat t.cc
#include <cmath>
#include <iostream>
#include <iomanip>
main() {
std::cout << tan(M_PI*90.0/180.0)<< "\n";
std::cout << std::setprecision(25) << tan(M_PI*90.0/180.0) << "\n";
}
don@serebryanya:~/src/openscad/binmaster$ g++ t.cc
don@serebryanya:~/src/openscad/binmaster$ ./a.out 
1.63312e+16
16331239353195370

```

Basically we are dealing with the difference between ideal "tan" function and "tan" as it exists in a computer, which is using approximations of an infinite process. 

I think the answer may lie somewhere within Niven's Theorem. Niven had some interesting things to say about when Sine has rational outputs aka value. There are only three angles, 0 30 and 90, producing 0, 1, 1/2. That's it. There are no other inputs to sine that will produce rational output.

Now those angles, 0 30 90, are rational numbers but if you convert to radians, they are 0, pi/6, pi/2, but pi is an irrational number with an infinite number of digits. So you can't get the rational results you want, if your system is based on radian input. There is no rational radian that is equivalent to "30 degrees". In other words, radians and degrees are not equivalent if one is working with a finite number of digits, which one always is inside of a computer. 

Soooo lets say you built a computer trigonometry system that was built on the idea of Sine having angles for input. Then you could give it exactly 30 degrees, or 90 degrees, and get a perfect results of  1/2 or 1 as output. That would be interesting, but it would also mean you re-implemented an entire mathematical system for the benefit of two special input numbers (and their multiples), out of several billion.

Now OpenSCAD actually does use angles for input.... and it converts them to radians. But what if OpenSCAD were to treat those "special angles" as "special numbers" for each trig function, and skip over C++ math functions and return the rational value? For example sin(30)=.5 instead of 0.4999999999999999444888488?

Then if you tried to do smooth rotation you might get these weird "jumps" at those angles? I don't know.  But if you tried to convert a python or C# or javascript or whatever program to OpenSCAD, you might get different results, because just about the whole world uses radians and the C style math concepts for their backend. I also wonder if there would be backwards compatability issues... let's say someone depended on Tan() producing a really big output, but they never expected it to be NaN. NaN will crash some calculations, IIRC matrix math or something. 

Those are my theories... could be wrong but..

https://en.wikipedia.org/wiki/Niven%27s_theorem

 Weisstein, Eric W. "Niven's Theorem." From MathWorld--A Wolfram Web Resource. http://mathworld.wolfram.com/NivensTheorem.html

nophead, 2017-12-13:
>But what if OpenSCAD were to treat those "special angles" as "special numbers" for each trig function, and skip over C++ math functions and return the rational value? For example sin(30)=.5 instead of 0.4999999999999999444888488?
It does exactly that. The user mode sin and cos functions have been tweaked to give exact results for special angles but tan hasn't. For example tan(45) != 1.

MichaelAtOz, 2017-12-14:
I like to think I live in a rational world, all evidence on the news not withstanding...
So the history of this is #1602 via [forum](http://forum.openscad.org/Bug-in-OpenScad-rendering-td16396.html#a16406).
While maths theory is fine, I like my solids to align in the real world, so if I rotate something 90 degrees to align with something else, I'd like it to rotate 90 degrees, not 89.0000019387t68548392.
Yes if it is some value other than 0/30/90 I may have to work with small differences, but 0 and 90 are so common in the real world it would be handy if they were rational, (30 I don't see much use for, but if you're going to do it, may as well make Niven happy...)

MichaelAtOz, 2017-12-14:
Tho I would think tan(90) should be [inf](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/General#Values_and_Data_Types)

thehans, 2017-12-14:
@MichaelAtOz its nan because it basically could be inf or -inf

doug-moen, 2017-12-14:
@thehans Mathematically, you are correct. tan(90) approaches inf from one side, and -inf from the other side. So it should be undefined.

However, we are using IEEE floating point arithmetic, not real arithmetic, and that changes things.

Suppose that somebody needs to compute the cotangent:
```
function cot(a) = 1 / tan(a);
```

cot(90) should equal 0. Right now, without your fixes, cot(90) returns 6.12323e-17, which is pretty close to 0. Mathematically, tan(90) is the superposition of inf and -inf. In IEEE float arithmetic, 1/inf is 0, and 1/(-inf) is -0. And 0 == -0.

In short, tan(90) should be inf so that cotangent calculations do not break.

thehans, 2017-12-14:
To me, that seems more like an argument for cot being an openscad native function just like tan so it too can handle special values.

@doug-moen What would you suggest for infinite other data points where:   a = 180*(i ∈ ℤ)+90 
Do you think they should all be positive infinity?

doug-moen, 2017-12-14:
There is also a backward compatibility issue. There may be existing OpenSCAD programs that use tan(a) as a divisor, because they are performing cotangent-like calculations. We don't want to break existing programs, and `inf` for tan(90) will be backward compatible with better accuracy in these cases.

My intuitive preference is to always use +inf, regardless of the sign of `a`. I know that the C library  `tan` function seems to disagree with this. However, I just checked a bunch of programming languages that have trig functions that work in degrees, and tan(-90)==+inf seems to be an industry standard.

A number of programming languages have a `tand` function which takes an argument measured in degrees.

In Matlab, For odd integers n , tand(n*90) is infinite.

In GNU Octave, Returns zero for elements where x/180 is an integer and Inf for elements where (x-90)/180 is an integer.

In R, the same: Zero is returned for any elements where x / 180 is an integer and Inf for elements where (x - 90) / 180 is an integer.

doug-moen, 2017-12-14:
I found a web browser implementation of Matlab. tand(-90) is -Inf, tand(-270) is Inf. Not sure why, it seems arbitrary.

I tried octave-online.net, and the result is always +Inf, never -Inf.

thehans, 2017-12-14:
Yeah that sounds like the third option I was kinda considering in my head but hadn't articulated.  

If you say that tand(-90) is -inf and tand(90) is +inf, and the function is periodic, then it should just alternate plus/minus every 180 degrees i guess.

thehans, 2017-12-14:
So the option of sign(a)*inf would break periodicity, which I think is no good.  Scrap that one.
I guess I could go either way with the two remaining options now:  alternating signs, or always positive.

doug-moen, 2017-12-14:
Tangent is (normally) periodic with a period of 180°. If tan(-90) is -inf and tan(90) is +inf, with alternating signs every 180°, then the function becomes periodic with a period of 360°.

So I guess we have to choose between oddness and a period of 180° ?

thehans, 2017-12-14:
I've just updated my PR to implement the alternating sign option.  After considering the options in our discussion here, I really feel its as close to correct as we can possibly do.  

The tests are cleaner now without the necessity to check for nan results, so that's nice.

And yes I did have to update the period to 360 in the test case.  Of all the aspects discussed I feel like that one is of the least consequence.  (I mean, its still periodic over 180 degrees if you omit singularities).

thehans, 2017-12-14:
And for what its worth, before my changes, tan failed the periodicity check for what looks like every angle.

doug-moen, 2017-12-15:
Matlab definitely implements alternating signs for tand of -90, 90, 270, ...

Think about what this implies for cotd(a) = 1/tand(a). What is the effect of alternating signs on this definition of cotd? If you think about that, then you realize that for consistency you also need alternating signs for the angles where tand yields 0, and that is exactly what Matlab implements as well.

tand(0) == 0, tand(180) == -0, tand(360) == 0, and so on.

As a result of this, cotd also ends up with alternating signs for inf/-inf and 0/-0 results.

thehans, 2017-12-15:
OK I added support for negative zero and corresponding tests.  

_Now_ is everyone satisfied?  :stuck_out_tongue:

thehans, 2017-12-26:
#2215 has been merged, closing

closed 2017-12-26 by thehans