openscad/openscad #2200: Setting $variable with -D on the command line gives them too much scope.

nophead, 2017-11-20:
Setting a $variable on the command line should just append that variable to the end of the main file. Indeed the code seems to do exactly that. However with recent versions they have a different effect. Instead of just setting the variable at file scope it seems to override any setting at module scope. Module scope should always trump file scope with dynamic scoping.

Here is an example that goes wrong:
use.scad:
------------
```
$bom = 0;

use <used.scad>

test();

let($bom = 0)
   test();
   
$bom = 1;
```
used.scad:
--------------
```
use <used2.scad>

module test() {
    echo("test:", $bom = $bom);
	test2();
}
```
used2.scad:
----------------
```
module test2() {
    echo("test2:", $bom=$bom);
}
```
In the GUI this gives:
```
ECHO: "test:", $bom = 1
ECHO: "test2:", $bom = 1
ECHO: "test:", $bom = 0
ECHO: "test2:", $bom = 0
```
From the command line with ```-D$bom=1``` it gives:
```
ECHO: "test:", $bom = 1
ECHO: "test2:", $bom = 1
ECHO: "test:", $bom = 1
ECHO: "test2:", $bom = 1
```

kintel, 2017-11-20:
This behavior changed in 2012, interestingly enough due to a bug report from you :)
See #211

nophead, 2017-11-20:
So it appends the -D options to all used files. But that only sets it at file scope everywhere. When you set the same variable in a module it overrides it. When it is a $variable it has dynamic scope visible in the children of that module and the modules it calls. Surely that should override file scope definition in used modules as well?

MichaelAtOz, 2017-11-20:
I put $bom=2 at end of used.scad & $bom=3 at end of used2.scad.
(also added msg= parameter & an assign() to test whether let() was a problem)
```
ECHO: "test:  msg=", "top", ", $bom='", 2
ECHO: "test2: msg=", "from test()", ", $bom='", 3
ECHO: "test:  msg=", "let", ", $bom='", 2
ECHO: "test2: msg=", "from test()", ", $bom='", 3
DEPRECATED: The assign() module will be removed in future releases. Use a regular assignment instead. 
ECHO: "test:  msg=", "asn", ", $bom='", 2
ECHO: "test2: msg=", "from test()", ", $bom='", 3
```
For test(), $bom=2 as the file scope wins becasue there is no $bom in the paramters of the test() call.
Ditto for test2.

i.e. let($bom=1) has less scope level to test() $bom file level.

I think that is as expected.
The issue is that #211 solution is not optimal...

nophead, 2017-11-21:
It shouldn't matter that $bom is not passed has a parameter. It has dynamic scope, so should trump file scope all the way up the call stack, not just up the call stack in the local file but not extend into the used one.

Yes #211 is not optimal and not actually needed. I found more recently that the file scope of the using module is passed to the used module but doesn't override a definition there. So I use this in my global include and actually test $_bom.
```
$_bom          = $bom             == undef ? 0      : $bom;  
```

This means a single definition in the main file works fine to affect all the others in the GUI and would on the command line if #211 was reverted.

kintel, 2017-11-22:
That's indeed a bit odd. I wonder if there is some lookup bug which prefers a file-scoped variable to a dynamically scoped one. Needs some careful investigation..

nophead, 2017-11-22:
They are looked up here:https://github.com/openscad/openscad/blob/0904a3b3a5b56b16bb423b177972468eeca775a9/src/context.cc#L132.

I think when a used module is instantiated its config variables are part of its context. It is higher on the stack so gets searched first when they are looked up.

There seems to be no segregation in the context between file scope config variables and module scope ones. I think there needs to be so module scope ones can be searched before file scope.

kintel, 2023-11-27:
#3761 cleans up special variable passing, and makes the behavior more well-defined. Closing this issue as obsolete. Please open a new issue if there is still unclear behavior present.

closed 2023-11-27 by kintel