openscad/openscad #2202: feature request ,  scad file import / export as plain ascii text and a print function

achiestdragon, 2017-11-20:
only spotted this when needing to pint a scad source file and also finding its not saved in txt format also 
sorry 2 requests one issue

kintel, 2017-11-21:
.scad _is_ saved in plain text, so you should be able to print from any program that can read text files. A separate print function is a perfectly reasonable feature request.

achiestdragon, 2017-11-21:
ho that explains a lot , had tried to open some i did some time back and just got something that looks like a bin file , so just assumed they where not , must be a corrupt archive , np , just checked  on a current one and its is ok , ,never mind , print would still be useful though

thehans, 2017-12-03:
Looks like your file data got corrupted somehow(failing disk drive possibly?) but yes valid `.scad` files are just plain text, and can be viewed and edited from the openscad editor.

@achiestdragon What do exactly do you mean by a print function?  You want to send the `.scad` source file directly to a paper printer from openscad?

achiestdragon, 2017-12-03:
yea   idealy with colour highlighting

amatulic, 2025-04-21:
Seven years later, maybe this issue can be closed? Unless there's an OS-independent "standard printer" spooler output device available in one of the C++ libraries used by OpenSCAD, I don't see this one getting done.

If I ever needed to print an OpenSCAD file, I'd just load the .scad file into an external editor and print it from there. And in the years I've been using OpenSCAD, the need to print my scripts has never come up. The laser printer we got 8 years ago hardly gets used anymore. Any text I need to communicate to others is sent electronically rather than on paper.

t-paul, 2025-04-21:
And the open issue bothers you why exactly? While the headline says "printing", it could be reinterpreted as "format to PDF" which would cover both printing on paper and sending/storing in places that do not like just any file format (like some archiving systems demanding PDF/A).

It's obviously not a very high priority topic, but there's no harm in keeping it in case someone is interested in taking it on.