openscad/openscad #2205: Feature request: first-class functions

thehans, 2017-11-28:
I'd like to be able to pass a function as a parameter to another function.

I feel like this is one of the keystones of functional programming, and a functional language without this capability is really handicapped.

Stretch goals(nice to have, but not as crucial in my opinion):
- function as a parameter to module
- function as return value of a function

kintel, 2017-11-28:
See also https://github.com/doug-moen/openscad2/blob/master/rfc/Functions.md

doug-moen, 2017-11-28:
The OpenSCAD2 project is no longer active; it's now called Curv, as I've
dropped backward compatibility with OpenSCAD as a goal. First class
function values in OpenSCAD was my original goal for the project.

There are a number of barriers to extending OpenSCAD with function values.
The latest notes I wrote on this problem, before I gave up, are here:
https://github.com/doug-moen/curv/blob/master/ideas/OpenSCADbis.md

There is the namespace problem. Function names belong to a different
namespace than parameter names. It is not syntactically possible to call a
function parameter as a function. The usual proposal involves introducing
some ugly syntax, such as @variablename(arg) for calling a function value
stored in a variable. The OpenSCADbis paper linked above describes an
alternative to inventing ugly syntax.

There is the lexically scoped closure problem. In any modern programming
language, functions are lexically scoped, and a function value captures the
values of non-local variables so that these values are still accessible
within the body of the function even after the non-local variables have
ended their lifetimes. The simplest implementation (at least, in a pure
functional language like OpenSCAD) is to copy the values of non-local
variables. In OpenSCAD, there are also non-local modules and non-local
functions (since functions, modules and variables exist in distinct
namespaces). OpenSCAD does not consistently use lexical scoping. I once
posted an example where the same non-local variable (not beginning with a
$) behaves as either lexically scoped, or dynamically scoped, depending on
the location within the program from which the function is called. It's not
clear if we can treat this as a bug and fix it. It's also not clear if we
are supposed to treat this as a feature, and faithfully duplicate those
semantics when functions become values. My proposal is to introduce a new
function definition syntax, which defines "new functions", which are
lexically scoped and can be treated as values. The "new function" idea is
also part of the scheme for avoiding ugly syntax.

There is the storage leak problem. When a recursive function is converted
into a value, the obvious implementation will result in a cyclic data
structure. This is fine if you use a tracing garbage collector, but
OpenSCAD uses reference counting. Cyclic data structures cause storage
leaks if you use reference counting. In Curv, I decided to take on the
challenge of implementing first class recursive function values, using
reference counting, without creating storage leaks. Although I succeeded in
implementing this, it requires an interesting data structure for function
values, and some tricky compiler infrastructure. So the choices are:
 1. Leak storage and don't worry about it.
 2. Convert OpenSCAD to using a tracing garbage collector. The easiest
approach might be to use the Boehm collector, as you could do this with few
code changes.
 3. Use the Curv design for reference counted function values. Ask me about
the details if you are serious.

Doug Moen.

ringerc, 2019-10-21:
It sounds to me like any "OpenSCAD v2" / Curv / etc would benefit from adopting a modern language runtime as a foundation, rather than developing a new language afresh from scratch. Then building a DSL (domain specific language) on top.

That way users would be able to access a wide variety of maths and geometry libs, rendering tools, and more.

The main issue seems to be restricted execution safety. You don't want your object definitions running off and uploading your password files to the Internet on the sly. But a number of runtimes have security models, like Java.

Personally I'd love to be able to write my OpenSCAD models in Scala or OCaml.

ringerc, 2019-10-21:
I, too, really want first class functions (and modules).

Sometimes invoking children by ordinal is a tolerable if ugly alternative. Brace blocks don't group up children, so you have to create wrapper modules, e.g.:

```
module outer()
{
    echo("doing foo children");
    children(0);

    echo("doing bar children");
    children(1);
}

module inner_foo(blah)
{
  echo("foo", blah);
}

module inner_bar(blah)
{
  echo("bar ", blah);
}

module foo_children()
{
    inner_foo("once");
    inner_foo("again");
}

module bar_children()
{
    inner_bar("works");
    inner_bar("but ugly");
}

outer()
{
    foo_children();
    bar_children();
}
```

Another way seems to be to pass child indices, but it's worse if anything:

```
module outer(foo_children=[], bar_children=[])
{
    echo("doing foo children");
    children(foo_children);

    echo("doing bar children");
    children(bar_children);
}

module inner_foo(blah)
{
  echo("foo", blah);
}

module inner_bar(blah)
{
  echo("bar ", blah);
}

outer(foo_children=[0:1:1], bar_children=[2:1:3])
{
    inner_foo("once");
    inner_foo("again");
    inner_bar("works");
    inner_bar("but ugly");
}
```

so I landed up using a `for` block with a single iteration as a poor-man's `with` expression, since OpenSCAD doesn't appear to have a way to define a variable that's both scoped within `{` blocks and inherited by child modules. So I have this monstrosity:

```
module outer()
{
    echo("doing foo children");
    for ($child = ["foo"])
    {
      children();
    }

    echo("doing bar children");
    
    for ($child = ["bar"])
    {
      children();
    }
}

module inner_foo(blah)
{
  echo("foo", blah);
}

module inner_bar(blah)
{
  echo("bar ", blah);
}

outer()
{
    echo(child=$child);
    if ($child == "foo")
    {
        inner_foo("once");
        inner_foo("again");
    }
    if ($child == "bar")
    {
        inner_bar("works");
        inner_bar("but ugly");
    }
}
```

ringerc, 2019-10-21:
See also the thread [passing parameters to children()](http://forum.openscad.org/Passing-parameters-to-children-td4808.html) which discusses this.

Though I think the conclusions it draws are completely erroneous. People there draw parallels to varargs in C and C++, but OpenSCAD's language is nothing like those.

With that said, the lack of `defined($variable)` tests, lack of associative arrays, etc, does make it harder.

Other related issues:

* https://github.com/openscad/openscad/issues/2104
* https://github.com/openscad/openscad/issues/1596

t-paul, 2019-10-21:
The issue is about first-class functions which are pretty much there (https://github.com/openscad/openscad/pull/3077). Further discussion about modules/objects in https://github.com/openscad/openscad/issues/3088.

Also there's a list of type test functions in v2019.05, including `is_undef()` - see [cheatsheet](http://www.openscad.org/cheatsheet/).

doug-moen, 2019-10-22:
@ringerc 
> Personally I'd love to be able to write my OpenSCAD models in Scala or OCaml.

You can. There are embeddings of OpenSCAD for a number of languages. For Scala, try <https://github.com/dzufferey/scadla>. I've also seen OpenSCAD embeddings for Javascript, Python, and Haskell.

OpenSCAD is a strange language, with a lot of non-standard semantics: 3 namespaces for variables, functions and modules; very odd semantics when there are multiple definitions of the same variable in a scope; a weird form of dynamic binding in module calls. It doesn't really map onto a modern language runtime. So a bidirectional interface between OpenSCAD and a conventional language (either language can call functions or modules in the other) would be incredibly complex and messy, and is unlikely to be implemented or used.

What can be implemented is:
* A reimplementation of OpenSCAD without the non-standard semantics, compatible with a modern language runtime, which can be interfaced with a modern language. This would have limited backward compatibility with existing OpenSCAD code, but existing OpenSCAD code could be manually ported in one direction to the new dialect.
* A code generator. You use an OpenSCAD-like DSL in Scala or Python or whatever to describe geometry, which is then compiled into OpenSCAD code.

None of these ideas is a good fit for the OpenSCAD project itself, which is commited to backward compatibility with the vast ecosystem of existing OpenSCAD scripts.

t-paul, 2019-10-22:
Well, namespaces for various things... maybe some looked at C++? :smiley: 
```
struct x { int x; };

x x(x x) { return x; }

int main() {
    return x([](){struct x x{.x = 42}; return x;}()).x;
}
```
On a more serious note... I don't see the language as that much of a problem. Yes, we do need a couple of extensions to make it more powerful, while keeping the simplicity of just `cube();` to get started. However even more important are extensions to the geometry engine.

t-paul, 2019-10-24:
Closed via #3077.

closed 2019-10-24 by t-paul