openscad/openscad #2206: cheatsheet or part of documentation should be part of app to work in offline mode

jtoy, 2017-11-30:
I use the app offline often and when I click on any of the help items, it takes me online which i cant go online. I think at least some basic stuff like the cheatsheet should be part of the app so people can use the app without having to be online.

donbright, 2017-12-12:
i find this request very interesting. would you be willing to tell more about your use case? what kind of setting and what kind of hardware / operating system? 

i think it is not too difficult, qt includes html rendering that we probably use somewhere already

jtoy, 2017-12-12:
The usecase is having developers be able to use the app in a fully self contained way. I am  offline often and when I use go to use the app I cannot look up any documentation because it just links to online. Not everyone has a 24/7 connection to the internet.

t-paul, 2017-12-12:
There's also a script that scrapes the Wiki manual and creates a standalone version. But there's no good idea yet how to integrate that. We can't scrape on every build.

thehans, 2017-12-12:
@t-paul Can you clarify why we can't?  Do you mean because it would make the build susceptible to vandalism?  If that's the issue, is it possible to "tag" a revision of the wiki which is known good?

t-paul, 2017-12-12:
Even just OBS is running about 40 builds for every check-in. I don't want to be blamed for DDOS-ing Wikibooks if we have multiple check-ins.
Well, OBS might not allow network access in the first place, so instead of scraping every time it probably would just fail. Not sure what's a good solution, maybe some kind of staging in a separate repo?

donbright, 2017-12-13:
yeah i didnt want to deal with the wikibook, thats too much, and too big. this is just for the minimal cheatsheet. given the cheatsheet itself, a reasonably dedicated user could figure out most of the stuff in the wikibook.

thehans, 2017-12-13:
BTW #2177, while not the same issue, would more or less address the root of the problem in my opinion.

MichaelAtOz, 2018-04-28:
$25 Bounty added.

ixil, 2019-06-11:
https://github.com/ixil/openscad_offliner So I was bored sometime last month - then forgot about it - then remembered and so here is this? 
it may be wip - I can't actually recall whether I finished this or not, let me know

t-paul, 2022-10-01:
Implemented via #3853.

@abshk-jr Looks like we missed this issue with the bounty attached, feel free to claim it.

@MichaelAtOz It's not yet enabled in the snapshot builds, but I've tested locally and people have used it in the meantime (#4214 / #4298), it's enabled in the Arch Linux [openscad-git](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=openscad-git) package.

closed 2022-10-01 by t-paul

MichaelAtOz, 2022-10-02:
@t-paul ok.
Note the Bountysource tool is broken. Not getting the BS added to new issues.
Shows an error on BS, I'll refresh my understanding, then try the power switch method... ;)

> This plugin is locked.
`Reason: Forbidden -- {:url=>"/repos/openscad/openscad.github.com/issues/72?access_token=<snip>&client_id=<snip>&client_secret=<snip>", :type=>"PATCH", :body=>{:body=>"The URLs in the appcast.xml file appear to be incorrect and are throwing 404 error...`

t-paul, 2022-10-03:
Hmm, that link is the website repo. Maybe it got confused?

MichaelAtOz, 2022-10-08:
Boutysource Tools fixed. That didn't prevent claims tho.

@abshk-jr if you want to claim the bounty go [here](https://app.bountysource.com/issues/52162376-cheatsheet-or-part-of-documentation-should-be-part-of-app-to-work-in-offline-mode?utm_campaign=plugin&utm_content=tracker%2F52063&utm_medium=issues&utm_source=github), if you haven't joined, just Sign-In with GitHub, via the Cat icon on the Sign-In page.

abshk-jr, 2022-10-11:
Just saw this and submitted a claim.
Thanks for letting me know!

MichaelAtOz, 2022-10-12:
Claim accepted. Thanks.