openscad/openscad #2217: Cannot open a folder that ends in ".scad" on Mac OSX

cfinke, 2017-12-15:
Steps to reproduce

1. Name a folder "test.scad" and put some files in it.
2. In OpenSCAD, choose `File > Open`.
3. Navigate to the folder containing the `test.scad` folder and try to open it.

What I expected: I would be able to open the `test.scad` folder and choose a file from within it.

What happened: OpenSCAD treated the directory as if it was a file, and it wrote this message to the console: `Failed to open file /Users/cfinke/test.scad: file to open is a directory`

Tested on both 2017.12.02 and 2015.03-3 on Mac OS X 10.12.5.

deckar01, 2017-12-27:
This is not an OpenSCAD bug, [it is a quirk of Mac OS][0]. When you give a folder an extension, Mac OS treats it as a "package". To view the package's contents you have to command click it in finder and select the "Show Package Contents" option. Any application that can only open files will behave the same way.

[0]: https://apple.stackexchange.com/questions/123001/renamed-folder-becomes-a-file-with-an-extension

cfinke, 2017-12-28:
That makes sense, but when I the same thing in Firefox (which can only open files), it opens the folder within the Open dialog without any issues.

kintel, 2017-12-28:
We use Qt to deal with file file dialogs. Perhaps Qt has some settings related to this?